OpenZWave/open-zwave #478: try compile but failed

nechry, 2015-03-17:
make
make -C /opt/python-openzwave/openzwave/cpp/build/ - 
make[1]: Entering directory '/opt/python-openzwave/openzwave/cpp/build'
Building vers.o
/opt/python-openzwave/openzwave/cpp/src/vers.cpp:4:30: error: expected primary-expression before ‘;’ token
/opt/python-openzwave/openzwave/cpp/build/support.mk:102: recipe for target '/opt/python-openzwave/openzwave/.lib/vers.o' failed
make[1]: *** [/opt/python-openzwave/openzwave/.lib/vers.o] Error 1
make[1]: Leaving directory '/opt/python-openzwave/openzwave/cpp/build'
Makefile:20: recipe for target 'all' failed
make: *** [all] Error 2

the cpp/src/vers.cpp dont contain the revision number

Fishwaldo, 2015-03-17:
This is due to the switch to Git... should be fixed for *nix platforms. I'll work on Windows builds soon.

Fishwaldo, 2015-03-19:
Should be fixed now.

closed 2015-03-19 by Fishwaldo

nechry, 2015-03-19:
hello, yes it's compile now, but the revision number is not right
#include "Defs.h"
uint16_t ozw_vers_major = 1;
uint16_t ozw_vers_minor = 3;
uint16_t ozw_vers_revision = 103;
char ozw_version_string[] = "1.3-103-g3be4793";

normally we must around 1028, the 103 it's long time ago. or is something like 1030 without the last 0 ?