OpenZWave/open-zwave #479: Fibaro FGS211 and FGS221 param3 not right

nechry, 2015-03-17:
Hello, 
I try to set the parameter 3 and I got with 2 possible option the default   
value. I take a look in manual and I found a mistake in actual   
configuration files. 

is the definition 
Parameter No. 3 - Auto off relay after specified time, with the 
possibility of manual override - immediate Off after button push. 
Default value: 0 
Available settings: 
0 - manual override disabled. After single button push the relay turns 
on and automatically turns off after specified time. 
1 - manual override enabled. After single button push the relay turns 
on and automatically turns off after specified time. Another button 
push turns the relay off immediately 

Is a patch for Fibaro FGS211 and FGS221. the file is on https://code.google.com/p/open-zwave/issues/detail?id=475

nechry, 2015-03-20:
the pull request #486 and #487 provide patch on config. 
https://github.com/OpenZWave/open-zwave/pull/486
https://github.com/OpenZWave/open-zwave/pull/487

is the right way to propose device patch?

ualex73, 2015-03-21:
Yes it is a good/right way to propose a patch - only we need to think about the Changelog update. This is still work-in-progress :-)

closed 2015-03-21 by ualex73