OpenZWave/open-zwave #481: Zipato RFid mini keypad find Scanned UserCode

nechry, 2015-03-19:
in reference to topic https://groups.google.com/forum/#!topic/openzwave/e-fphZ8k0Z8 

I use the badge reader RFid Zipato mini keypad a rebranded  BeNext tagreader.

I'm looking a way to detect witch UserCode index has been scanned. I know it is possible to retrieve this information, but isn't implemented in openzwave librarie.

Another element, the Alarm Level  follow Home(0)/Away(255) value, but normally I' think is for sabotage.

The home/away mode can be handle with value "Access Control"  from commandClass (0x71) index 9
Value=6 for Home and 5 for Away.

Here the frame when I scan a badge in Home :

2015-03-12 00:23:07.147 Detail, 
2015-03-12 00:23:07.147 Info, Node060, Received Alarm report: type=0, level=0
2015-03-12 00:23:07.148 Detail, Node060, Refreshed Value: old value=0, new value=0, type=uint8
2015-03-12 00:23:07.148 Detail, Node060, Changes to this value are not verified
2015-03-12 00:23:07.149 Detail, Node060, Refreshed Value: old value=255, new value=0, type=uint8
2015-03-12 00:23:07.149 Detail, Node060, Changes to this value are not verified
2015-03-12 00:23:07.150 Detail, Node060, Refreshed Value: old value=255, new value=255, type=uint8
2015-03-12 00:23:07.150 Detail, Node060, Changes to this value are not verified
2015-03-12 00:23:07.151 Detail, Node060, Refreshed Value: old value=5, new value=6, type=uint8
2015-03-12 00:23:07.151 Detail, Node060, Changes to this value are not verified
2015-03-12 00:23:07.163 Detail, Node060,   Received: 0x01, 0x08, 0x00, 0x04, 0x00, 0x3c, 0x02, 0x84, 0x07, 0x4e
2015-03-12 00:23:07.163 Detail, 
2015-03-12 00:23:07.164 Info, Node060, Received Wakeup Notification from node 60
2015-03-12 00:23:07.164 Info, Node060,   Node 60 has been marked as awake
2015-03-12 00:23:07.165 Detail, Node060, Queuing (WakeUp) WakeUpCmd_NoMoreInformation (Node=60): 0x01, 0x09, 0x00, 0x13, 0x3c, 0x02, 0x84, 0x08, 0x25, 0x87, 0xf5
2015-03-12 00:23:07.166 Detail, 
2015-03-12 00:23:07.166 Info, Node060, Sending (WakeUp) message (Callback ID=0x87, Expected Reply=0x13) - WakeUpCmd_NoMoreInformation (Node=60): 0x01, 0x09, 0x00, 0x13, 0x3c, 0x02, 0x84, 0x08, 0x25, 0x87, 0xf5
2015-03-12 00:23:07.174 Detail, Node060,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2015-03-12 00:23:07.174 Detail, Node060,   ZW_SEND_DATA delivered to Z-Wave stack
2015-03-12 00:23:07.185 Detail, Node060,   Received: 0x01, 0x05, 0x00, 0x13, 0x87, 0x00, 0x6e
2015-03-12 00:23:07.185 Detail, Node060,   ZW_SEND_DATA Request with callback ID 0x87 received (expected 0x87)
2015-03-12 00:23:07.186 Info, Node060, Request RTT 18 Average Request RTT 18
2015-03-12 00:23:07.186 Info, Node060,   Node 60 has been marked as asleep
2015-03-12 00:23:07.187 Detail, Node060,   Expected reply was received
2015-03-12 00:23:07.187 Detail, Node060,   Message transaction complete

Here the frame when I scan a badge in Away:
2015-03-12 00:22:53.810 Detail, Node060,   Received: 0x01, 0x10, 0x00, 0x04, 0x00, 0x3c, 0x0a, 0x71, 0x05, 0x00, 0xff, 0x00, 0xff, 0x06, 0x05, 0x01, 0x04, 0xaf
2015-03-12 00:22:53.810 Detail, 
2015-03-12 00:22:53.811 Info, Node060, Received Alarm report: type=0, level=255
2015-03-12 00:22:53.811 Detail, Node060, Refreshed Value: old value=0, new value=0, type=uint8
2015-03-12 00:22:53.811 Detail, Node060, Changes to this value are not verified
2015-03-12 00:22:53.811 Detail, Node060, Refreshed Value: old value=0, new value=255, type=uint8
2015-03-12 00:22:53.812 Detail, Node060, Changes to this value are not verified
2015-03-12 00:22:53.812 Detail, Node060, Refreshed Value: old value=255, new value=255, type=uint8
2015-03-12 00:22:53.812 Detail, Node060, Changes to this value are not verified
2015-03-12 00:22:53.812 Detail, Node060, Refreshed Value: old value=6, new value=5, type=uint8
2015-03-12 00:22:53.813 Detail, Node060, Changes to this value are not verified
2015-03-12 00:22:53.826 Detail, Node060,   Received: 0x01, 0x08, 0x00, 0x04, 0x00, 0x3c, 0x02, 0x84, 0x07, 0x4e
2015-03-12 00:22:53.827 Detail, 
2015-03-12 00:22:53.827 Info, Node060, Received Wakeup Notification from node 60
2015-03-12 00:22:53.827 Detail, Node060, Queuing (WakeUp) WakeUpCmd_NoMoreInformation (Node=60): 0x01, 0x09, 0x00, 0x13, 0x3c, 0x02, 0x84, 0x08, 0x25, 0x86, 0xf4
2015-03-12 00:22:53.828 Detail, 
2015-03-12 00:22:53.828 Info, Node060, Sending (WakeUp) message (Callback ID=0x86, Expected Reply=0x13) - WakeUpCmd_NoMoreInformation (Node=60): 0x01, 0x09, 0x00, 0x13, 0x3c, 0x02, 0x84, 0x08, 0x25, 0x86, 0xf4
2015-03-12 00:22:53.835 Detail, Node060,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2015-03-12 00:22:53.835 Detail, Node060,   ZW_SEND_DATA delivered to Z-Wave stack
2015-03-12 00:22:53.847 Detail, Node060,   Received: 0x01, 0x05, 0x00, 0x13, 0x86, 0x00, 0x6f
2015-03-12 00:22:53.848 Detail, Node060,   ZW_SEND_DATA Request with callback ID 0x86 received (expected 0x86)
2015-03-12 00:22:53.848 Info, Node060, Request RTT 19 Average Request RTT 18
2015-03-12 00:22:53.848 Info, Node060,   Node 60 has been marked as asleep
2015-03-12 00:22:53.848 Detail, Node060,   Expected reply was received
2015-03-12 00:22:53.849 Detail, Node060,   Message transaction complete

thank, if you can implement this feature in a future milestone. Also I'm available to test some patch

nechry, 2015-04-22:
hello, In Zipato and Benext documentation for Alarm Class 0x71 is wrote:
CLASS: 0X71 COMMAND_CLASS_ALARM_V2 
In the Mini Keypad RFiD/Z-Wave, this Command Class has two purposes:

1 | Identify the state of the tamper alarm. The device will send an unsolicited report to the devices in its association group if tempering is detected. The state of the tamper alarm can also be requested
by any other device.

2 | Report tags or codes that are entered. The Mini Keypad RFiD/ZWave will send an unsolicited report to the devices in its association group with the UID that belongs to the code or tag and whether the alarm system should be armed (Away) or disarmed (Home).

Every other alarm type that is requested will be ignored by application.

The actuel specific configuration file have this Element:

<CommandClass id="113" getsupported="false" />

Is can be just remove this part to fix this issues? or I don't understand the meaning of this element

nechry, 2016-06-08:
complement / duplicate:
https://github.com/OpenZWave/open-zwave/issues/145

Fishwaldo, 2016-06-28:
I've just re-factored the Alarm CC on the Dev Branch. Its changing the way the ValueID's are exposed compared to to 1.4 version. Please see https://github.com/OpenZWave/open-zwave/wiki/Alarm-Command-Class for details on the new implementation.

closed 2016-06-28 by Fishwaldo