OpenZWave/open-zwave #482: OZW Google Code Website has Moved?

alphe, 2015-03-19:
Hello, I'm a great fan of your work. 
I would like to know why your usual tracker code distribution site is gone?
I don t see any anouncement of the site moving from OZW dev team part, so I wonder what happend?
How can we get the most recent release from github?

Fishwaldo, 2015-03-19:
It was discussed here: https://groups.google.com/d/msg/openzwave/hHX4OIySalE/CGie1_hJ0gMJ

New Item posted here: http://openzwave.com/home/news/2015/03/19/moving_to_github/more

and you can either use the snapshots from the main page, or learn how to use git. :)

closed 2015-03-19 by Fishwaldo

kintoki2015, 2015-03-19:
thank you again for your reply !? the official website still points to google code to code tracker mailling list etc...

Fishwaldo, 2015-03-19:
the mailing list will stay. 
The code/issues/developer wiki has moved to GitHub. 

The website will be updated soon... but the redirects points here, so its not critical.