OpenZWave/open-zwave #484: two optional command class are not displayed but are supported why?

kintoki2015, 2015-03-19:
Hello I have the following problem:
On a doorlock device I receive as node information update message this frame:
0x01, 0x13, 0x00, 0x49, 0x84, 0x04, 0x0d, 0x04, 0x40, 0x03, 0x22, 0x5d, 0x85, 0x80, 0x70, 0x4c, 0x77, 0x72, 0x86, 0x98, 0x32
as I try to understand what is the meaning of each 8bits word (or hexa number) on that frame. I stumble on the fact that from 0x22 to the checksum 0x98 there is two command_class that are implemented (there exists a .cpp and .h for those command class) but there is no disply of them existing so I get a gap betwin what the log shows me and reallity of the message.

I know that those class are supported just for "fun" but I would like instead of skipping them in the logs to have them a categori like Irrelevant optionnal class. I didn t understand yet how in the code those two class are skiped. 

the irrelevant skipped classes are COMMAND_CLASS_VERSION 0x86 and COMMAND_CLASS_MANIFACTURER_SPECIFIC 0x72 so my log should look like:

2015-03-17 13:36:44.621 Info, Node004, FUNC_ID_ZW_REQUEST_NODE_INFO Request successful.
2015-03-17 13:36:44.653 Detail, Node004,   Received: 0x01, 0x13, 0x00, 0x49, 0x84, 0x04, 0x0d, 0x04, 0x40, 0x03, 0x22, 0x5d, 0x85, 0x80, 0x70, 0x4c, 0x77, 0x72, 0x86, 0x98, 0x32
2015-03-17 13:36:44.653 Detail, 
2015-03-17 13:36:44.653 Info, Node004, UPDATE_STATE_NODE_INFO_RECEIVED from node 4
2015-03-17 13:36:44.653 Info, Node004,   Optional command classes for node 4:
2015-03-17 13:36:44.653 Info, Node004,     COMMAND_CLASS_APPLICATION_STATUS
2015-03-17 13:36:44.653 Info, Node004,   CommandClass 0x5d - NOT REQUIRED
2015-03-17 13:36:44.653 Info, Node004,     COMMAND_CLASS_ASSOCIATION
2015-03-17 13:36:44.653 Info, Node004,     COMMAND_CLASS_BATTERY
2015-03-17 13:36:44.653 Info, Node004,     COMMAND_CLASS_CONFIGURATION
2015-03-17 13:36:44.653 Info, Node004,     COMMAND_CLASS_DOOR_LOCK_LOGGING
2015-03-17 13:36:44.653 Info, Node004,     COMMAND_CLASS_NODE_NAMING
2015-03-17 13:36:44.653 Info, Node004,     Optional command classes that are total useless:
2015-03-17 13:36:44.653 Info, Node004,     COMMAND_CLASS_MANUFACTURER_SPECIFIC
2015-03-17 13:36:44.653 Info, Node004,     COMMAND_CLASS_VERSION  
2015-03-17 13:36:44.653 Detail, Node004, AdvanceQueries queryPending=1 queryRetries=0 queryStage=NodeInfo live=1
2015-03-17 13:36:44.653 Detail, Node004,   Expected reply was received
2015-03-17 13:36:44.653 Detail, Node004,   Message transaction complete
2015-03-17 13:36:44.653 Detail,

kintoki2015, 2015-03-19:
As an update on this topic I can say that if I have the NODE INFO to that same doorlock without providing the network key to OZW I get the 2 missing command_class.

015-03-19 11:37:37.117 Info, Node004, FUNC_ID_ZW_REQUEST_NODE_INFO Request successful.
2015-03-19 11:37:37.146 Detail, Node004,   Received: 0x01, 0x0c, 0x00, 0x49, 0x84, 0x04, 0x06, 0x04, 0x40, 0x03, 0x72, 0x86, 0x98, 0x17
2015-03-19 11:37:37.146 Detail, 
2015-03-19 11:37:37.146 Info, Node004, UPDATE_STATE_NODE_INFO_RECEIVED from node 4
2015-03-19 11:37:37.146 Info, Node004,   Optional command classes for node 4:
2015-03-19 11:37:37.146 Info, Node004,     COMMAND_CLASS_MANUFACTURER_SPECIFIC
2015-03-19 11:37:37.146 Info, Node004,     COMMAND_CLASS_VERSION
2015-03-19 11:37:37.146 Info, Node004,   Using Configured Network Key (AddingNode: false KeySet: false)
2015-03-19 11:37:37.146 Warning, Invalid Network Key. Does not contain 16 Bytes - Contains 0
2015-03-19 11:37:37.146 Warning, Raw Key: 
2015-03-19 11:37:37.146 Warning, Parsed Key:
2015-03-19 11:37:37.146 Error, Exception: Driver.cpp:6304 - 103 - Failed to Read Network Key

As a side ask I dont understand the numer that comme at the end of the frame after the checksum.

Fishwaldo, 2015-03-19:
VERSION give us the versions of the CC's implemented on the device. 
MANUFACTURER_SPECIFIC gives us the manufacturer Id and product codes. Both are used during the QueryStage processing. 

Please use the google groups list for general questions, not the issue tracker.

closed 2015-03-19 by Fishwaldo

kintoki2015, 2015-03-19:
The issue is the fact that it version y manufacturer are not shown but
suddently appears when the nwtwork key is not provided.