OpenZWave/open-zwave #485: More Aeon Labs Multisensor Gen5 issues

maartenbreddels, 2015-03-19:
I have a zwave-me usb stick on a raspberry pi 2 running domoticz and openzwave 1.3.95. I have a fibaro FGS221 Double Relay Switch 2x1.5kW running fine.
 - I can change settings, and it's stable.
 - Now I attach a  Aeon Labs Multisensor (Gen5) to power (USB), and it is still fine. 
 - I touch the zwave button on the multisensor device, or I try to change a setting and the Fibaro doesn't work anymore. Futhermore I cannot change any setting of the multisensor. It only sometimes works, just after it is connected, but most of the time it is not responding. I do get temperature, humidity and light levels.

Motion detection isn't working, but that is another issue (already posted). I also tested it under Window with the openzwave that came with domoticz (3.1.974), and it showed the same results (although the Fibaro only stopped responding after I tried to change a setting of the Multisensor, not when I clicked its zwave button, if I remember correctly)

I attached the log and and configurations, I hope this is enough to tell what is going on.

If I can do anything to help solving this issue let me know soon. I have only a limited amount of time, because if I can't get it working I will send it back to the shop, but they only give 2 weeks for that.

Note that I have to multisensors, just one attached.

Log: http://pastebin.com/0awMepKD
Configuration: http://pastebin.com/x4FicjQN

Fishwaldo, 2015-03-20:
All the new Aeon Labs Gen5 gear is broken right now. I have to refactor how the Security CC is being implemented.  

What I promise is that it will work (with or without Security enabled) once that work is finished.

maartenbreddels, 2015-03-20:
Ok, great to hear. Any ideas on why the Fibaro stopped responding? Can it be the Fibaro itself, or OpenZWave? When I disconnect the Multisensors and restart Domoticz it works fine again (never tried to disconnect the multisensors and wait btw).

Webunity, 2015-03-23:
@maartenbreddels i had exactly thesame issues with Domoticz (which is using Openzwave) and Fibaro switches icm with the Aeon Labs Multisensor (old version and Gen5). Must be somewhat related; following this issue!

Fishwaldo, 2015-03-23:
Please track Issue #352 and Issue #467

closed 2015-03-23 by Fishwaldo