OpenZWave/open-zwave #490: Problem with Linear GD00Z-4 garage door opener

bytedreamer, 2015-03-21:
I added the Linear GD00Z-4 to my Z-Wave network. The device is recognized, however it won't properly complete the initialization. As long as it is associated with the network, OpenZWave will not finish the initialization routine.

Here is the section of the log that I believe is causing the issue:

```
2015-03-21 13:32:39.621 Info, Node016, Sending (Query) message (Callback ID=0x00, Expected Reply=0x49) - Request Node Info (Node=16): 0x01, 0x04, 0x00, 0x60, 0x10, 0x8b
2015-03-21 13:32:39.648 Detail, Node016,   Received: 0x01, 0x04, 0x01, 0x60, 0x01, 0x9b
2015-03-21 13:32:39.648 Detail, 
2015-03-21 13:32:39.648 Info, Node016, FUNC_ID_ZW_REQUEST_NODE_INFO Request successful.
2015-03-21 13:32:39.725 Detail, Node016,   Received: 0x01, 0x0b, 0x00, 0x49, 0x84, 0x10, 0x05, 0x04, 0x40, 0x07, 0x72, 0x98, 0x85
2015-03-21 13:32:39.725 Detail, 
2015-03-21 13:32:39.725 Info, Node016, UPDATE_STATE_NODE_INFO_RECEIVED from node 16
2015-03-21 13:32:39.725 Info, Node016,   Optional command classes for node 16:
2015-03-21 13:32:39.725 Info, Node016,     COMMAND_CLASS_MANUFACTURER_SPECIFIC
2015-03-21 13:32:39.725 Info, Node016,   Using Configured Network Key (AddingNode: false KeySet: false)
2015-03-21 13:32:39.725 Info, Node016,     COMMAND_CLASS_SECURITY
2015-03-21 13:32:39.725 Detail, Node016,   Expected2 reply was received
2015-03-21 13:32:39.726 Detail, Node016,   Message transaction complete
2015-03-21 13:32:39.726 Detail, 
2015-03-21 13:32:39.726 Detail, Node016, Removing current message
2015-03-21 13:32:39.726 Detail, Node016, Query Stage Complete (NodeInfo)
2015-03-21 13:32:39.726 Detail, Node016, AdvanceQueries queryPending=0 queryRetries=0 queryStage=SecurityReport live=1
2015-03-21 13:32:39.726 Detail, Node016, QueryStage_SecurityReport
2015-03-21 13:32:39.726 Detail, Node016, Queuing (Security) SecurityCmd_NonceGet (Node=16): 0x01, 0x09, 0x00, 0x13, 0x10, 0x02, 0x98, 0x40, 0x05, 0x10, 0x3a
2015-03-21 13:32:39.726 Detail, 
2015-03-21 13:32:39.726 Info, Node016, Sending (Security) message (Callback ID=0x10, Expected Reply=0x04) - SecurityCmd_NonceGet (Node=16): 0x01, 0x09, 0x00, 0x13, 0x10, 0x02, 0x98, 0x40, 0x05, 0x10, 0x3a
2015-03-21 13:32:39.731 Detail, Node016,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2015-03-21 13:32:39.731 Detail, Node016,   ZW_SEND_DATA delivered to Z-Wave stack
2015-03-21 13:32:39.802 Detail, Node016,   Received: 0x01, 0x05, 0x00, 0x13, 0x10, 0x00, 0xf9
2015-03-21 13:32:39.802 Detail, Node016,   ZW_SEND_DATA Request with callback ID 0x10 received (expected 0x10)
2015-03-21 13:32:39.802 Info, Node016, Request RTT 75 Average Request RTT 70
2015-03-21 13:33:19.728 Detail, 
2015-03-21 13:33:19.728 Info, Node016, Sending (Security) message (Attempt 2, Callback ID=0x11, Expected Reply=0x04) - SecurityCmd_NonceGet (Node=16): 0x01, 0x09, 0x00, 0x13, 0x10, 0x02, 0x98, 0x40, 0x05, 0x11, 0x3b
2015-03-21 13:33:19.733 Detail, Node016,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2015-03-21 13:33:19.733 Detail, Node016,   ZW_SEND_DATA delivered to Z-Wave stack
2015-03-21 13:33:19.767 Detail, Node016,   Received: 0x01, 0x05, 0x00, 0x13, 0x11, 0x00, 0xf8
2015-03-21 13:33:19.767 Detail, Node016,   ZW_SEND_DATA Request with callback ID 0x11 received (expected 0x11)
2015-03-21 13:33:19.768 Info, Node016, Request RTT 39 Average Request RTT 54
2015-03-21 13:33:59.729 Detail, 
2015-03-21 13:33:59.729 Info, Node016, Sending (Security) message (Attempt 3, Callback ID=0x12, Expected Reply=0x04) - SecurityCmd_NonceGet (Node=16): 0x01, 0x09, 0x00, 0x13, 0x10, 0x02, 0x98, 0x40, 0x05, 0x12, 0x38
2015-03-21 13:33:59.734 Detail, Node016,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2015-03-21 13:33:59.734 Detail, Node016,   ZW_SEND_DATA delivered to Z-Wave stack
2015-03-21 13:33:59.769 Detail, Node016,   Received: 0x01, 0x05, 0x00, 0x13, 0x12, 0x00, 0xfb
2015-03-21 13:33:59.769 Detail, Node016,   ZW_SEND_DATA Request with callback ID 0x12 received (expected 0x12)
2015-03-21 13:33:59.769 Info, Node016, Request RTT 39 Average Request RTT 46
2015-03-21 13:34:39.730 Error, Node016, ERROR: Dropping command, expected response not received after 3 attempt(s)
2015-03-21 13:34:39.730 Detail, Node016, Removing current message
```

Fishwaldo, 2015-03-22:
Did you follow https://github.com/OpenZWave/open-zwave/wiki/Adding-Security-Devices-to-OZW

bytedreamer, 2015-03-22:
Yes, I already had configured a Yale Door Lock to the Z-Wave network. I placed the device right next to the controller during the inclusion process. The controller is the Aeon S2 USB stick.

When I get a chance, I'll try the inclusion process again and capture the logs.

Fishwaldo, 2015-03-23:
The Log Snippet indicates it wasn't included Securely.

bytedreamer, 2015-03-23:
K, Thanks.  I will include it back tonight and check the logs.

bytedreamer, 2015-03-24:
Looks like there is an issue with the secure inclusion process. I'll work to get the latest OZW GitHub code to compile on my laptop. Let me know if you see anything interesting in the logs.

```
2015-03-23 20:06:11.538 Detail, contrlr, Queuing (Controller) Add Device
2015-03-23 20:06:11.539 Info, Add Device
2015-03-23 20:06:11.539 Detail, contrlr, Queuing (Command) AddDevice: 0x01, 0x05, 0x00, 0x4a, 0x81, 0x8d, 0xbc
2015-03-23 20:06:11.539 Detail, 
2015-03-23 20:06:11.539 Info, contrlr, Sending (Command) message (Callback ID=0x8d, Expected Reply=0x4a) - AddDevice: 0x01, 0x05, 0x00, 0x4a, 0x81, 0x8d, 0xbc
2015-03-23 20:06:11.543 Detail, contrlr,   Received: 0x01, 0x07, 0x00, 0x4a, 0x8d, 0x01, 0x00, 0x00, 0x3e
2015-03-23 20:06:11.543 Detail, 
2015-03-23 20:06:11.543 Info, contrlr, FUNC_ID_ZW_ADD_NODE_TO_NETWORK:
2015-03-23 20:06:11.543 Info, contrlr, ADD_NODE_STATUS_LEARN_READY
2015-03-23 20:06:11.543 Detail, contrlr,   Expected callbackId was received
2015-03-23 20:06:11.543 Detail, contrlr,   Expected reply was received
2015-03-23 20:06:11.543 Detail, contrlr,   Message transaction complete
2015-03-23 20:06:11.543 Detail, 
2015-03-23 20:06:11.543 Detail, contrlr, Removing current message
2015-03-23 20:06:11.543 Info, WriteNextMsg Controller nothing to do
2015-03-23 20:06:50.363 Detail,   Received: 0x01, 0x07, 0x00, 0x4a, 0x8d, 0x02, 0x00, 0x00, 0x3d
2015-03-23 20:06:50.363 Detail, 
2015-03-23 20:06:50.363 Info, FUNC_ID_ZW_ADD_NODE_TO_NETWORK:
2015-03-23 20:06:50.363 Info, ADD_NODE_STATUS_NODE_FOUND
2015-03-23 20:06:50.497 Detail,   Received: 0x01, 0x0d, 0x00, 0x4a, 0x8d, 0x03, 0x11, 0x06, 0x04, 0x40, 0x07, 0x72, 0x98, 0x5a, 0xd2
2015-03-23 20:06:50.497 Detail, 
2015-03-23 20:06:50.497 Info, FUNC_ID_ZW_ADD_NODE_TO_NETWORK:
2015-03-23 20:06:50.497 Info, ADD_NODE_STATUS_ADDING_SLAVE
2015-03-23 20:06:50.497 Info, Adding node ID 17
2015-03-23 20:06:50.497 Detail, contrlr, Queuing (Command) Add Node Stop: 0x01, 0x04, 0x00, 0x4a, 0x05, 0xb4
2015-03-23 20:06:51.497 Detail, 
2015-03-23 20:06:51.497 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x4a) - Add Node Stop: 0x01, 0x04, 0x00, 0x4a, 0x05, 0xb4
2015-03-23 20:06:51.551 Detail, contrlr,   Received: 0x01, 0x07, 0x00, 0x4a, 0x8d, 0x06, 0x11, 0x00, 0x28
2015-03-23 20:06:51.551 Detail, 
2015-03-23 20:06:51.551 Info, contrlr, FUNC_ID_ZW_ADD_NODE_TO_NETWORK:
2015-03-23 20:06:51.551 Info, contrlr, ADD_NODE_STATUS_DONE
2015-03-23 20:06:51.551 Detail, Node017, AdvanceQueries queryPending=0 queryRetries=0 queryStage=None live=1
2015-03-23 20:06:51.551 Detail, Node017, QueryStage_ProtocolInfo
2015-03-23 20:06:51.551 Detail, Node017, Queuing (Query) Get Node Protocol Info (Node=17): 0x01, 0x04, 0x00, 0x41, 0x11, 0xab
2015-03-23 20:06:51.551 Detail, Node017, Queuing (Query) Query Stage Complete (ProtocolInfo)
2015-03-23 20:06:51.551 Info, Initilizing Node. New Node: true (true)
2015-03-23 20:06:51.551 Detail, contrlr,   Expected reply was received
2015-03-23 20:06:51.551 Detail, contrlr,   Message transaction complete
2015-03-23 20:06:51.551 Detail, 
2015-03-23 20:06:51.551 Detail, contrlr, Removing current message
2015-03-23 20:06:51.552 Detail, 
2015-03-23 20:06:51.552 Info, Node017, Sending (Query) message (Callback ID=0x00, Expected Reply=0x41) - Get Node Protocol Info (Node=17): 0x01, 0x04, 0x00, 0x41, 0x11, 0xab
2015-03-23 20:06:51.556 Detail, Node017,   Received: 0x01, 0x09, 0x01, 0x41, 0xd3, 0x9c, 0x00, 0x04, 0x40, 0x07, 0xba
2015-03-23 20:06:51.556 Detail, 
2015-03-23 20:06:51.556 Info, Node017, Received reply to FUNC_ID_ZW_GET_NODE_PROTOCOL_INFO
2015-03-23 20:06:51.556 Info, Node017,   Protocol Info for Node 17:
2015-03-23 20:06:51.556 Info, Node017,     Listening     = true
2015-03-23 20:06:51.556 Info, Node017,     Beaming       = true
2015-03-23 20:06:51.556 Info, Node017,     Routing       = true
2015-03-23 20:06:51.556 Info, Node017,     Max Baud Rate = 40000
2015-03-23 20:06:51.556 Info, Node017,     Version       = 4
2015-03-23 20:06:51.556 Info, Node017,     Security      = false
2015-03-23 20:06:51.556 Info, Node017,   Basic device class    (0x04) - Routing Slave
2015-03-23 20:06:51.556 Info, Node017,   Generic device Class  (0x40) - Entry Control
2015-03-23 20:06:51.556 Info, Node017,   No specific device class defined
2015-03-23 20:06:51.556 Info, Node017,     COMMAND_CLASS_BASIC is not mapped
2015-03-23 20:06:51.556 Info, Node017,   Mandatory Command Classes for Node 17:
2015-03-23 20:06:51.556 Info, Node017,     COMMAND_CLASS_BASIC
2015-03-23 20:06:51.556 Info, Node017,   Mandatory Command Classes controlled by Node 17:
2015-03-23 20:06:51.556 Info, Node017,     None
2015-03-23 20:06:51.556 Detail, Node017,   Expected reply was received
2015-03-23 20:06:51.556 Detail, Node017,   Message transaction complete
2015-03-23 20:06:51.556 Detail, 
2015-03-23 20:06:51.556 Detail, Node017, Removing current message
2015-03-23 20:06:51.557 Detail, Node017, Query Stage Complete (ProtocolInfo)
2015-03-23 20:06:51.557 Detail, Node017, AdvanceQueries queryPending=0 queryRetries=0 queryStage=Probe live=1
2015-03-23 20:06:51.557 Detail, Node017, QueryStage_Probe
[... 45 lines cut ...]
2015-03-23 20:07:32.144 Detail, Node017, Queuing (Security) SecurityCmd_SchemeGet (Node=17): 0x01, 0x0a, 0x00, 0x13, 0x11, 0x03, 0x98, 0x04, 0x00, 0x25, 0x8f, 0xc2
2015-03-23 20:07:32.144 Detail, 
[... cut ...]
2015-03-23 20:08:52.152 Detail, Node017,   ZW_SEND_DATA delivered to Z-Wave stack
2015-03-23 20:08:52.187 Detail, Node017,   Received: 0x01, 0x05, 0x00, 0x13, 0x91, 0x00, 0x78
2015-03-23 20:08:52.187 Detail, Node017,   ZW_SEND_DATA Request with callback ID 0x91 received (expected 0x91)
2015-03-23 20:08:52.187 Info, Node017, Request RTT 40 Average Request RTT 102
2015-03-23 20:09:32.147 Error, Node017, ERROR: Dropping command, expected response not received after 3 attempt(s)
2015-03-23 20:09:32.147 Detail, Node017, Removing current message
```

openzwave-dev-watcher, 2015-03-24:
Put the log through the Log Analyser on the homepage. It will tell you if your reaching a timeout for Secure Inclusion.

Fishwaldo, 2015-03-24:
Oops. That last comment is from me.

bytedreamer, 2015-03-24:
@Fishwaldo I'll try including the device a few more times. What should I do if inclusion still has timeout issues?

Here are the results of the [Log Analyser](http://www.openzwave.com/log-analyzer/public/8715e59d-235c-4dfb-9f48-01e4dbda9014/)

Fishwaldo, 2015-03-24:
You should wait till OZW is fully started up (allnodesqueried* notification) before trying the inclusion. As noted in the errors tab for node 17, it took 54 seconds to do, which is way over the timeout value for secure inclusion. 

If this device is like most locks, its just a matter of keep trying till it includes within 10 seconds or so... Hopefully the security rewrite should speed up the inclusion on OZW side, but the 10 second timeout is a device constraint, not OZW.

bytedreamer, 2015-03-28:
@Fishwaldo Please reopen this issue. I've tried included the Linear device many times with no luck. To be sure, I have the device a couple of inches of the USB Z-Wave controller. I can debug the code on my laptop to diagnose the issue. Is there anything specific I should look at within the code.

Is there any chance your rewrite of the Security code will fix my issue?

Thanks

Fishwaldo, 2015-03-28:
If its taking more than 10 seconds from the AddNode command to the Secure Inclusion starting, then you need to understand why? 

The rewrite might help,  buts its not ready yet. 

If you cant get it to happen in under 10 seconds, there isnt much i can do. The timeout is imposed on the device not OZW.

bytedreamer, 2015-03-28:
I’ll try the including the hardware with the Razberry daughter board, http://razberry.z-wave.me/ <http://razberry.z-wave.me/>.  I should have it early next week. I’m curious if there will be any issues with the Z-Way library.

Thanks

bytedreamer, 2015-04-01:
@Fishwaldo The z-wave.me ZWay library is able to include the garage door opener device on the first attempt.

satzg, 2015-04-01:
> !DSPAM:551c07ce273361027123211!

Can you capture any logs of the controller traffic?

Thanks,
Greg

ualex73, 2015-04-01:
the logfile should be in /var/log/razberry.log if I remember correctly.

bytedreamer, 2015-04-01:
Yes, I will get the logs later today when I get home.

Fishwaldo, 2015-04-02:
I'll lay money its just related to timeout issues.

bytedreamer, 2015-04-02:
I posted the zwave.me inclusion logs at [here](http://pastebin.com/g1wcvN9e)

Fishwaldo, 2015-04-02:
Yeah. It completes the SecureInclusion in about 2-3 seconds. 

I'm sure you can get it working with the current OZW, it's just a matter of making sure OZW is idle when you start the inclusion process.

Fishwaldo, 2015-04-02:
If zwave.me stores the network key somewhere, and your using the same dongle for OZW and Zwave.me, then you can include it via zwave.me, copy the network key to OZW Options.xml and it should stay associated and encrypted.

SurfingNinjas, 2015-09-14:
@bytedreamer I'm having the same problem, did you ever get this working?

bytedreamer, 2015-09-14:
@SurfingNinjas I was not able to get the Garge Door opener working with the OpenZwave library. I switched to a different library that supported the device.

SurfingNinjas, 2015-09-14:
@Fishwaldo Can we reopen/investigate this? I am having the same issue with a very similar log. After going my own inclusion logs, it appears that open-zwave does not support command class 0x66 (102)(COMMAND_CLASS_BARRIER_OPERATOR) . I would assume that would stop a garage door from ever working with open-zwave, unless I'm missing something?

Fishwaldo, 2015-09-15:
Yep - We don't support that CC. Someone would need to figure it out so we could implement it. :)

What other libraries work with this device?

SurfingNinjas, 2015-09-22:
@bytedreamer What library did you use to get this working?

bytedreamer, 2015-09-28:
@SurfingNinjas The [ZWay Lib from Z-Wave.Me] (http://razberry.z-wave.me/index.php?id=30) supports the Barrier Operator commands. I've been using this library for a few months with no issues with the garage door opener.

gottsman, 2015-10-02:
@bytedreamer Can you tell me where you sourced the razberry card? The couple of US stores listed from the razberry site didn't have any in stock. Perhaps they are switching stock to the newer version.

bytedreamer, 2015-10-11:
@gottsman I purchased mine from Amazon. Make sure to get the 500 series version.

http://www.amazon.com/Z-Wave-Me/b/ref=bl_dp_s_web_11388845011?ie=UTF8&node=11388845011&field-lbr_brands_browse-bin=Z-Wave.Me

Fishwaldo, 2015-11-13:
I just got confirmation that Aeotec are shipping me their new garage door opener, so I can work on this soon.

nechry, 2015-11-13:
Nice! I'm also supose to receive one, so I will enable to test from my side.

SurfingNinjas, 2015-12-30:
Any update on this?

nechry, 2015-12-30:
From my side nothing, I did not receive the GDC

Fishwaldo, 2016-01-02:
I have the GDC, but no change to explore the Barrier CC yet

Fishwaldo, 2016-01-04:
Some details on the new CC:
http://dz.prosyst.com/pdoc/mBS_SH_SDK_8.0/modules/zwave/api/driver/com/prosyst/mbs/services/zwave/commandclasses/CCBarrierOperator.html

nechry, 2016-02-05:
hello Justin I receive the GDC, I just wrote the config file
https://github.com/jeedom/plugin-openzwave/blob/master/ressources/openzwave/config/aeotec/zw062_garage_door_controller.xml
Base on Rev 4 Engineering Spec
Not all parameters work, so is a preliminary version. If you need some test let me know

srirams, 2016-05-12:
I've created a patch to support BarrierOperator command class based on: https://github.com/dbadia/openhab/commit/eb5a203303d5029ed97941de5abacab4af676608. I just basically copied one of the other classes with some modifications, so it needs to be verified and cleaned up by someone more expertise....

It works on my Iris/Linear GDC. I am able to open/close the door as well as receive status updates from the tilt sensor. The tilt sensor is also supposed to send out battery levels (or at least updates when the battery is low) which I don't know how. Maybe the "Burglar" field in the Received Alarm report? The GDC also sets the Access Control of the Received Alarm report to 70 when you request an action (eg, open or close) and there is no change from the tilt sensor. Again, I don't know what this means. Possible related to how the GDC will only make two attempts to perform an action before you have to do it manually.

[0001-add-BarrierOperator.patch.txt](https://github.com/OpenZWave/open-zwave/files/261578/0001-add-BarrierOperator.patch.txt)

Fishwaldo, 2016-05-13:
Thanks @srirams I've added it to the dev branch, and will add the Signals Support soon as well.

I might also change the ValueID's we export from this CC so we can represent the Opening/Closing/Stopped Events as well.

nechry, 2016-05-13:
I try yesterday the @srirams patch. is ok for open and close but don't report events Opening/Closing/Stopped.
I do test on Aeon-labs Garage door Controller

Fishwaldo, 2016-05-16:
@nechry: yes - those are not supported in @srirams patch, hence why I said I might change the ValueID exported so we can report these. I'll think about the best approach soon. (as a List - or as a new ValueID etc)

openzwave-dev-watcher, 2016-06-06:
How can I help fix this?Also, I would like to get the patch to try it out. 
Thanks.

kb3dow, 2016-06-25:
@openzwave-dev-watcher checkout the dev branch by doing (in your open-zwave directory) - 
git checkout remotes/origin/Dev
make -j 4
Then restart your domoticz (from /etc/init.d/domoticz - assuming you did it that way).
I just got the garage opener 2 days ago and will try out this fix this weekend.

Fishwaldo, 2016-06-25:
Dev branch is only good for playing around at the moment. Lots of other changes in there and it hasn't had a lot of testing.

dtraub, 2016-06-28:
@kb3dow any luck with this opener?  I am considering testing out dev to get it working as well just curious what your inputs are?

dtraub, 2016-06-28:
I have the GD00Z-4

kb3dow, 2016-06-29:
@dtraub  I couldn't get it to work. I use Domoticz with an Aeon Z series stick 2 using Debian on an intel NUC box. I was unable to have it learn the device via the browser (both as a switch and as a security device). When I used the Z stick by itself to pair it, it behaved as if it paired (by they way it beeped - a short beep followed by a long beep) - but the opener wouldn't show up in Domoticz. This was with the dev branch. I restarted the domoticz service after recompiling openzwave. I wouldn't think it needs to be recompiled as the lib for openzwave would be referenced at run time.

legrego, 2016-07-05:
@kb3dow I also tried @srirams 's patched version on the dev branch with the GD00Z-4, with no luck. I was able to pair the device with my Aeon Z series Gen 5 stick via the ozwcp, but I could not get or set the device status at that point. It would give initial values for state, alarm, etc, but would not change them, or allow me to change them.

The OZW log indicated this error everytime I changed the orientation of the tilt sensor:
`ApplicationCommandHandler - Unhandled Command Class 0x2b`

legrego, 2016-07-12:
Ignore my last comment, I had a stale build.

I was able to get the GD00Z-4 working using @srirams patch against master, with just a small configuration change (see c808b74).

I also added a corresponding change to OZWCP at https://github.com/legrego/open-zwave-control-panel/commit/0925251a566a36543e0c6b5cb0b2c1498bddfa68 to support the new command class.

I'm not sure of the best way to integrate this (if we even need to), since the patch is already applied on the dev branch.

JoshAaronJones, 2016-07-21:
I am also trying to get the Linear GD00Z-4 to work.  I haven't actually tried it yet, because I'm almost certain it's not being included securely.  Log is similar to @bytedreamer -- Today I applied the patches mentioned, both in ozw and ozwcp.  The log seems to indicate that the Barrier stuff is successful.  However, I don't think secure inclusion is happening.  I have the network key set, and recognize it in the log.  Device is nearby, and I'm pressing the include button about 1 second after clicking the "Add Device (Secure) -> Go" button.  I'm going to post the log, and if anyone sees anything, please let me know what you think.

`2016-07-21 01:05:10.841 Detail, Queuing (Controller) Add Device
2016-07-21 01:05:10.842 Info, Add Device
post: key=fun data= size=0
2016-07-21 01:05:10.842 Detail, contrlr, Queuing (Command) ControllerCommand_AddDevice: 0x01, 0x05, 0x00, 0x4a, 0xc1, 0x1a, 0x6b
2016-07-21 01:05:10.842 Detail, Notification: ControllerComand - Starting
2016-07-21 01:05:10.842 Info, Notification: ControllerCommand home d5663088 Event 1 Error 0
2016-07-21 01:05:10.842 Detail, 
2016-07-21 01:05:10.842 Info, contrlr, Sending (Command) message (Callback ID=0x1a, Expected Reply=0x4a) - ControllerCommand_AddDevice: 0x01, 0x05, 0x00, 0x4a, 0xc1, 0x1a, 0x6b
2016-07-21 01:05:10.844 Detail, contrlr,   Received: 0x01, 0x07, 0x00, 0x4a, 0x1a, 0x01, 0x00, 0x00, 0xa9
2016-07-21 01:05:10.844 Detail, 
2016-07-21 01:05:10.844 Info, contrlr, FUNC_ID_ZW_ADD_NODE_TO_NETWORK:
2016-07-21 01:05:10.845 Info, contrlr, ADD_NODE_STATUS_LEARN_READY
2016-07-21 01:05:10.845 Detail, Node001,   Expected callbackId was received
2016-07-21 01:05:10.845 Detail, Node001,   Expected reply was received
2016-07-21 01:05:10.845 Detail, Node001,   Message transaction complete
2016-07-21 01:05:10.845 Detail, 
2016-07-21 01:05:10.845 Detail, contrlr, Removing current message
2016-07-21 01:05:10.845 Detail, Notification: ControllerCommand - Waiting
2016-07-21 01:05:10.845 Info, Notification: ControllerCommand home d5663088 Event 4 Error 0
2016-07-21 01:05:23.570 Detail,   Received: 0x01, 0x07, 0x00, 0x4a, 0x1a, 0x02, 0x00, 0x00, 0xaa
2016-07-21 01:05:23.570 Detail, 
2016-07-21 01:05:23.570 Info, FUNC_ID_ZW_ADD_NODE_TO_NETWORK:
2016-07-21 01:05:23.570 Info, ADD_NODE_STATUS_NODE_FOUND
2016-07-21 01:05:23.570 Detail, Notification: ControllerCommand - InProgress
2016-07-21 01:05:23.570 Info, Notification: ControllerCommand home d5663088 Event 6 Error 0
2016-07-21 01:05:23.781 Detail,   Received: 0x01, 0x0f, 0x00, 0x4a, 0x1a, 0x03, 0x09, 0x08, 0x04, 0x40, 0x07, 0x5e, 0x5a, 0x72, 0x73, 0x98, 0x7c
2016-07-21 01:05:23.781 Detail, 
2016-07-21 01:05:23.781 Info, FUNC_ID_ZW_ADD_NODE_TO_NETWORK:
2016-07-21 01:05:23.781 Info, ADD_NODE_STATUS_ADDING_SLAVE
2016-07-21 01:05:23.781 Info, Adding node ID 9 - Secure
2016-07-21 01:05:23.781 Detail, Notification: ControllerCommand - InProgress
2016-07-21 01:05:23.781 Info, Notification: ControllerCommand home d5663088 Event 6 Error 0
2016-07-21 01:05:24.977 Detail,   Received: 0x01, 0x07, 0x00, 0x4a, 0x1a, 0x05, 0x09, 0x00, 0xa4
2016-07-21 01:05:24.977 Detail, 
2016-07-21 01:05:24.977 Info, FUNC_ID_ZW_ADD_NODE_TO_NETWORK:
2016-07-21 01:05:24.977 Info, ADD_NODE_STATUS_PROTOCOL_DONE
2016-07-21 01:05:24.977 Detail, contrlr, Queuing (Command) Add Node Stop: 0x01, 0x04, 0x00, 0x4a, 0x05, 0xb4
2016-07-21 01:05:24.977 Detail, Notification: ControllerCommand - InProgress
2016-07-21 01:05:24.977 Info, Notification: ControllerCommand home d5663088 Event 6 Error 0
2016-07-21 01:05:24.977 Detail, 
2016-07-21 01:05:24.977 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x4a) - Add Node Stop: 0x01, 0x04, 0x00, 0x4a, 0x05, 0xb4
2016-07-21 01:05:25.028 Detail, contrlr,   Received: 0x01, 0x07, 0x00, 0x4a, 0x1a, 0x06, 0x09, 0x00, 0xa7
2016-07-21 01:05:25.028 Detail, 
2016-07-21 01:05:25.028 Info, contrlr, FUNC_ID_ZW_ADD_NODE_TO_NETWORK:
2016-07-21 01:05:25.028 Info, contrlr, ADD_NODE_STATUS_DONE
2016-07-21 01:05:25.028 Info, Node009,   Basic device class    (0x04) - Routing Slave
2016-07-21 01:05:25.028 Info, Node009,   Generic device Class  (0x40) - Entry Control
2016-07-21 01:05:25.028 Info, Node009,   Specific device class (0x07) - Secure Barrier AddOn
2016-07-21 01:05:25.028 Info, Node009,     COMMAND_CLASS_BASIC is not mapped
2016-07-21 01:05:25.028 Info, Node009,   Mandatory Command Classes for Node 9:
2016-07-21 01:05:25.028 Info, Node009,     COMMAND_CLASS_BASIC
2016-07-21 01:05:25.028 Info, Node009,   Mandatory Command Classes controlled by Node 9:
2016-07-21 01:05:25.028 Info, Node009,     None
2016-07-21 01:05:25.028 Detail, Node009, Queuing (Security) SecurityCmd_SchemeGet (Node=9): 0x01, 0x0a, 0x00, 0x13, 0x09, 0x03, 0x98, 0x04, 0x00, 0x25, 0x1b, 0x4e
2016-07-21 01:05:25.028 Info, Node009,   Optional command classes for node 9:
2016-07-21 01:05:25.028 Info, Node009,     COMMAND_CLASS_ZWAVE_PLUS_INFO
2016-07-21 01:05:25.028 Info, Node009,     COMMAND_CLASS_DEVICE_RESET_LOCALLY
2016-07-21 01:05:25.028 Info, Node009,     COMMAND_CLASS_MANUFACTURER_SPECIFIC
2016-07-21 01:05:25.028 Info, Node009,     COMMAND_CLASS_POWERLEVEL
2016-07-21 01:05:25.028 Info, Node009,     COMMAND_CLASS_SECURITY (Existing)
2016-07-21 01:05:25.028 Info, Node009, Initilizing Node. New Node: true (true)
2016-07-21 01:05:25.028 Detail, Node006,   Expected reply was received
2016-07-21 01:05:25.028 Detail, Node006,   Message transaction complete
2016-07-21 01:05:25.028 Detail, 
[... cut ...]
2016-07-21 01:14:27.372 Detail, Node009, AdvanceQueries queryPending=0 queryRetries=0 queryStage=Configuration live=1
2016-07-21 01:14:27.372 Detail, Node009, QueryStage_Configuration
2016-07-21 01:14:27.372 Detail, Node009, QueryStage_Complete
2016-07-21 01:14:27.372 Warning, CheckCompletedNodeQueries m_allNodesQueried=0 m_awakeNodesQueried=1
2016-07-21 01:14:27.372 Warning, CheckCompletedNodeQueries all=0, deadFound=0 sleepingOnly=1
2016-07-21 01:14:27.372 Detail, Node009, Notification: NodeQueriesComplete
2016-07-21 01:14:27.372 Info, Notification: Node 9 Queries Complete`

Fishwaldo, 2016-07-21:
Your Controller is dropping one of the Security Messages, so Secure Inclusion doesn't happen. I'd retry a few times... sometimes RF issues etc can cause this.

tobiebooth, 2016-08-02:
I was able to get the GD00Z-4 working with the commits in @legrego's comment. We've also added support for it to Home Assistant (https://github.com/home-assistant/home-assistant/pull/2712) which works great.

Is there any ETA on when this could make it to master so that we have a timeline of availability in Home Assistant?

openzwave-dev-watcher, 2016-08-28:
> to Home Assistant (home-assistant/home-assistant#2712

I just purchased a GD00Z-4 and I'm having similar issues. Is there a branch 
that makes this work, I couldn't follow links for legrego's commits. Is 
there a way I can help?

openzwave-dev-watcher, 2016-08-28:
My recent log

http://openzwave.com/log-analyzer/public/2df4b736-30e2-4411-a71f-238f3a8c02eb/

2016-08-28 14:31:03.817 Always, OpenZwave Version 1.4.0 Starting Up
2016-08-28 14:31:03.819 Info, Setting Up Provided Network Key for Secure Communications
2016-08-28 14:31:03.819 Info, mgr,     Added driver for controller /dev/ttyACM0
2016-08-28 14:31:03.819 Info,   Opening controller /dev/ttyACM0
2016-08-28 14:31:03.819 Info, Trying to open serial port /dev/ttyACM0 (attempt 1)
2016-08-28 14:31:03.820 Info, Serial port /dev/ttyACM0 opened (attempt 1)
2016-08-28 14:31:03.821 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_VERSION: 0x01, 0x03, 0x00, 0x15, 0xe9
2016-08-28 14:31:03.821 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_MEMORY_GET_ID: 0x01, 0x03, 0x00, 0x20, 0xdc
2016-08-28 14:31:03.821 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_CONTROLLER_CAPABILITIES: 0x01, 0x03, 0x00, 0x05, 0xf9
2016-08-28 14:31:03.821 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_GET_CAPABILITIES: 0x01, 0x03, 0x00, 0x07, 0xfb
2016-08-28 14:31:03.821 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_SUC_NODE_ID: 0x01, 0x03, 0x00, 0x56, 0xaa
2016-08-28 14:31:03.821 Detail, 
2016-08-28 14:31:03.821 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x15) - FUNC_ID_ZW_GET_VERSION: 0x01, 0x03, 0x00, 0x15, 0xe9
2016-08-28 14:31:03.823 Detail, contrlr,   Received: 0x01, 0x10, 0x01, 0x15, 0x5a, 0x2d, 0x57, 0x61, 0x76, 0x65, 0x20, 0x33, 0x2e, 0x39, 0x35, 0x00, 0x01, 0x99
2016-08-28 14:31:03.823 Detail, 
2016-08-28 14:31:03.823 Info, contrlr, Received reply to FUNC_ID_ZW_GET_VERSION:
2016-08-28 14:31:03.823 Info, contrlr,     Static Controller library, version Z-Wave 3.95
2016-08-28 14:31:03.823 Detail, Node045,   Expected reply was received
2016-08-28 14:31:03.823 Detail, Node045,   Message transaction complete
2016-08-28 14:31:03.823 Detail, 
2016-08-28 14:31:03.823 Detail, contrlr, Removing current message
2016-08-28 14:31:03.823 Detail, 
2016-08-28 14:31:03.824 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x20) - FUNC_ID_ZW_MEMORY_GET_ID: 0x01, 0x03, 0x00, 0x20, 0xdc
2016-08-28 14:31:03.826 Detail, contrlr,   Received: 0x01, 0x08, 0x01, 0x20, 0xfc, 0xd2, 0x5b, 0x68, 0x01, 0xca
2016-08-28 14:31:03.826 Detail, 
2016-08-28 14:31:03.826 Info, contrlr, Received reply to FUNC_ID_ZW_MEMORY_GET_ID. Home ID = 0xfcd25b68.  Our node ID = 1
2016-08-28 14:31:03.826 Detail, Node210,   Expected reply was received
2016-08-28 14:31:03.826 Detail, Node210,   Message transaction complete
2016-08-28 14:31:03.826 Detail, 
2016-08-28 14:31:03.826 Detail, contrlr, Removing current message
2016-08-28 14:31:03.826 Detail, 
2016-08-28 14:31:03.826 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x05) - FUNC_ID_ZW_GET_CONTROLLER_CAPABILITIES: 0x01, 0x03, 0x00, 0x05, 0xf9
2016-08-28 14:31:03.829 Detail, contrlr,   Received: 0x01, 0x04, 0x01, 0x05, 0x08, 0xf7
2016-08-28 14:31:03.829 Detail, 
2016-08-28 14:31:03.829 Info, contrlr, Received reply to FUNC_ID_ZW_GET_CONTROLLER_CAPABILITIES:
2016-08-28 14:31:03.829 Info, contrlr,     There is no SUC ID Server (SIS) in this network.
2016-08-28 14:31:03.829 Info, contrlr,     The PC controller is a primary controller.
2016-08-28 14:31:03.829 Detail, Node247,   Expected reply was received
2016-08-28 14:31:03.829 Detail, Node247,   Message transaction complete
2016-08-28 14:31:03.829 Detail, 
2016-08-28 14:31:03.829 Detail, contrlr, Removing current message
2016-08-28 14:31:03.829 Detail, 
2016-08-28 14:31:03.829 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x07) - FUNC_ID_SERIAL_API_GET_CAPABILITIES: 0x01, 0x03, 0x00, 0x07, 0xfb
2016-08-28 14:31:03.833 Detail, contrlr,   Received: 0x01, 0x2b, 0x01, 0x07, 0x01, 0x00, 0x00, 0x86, 0x01, 0x01, 0x00, 0x5a, 0xfe, 0x81, 0xff, 0x88, 0x4f, 0x1f, 0x00, 0x00, 0xfb, 0x9f, 0x7d, 0xa0, 0x67, 0x00, 0x00, 0x80, 0x00, 0x80, 0x86, 0x00, 0x00, 0x00, 0xe8, 0x73, 0x00, 0x00, 0x0e, 0x00, 0x00, 0x60, 0x00, 0x00, 0xfa
2016-08-28 14:31:03.833 Detail, 
2016-08-28 14:31:03.833 Info, contrlr,  Received reply to FUNC_ID_SERIAL_API_GET_CAPABILITIES
2016-08-28 14:31:03.833 Info, contrlr,     Serial API Version:   1.0
2016-08-28 14:31:03.833 Info, contrlr,     Manufacturer ID:      0x0086
2016-08-28 14:31:03.833 Info, contrlr,     Product Type:         0x0101
2016-08-28 14:31:03.833 Info, contrlr,     Product ID:           0x005a
2016-08-28 14:31:03.833 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_RANDOM: 0x01, 0x04, 0x00, 0x1c, 0x20, 0xc7
2016-08-28 14:31:03.834 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_GET_INIT_DATA: 0x01, 0x03, 0x00, 0x02, 0xfe
2016-08-28 14:31:03.834 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_SET_TIMEOUTS: 0x01, 0x05, 0x00, 0x06, 0x64, 0x0f, 0x97
2016-08-28 14:31:03.834 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_APPL_NODE_INFORMATION: 0x01, 0x07, 0x00, 0x03, 0x01, 0x02, 0x01, 0x00, 0xf9
2016-08-28 14:31:03.834 Detail,   Expected reply was received
2016-08-28 14:31:03.834 Detail,   Message transaction complete
2016-08-28 14:31:03.834 Detail, 
2016-08-28 14:31:03.834 Detail, contrlr, Removing current message
2016-08-28 14:31:03.834 Detail, 
2016-08-28 14:31:03.834 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x56) - FUNC_ID_ZW_GET_SUC_NODE_ID: 0x01, 0x03, 0x00, 0x56, 0xaa
2016-08-28 14:31:03.836 Detail, contrlr,   Received: 0x01, 0x04, 0x01, 0x56, 0x00, 0xac
[... cut ...]
2016-08-28 14:31:26.343 Detail,   Expected callbackId was received
2016-08-28 14:31:26.359 Detail, Node003,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x03, 0x03, 0x27, 0x03, 0xff, 0x29
2016-08-28 14:31:26.359 Detail, 
2016-08-28 14:31:26.359 Info, Node003, Response RTT 48 Average Response RTT 48
2016-08-28 14:31:26.359 Detail, Node003, Initial read of value
2016-08-28 14:31:26.359 Info, Node003, Received SwitchAll report from node 3: On and Off Enabled
2016-08-28 14:31:26.359 Detail, Node003,   Expected reply and command class was received
2016-08-28

openzwave-dev-watcher, 2016-08-29:
The patch is in the Dev Tree. But that's going through a bit of work currently so might not be stable. I'm hoping to release very soon though. 

On 29 Aug 2016, at 3:04 AM, Vincent Miceli <vjm@bitdog.io <mailto:vjm@bitdog.io> > wrote:

My recent log

http://openzwave.com/log-analyzer/public/2df4b736-30e2-4411-a71f-238f3a8c02eb/ <http://openzwave.com/log-analyzer/public/2df4b736-30e2-4411-a71f-238f3a8c02eb/> 

On Sunday, August 28, 2016 at 2:55:01 PM UTC-4, Vincent Miceli wrote:

I just purchased a GD00Z-4 and I'm having similar issues. Is there a branch that makes this work, I couldn't follow links for legrego's commits. Is there a way I can help?

On Tuesday, August 2, 2016 at 12:24:01 PM UTC-4, Tobie Booth wrote:

I was able to get the GD00Z-4 working with the commits in @legrego's comment. We've also added support for it to Home Assistant (home-assistant/home-assistant#2712) which works great.

Is there any ETA on when this could make it to master so that we have a timeline of availability in Home Assistant? 

—

 

You received this message because you are subscribed to the Google Groups "OpenZWave" group.
To unsubscribe from this group and stop receiving emails from it, send an email to openzwave+unsubscribe@googlegroups.com <mailto:openzwave+unsubscribe@googlegroups.com> .
For more options, visit https://groups.google.com/d/optout.
<OZW_Log.txt>

blinduvula, 2016-09-20:
Any recent updates on this? I picked one of these up at a local big box store on sale. Support would be amazing.

GFBOX, 2016-10-05:
Hi, I just purchase the GDO Linear GD00Z-4 so any update ? Or just an estimate for the support ?
Thx a lot

colinodell, 2016-10-05:
Any idea when the next release might be?  (A few days? A few weeks?)  I might try patching it myself if the next release is a long ways off.

Thanks for everyone's work on this!

nechry, 2016-10-06:
the Barrier Operator and Central Scene are planned after the release 1.6 of ozw library
https://groups.google.com/forum/#!topic/openzwave/3rmb5a-IDWM

nechry, 2016-12-07:
base on SDK we rewrite the Barrier Operator in beta branch of zwave plugin for jeedom
https://github.com/jeedom/plugin-openzwave/blob/beta/resources/python-openzwave/openzwave/cpp/src/command_classes/BarrierOperator.cpp

We currently only tested with Aeotec Garage Door Controller.

chilicheech, 2017-03-14:
Any updates on this? When do we plan to release and/or add this to OZW?

oblogic7, 2017-04-18:
@nechry Any idea when 1.6 will be released?  Very eager for Barrier Operator to be implemented.

[45 more replies]

closed 2019-05-13 by Fishwaldo