OpenZWave/open-zwave #493: Z-Stick does not go into learning mode

imv, 2015-03-25:
Intermittently, Aeon Labs Z-Stick S2 doesn't go into learning mode. Instead it just reports success.
The relevant part of the log:

```
2015-03-25 12:17:58.843 Detail, contrlr, Queuing (Controller) Add Device
2015-03-25 12:17:58.843 Info, Add Device
2015-03-25 12:17:58.844 Detail, contrlr, Queuing (Command) AddDevice: 0x01, 0x05, 0x00, 0x4a, 0x81, 0x88, 0xb9
2015-03-25 12:17:58.854 Detail, 
2015-03-25 12:17:58.854 Info, contrlr, Sending (Command) message (Callback ID=0x88, Expected Reply=0x4a) - AddDevice: 0x01, 0x05, 0x00, 0x4a, 0x81, 0x88, 0xb9
2015-03-25 12:17:58.865 Detail, contrlr,   Received: 0x01, 0x07, 0x00, 0x4a, 0x88, 0x07, 0x00, 0x00, 0x3d
2015-03-25 12:17:58.865 Detail, 
2015-03-25 12:17:58.866 Info, contrlr, FUNC_ID_ZW_ADD_NODE_TO_NETWORK:
2015-03-25 12:17:58.866 Info, contrlr, ADD_NODE_STATUS_FAILED
2015-03-25 12:17:58.866 Detail, contrlr, Removing current message
2015-03-25 12:17:58.866 Detail, contrlr, Queuing (Command) Add Node Stop: 0x01, 0x04, 0x00, 0x4a, 0x05, 0xb4
2015-03-25 12:17:58.866 Detail, 
2015-03-25 12:17:58.867 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x4a) - Add Node Stop: 0x01, 0x04, 0x00, 0x4a, 0x05, 0xb4
2015-03-25 12:17:58.920 Detail, contrlr,   Received: 0x01, 0x07, 0x00, 0x4a, 0x88, 0x06, 0x28, 0x00, 0x14
2015-03-25 12:17:58.920 Detail, 
2015-03-25 12:17:58.920 Info, contrlr, FUNC_ID_ZW_ADD_NODE_TO_NETWORK:
2015-03-25 12:17:58.920 Info, contrlr, ADD_NODE_STATUS_DONE
2015-03-25 12:17:58.921 Detail, contrlr,   Expected reply was received
2015-03-25 12:17:58.921 Detail, contrlr,   Message transaction complete
2015-03-25 12:17:58.921 Detail, 
2015-03-25 12:17:58.921 Detail, contrlr, Removing current message
```

Full log:
http://www.openzwave.com/log-analyzer/public/8f9f6194-1c9c-4893-ae6d-5d385051581a/

Fishwaldo, 2015-03-26:
This is completely outside the control of OZW. The stick replied:
2015-03-25 12:17:58.866 Info, contrlr, FUNC_ID_ZW_ADD_NODE_TO_NETWORK:
2015-03-25 12:17:58.866 Info, contrlr, ADD_NODE_STATUS_FAILED

In this case, it was probably doing something at that point in time. You just have to retry it.

closed 2015-03-26 by Fishwaldo

imv, 2015-03-26:
Yes, but then it replied
```
2015-03-25 12:17:58.920 Info, contrlr, FUNC_ID_ZW_ADD_NODE_TO_NETWORK:
2015-03-25 12:17:58.920 Info, contrlr, ADD_NODE_STATUS_DONE
```
And called the pfnControllerCallback_t callback with ControllerState_Completed. It doesn't call it with ControllerState_Failed.

Fishwaldo, 2015-03-26:
AddNodeStatusDone is returned as a result of us sending node stop command and ControllerStateFailed is called if it would fail in entering/processing a ControllerMode, not a command. 
(When the dongle is in any of the various controller modes, you can't so normal communications, hence why we have error handing for these states and not for individual commands