OpenZWave/open-zwave #497: HomeId overwritten in scene configuration file

JohanCosemans, 2015-04-01:
Hi OZW enthusiasts,

I am setting up a new ZWave network and have an issue with defining scenes. After defining a scene, this is the XML file that is written:

```xml
<?xml version="1.0" encoding="utf-8" ?>
<Scenes xmlns="http://code.google.com/p/open-zwave/" version="1">
	<Scene id="1" label="Test">
		<Value homeId="0xc1162840" nodeId="5" genre="user" commandClassId="38" instance="1" index="0" type="byte">99</Value>
	</Scene>
</Scenes>
```

After I restart OZW and create a new scene, the above XML is changed:
```xml
	<Scene id="4" label="Test##images/User/scenes/light_on.png">
		<Value homeId="0x7fffffff" nodeId="5" genre="user" commandClassId="38" instance="1" index="0" type="byte">99</Value>
		<Value homeId="0xc1162840" nodeId="6" genre="user" commandClassId="37" instance="2" index="0" type="bool">True</Value>
	</Scene>
```

--> The homeId is set to the max int value 2147483647.

When I restart again, the home id of the new value will be changed etc. This makes my scenes unusable. I have the impression that it is going wrong when the controller is reading the scenes from the XML files.

I'm not a C expert, but I noticed that the OZW file is also storing the homeId, and reads it like this from zwcf_<homeid>.xml:
(driver.cpp, r646)
uint32 homeId = (uint32)**strtoul**( homeIdStr, &p, 0 );

The file Scene.cpp, r231 reads it as follows:
uint32 homeId = (uint32)**strtol**( str, &p, 0 );

Is this second line correct? Can you please help me resolve my issue?

I build OZW from the git repository on 28/3/15.

Thanks!

Fishwaldo, 2015-04-02:
Hrm. Thanks for the report. I will look into it shortly.

Fishwaldo, 2017-05-30:
I'm depreciating the Scene Support so I wont be working on this.

closed 2017-05-30 by Fishwaldo