OpenZWave/open-zwave #499: double Power values for Fibaro Wall Plug and AEON DSC 08101

slawekgh, 2015-04-02:
For FibaroWallPlug and AEON DSC08101 dimmer I have double Power metrics - first one is ok , 2nd is at a OZW startup smth like zero or a strange number 

After couple of minutes both power metrics show almost the same - so I think it is the same hardware gauge on Fibaro Wall Plug but it is being sent twice, so python-ozw thinks they are two separate metrics and show them as two independent power usage?

First time I saw it I thought that there is something wrong with my fibaro wall plug and it started send double metrics, but then I connnect AEON DSC which has the same situation - double power gauges

any ideas? I have no other power metrics zwave device to test.

sl

Fishwaldo, 2015-04-02:
Please post your question to the google groups list with a log file. 

https://groups.google.com/forum/?nomobile=true#!forum/openzwave

closed 2015-04-02 by Fishwaldo

openzwave-dev-watcher, 2015-04-02:
my LOG file : 

http://openzwave.com/OZWLog.log?wtd=u2ale1CYxC2Pphre&request=resource&resource=onkl8gs&rand=9

W dniu czwartek, 2 kwietnia 2015 16:53:22 UTC+2 użytkownik slawekgh napisał: