OpenZWave/open-zwave #501: ThermostatFanMode broken in newish builds...

madarwin, 2015-04-04:
Created from 
https://groups.google.com/forum/#!topic/openzwave/Rz4tAsjbxhw

I think there might be an issue with some changes to ThermostatFanMode.cpp from this commit:
https://github.com/OpenZWave/open-zwave/commit/e14ce4435be634fe7f106571def6069ee64e5d10

For my thermostat (Honeywell) it only has 3 supported Fan modes (on, auto and circulate) and thus m_supportedModes.size() is 3. But one of the supported modes is "circulate" which has a value of 6. So the code at line 238 won't be true

        if( mode < m_supportedModes.size() )
               
The old way worked. So maybe this new code only works with some thermostats? I can provide logs if needed.

I will attached the full log but here is the except that matters: 3 fan modes are discovered. Circulate is mode 6. When attempting to go to mode 6 (circulate), it fails.

2015-04-04 13:16:07.100 Info, Node011, Received supported thermostat fan modes
2015-04-04 13:16:07.100 Info, Node011,     Added fan mode: Auto Low
2015-04-04 13:16:07.100 Info, Node011,     Added fan mode: On Low
2015-04-04 13:16:07.100 Info, Node011,     Added fan mode: Circulate
2015-04-04 13:20:09.636 Info, Node011, Received thermostat fan mode: Auto Low
2015-04-04 13:21:26.682 Info, Node011, Received unknown thermostat fan mode: 6

Fishwaldo, 2015-09-15:
Sorry this took so long. Should be fixed now. Appreciate if you can test.

closed 2015-09-15 by Fishwaldo

madarwin, 2015-09-17:
Took it for a quick test. Seems to work great!

Thanks!
MD

Fishwaldo, 2015-09-17:
thanks for testing!