OpenZWave/open-zwave #502: Adding support for DragonTech In Wall Dimmer

stevejbauer, 2015-04-06:
The devices from DragonTech are not support in the database.  The device that I have is a DragonTech In Wall Dimmer which is a new zwave plus in wall dimmer for the US.  The link to buy this is on Amazon at:  http://www.amazon.com/Dragon-Tech-WD-100-Z-Wave-Multi-Scene/dp/B00T585DUU/ref=sr_1_1?ie=UTF8&qid=1428292643&sr=8-1&keywords=dragon+tech

I have the database files created for this device.  Did you want me to create a pull request for it?  Also, if I scan the manual, where would you like me to put that at?

Fishwaldo, 2015-04-06:
Yes. Please open a pull request. 

I assume you followed these instructions:

https://github.com/OpenZWave/open-zwave/wiki/Adding-Devices

Fishwaldo, 2015-04-06:
and you can email me the manual, i'll upload it to the wiki.

stevejbauer, 2015-04-06:
Here is the manual.  I had to scan it since I can't find this version of it
online.

Fishwaldo, 2015-04-07:
You need to email it direct to me. GitHub filters it out... :)

Fishwaldo, 2015-04-08:
added the doc. Thanks

Fishwaldo, 2016-10-07:
Closing this.

closed 2016-10-07 by Fishwaldo