OpenZWave/open-zwave #503: Setting poll frequency: potential issue in Driver.cpp

turboproc, 2015-04-06:
Trying to set the poll frequency of some battery powered devices, i.e. poll every 2 hours, it occurred to me it was working weird. Setting the poll interval to 30000 (30 seconds) I wasn't able to set the establish the want polling frequency by setting the intensity at 240 (240 time 30 seconds, being 120 mins, 2 hrs).

Setting some debugging logs in the library showed to me that the wait::single call is not executed in most of the cycles. To my understanding this is caused by the continue statement in line 4025 of driver.cpp which causes a jump to the end of the while loop, skipping the wait::single.

I repaired this by omitting the continue and put most of the code below the continue in an else clause up to the block where the wait call is made.
Could someone confirm this is an issue or tell me I repaired something that wasn't broken in the first place

Fishwaldo, 2015-04-06:
Battery Powered Devices will introduce some complexity to the whole polling interval - Reason being is that battery powered devices (not FLiRS) will only wake up every "wakeup interval" (as set by the Wakeup CC) which might not match the poll interval (or be at different times)..

eg, if Wakeup Interval is 7200 (every hour) and the first wakeup interval triggers at say, 1:10, then then next time the device will process commands from OZW will be at 2:10.

Polling will only success on Sleeping devices at 1:10, 2:10, 3:10, 4:10 etc, regardless of what the poll frequency is. (10 minutes, 20 minutes etc). In your example, 2 Hours it would be say at 1:10, 3:10 etc, even if it fires before then.

As for the continue statement, that relating to the poll intensity setting. Setting Poll Intensity to 1 for each value you are polling means each iteration of the Poll Loop will trigger a actual poll. (setting it to 2 means each 2nd iteration of the Poll Loop etc. 

Does that answer your question?

turboproc, 2015-04-06:
Hi Justin,

 

Completely aware of sleeping devices and their interval to wake up. Don’t think that is the issue when keeping the poll interval is long.

 

Not sure whether we touched the real issue. Let’s try it with an example.

 

Taking 1 device that needs to be polled every 60 minutes. Poll interval set at 60000 (1 minute), poll intensity set at 60. If I understand the principle, this should result in polling every 60 minutes.

 

Looking at the code, on the first cycle pollCounter will have a value 60 (origins from poll intensity, line 3795). Hence, pollCounter will be decrease by 1, and the continue statement will be encountered, bringing the process to the end of the while loop, skipping the wait::single. This will repeat until pollCounter reaches the value 1. This time the continue statement will not be encountered, all intelligence is executed and the wait::single will be executed.

 

Hence, assuming the cycles will take zero time by itself, the distance beween the polls become 59*0+1*pollInterval rather that the expected 60*pollInterval.

 

Makes sence?

 

Roland

Fishwaldo, 2015-04-06:
ahhh... I see, yep, it will busy loop decrementing the m_pollCounter if there is only a single Poll Value on the list... 

Can you show me the patch you propose? I'm trying to see how to handle multiple ValueID's each iteration etc.

turboproc, 2015-04-06:
Sure, I patched it in version 1.0.791 but you will get the idea anyway. This part hasn't changed a lot.

First move the calculalation of the pollinterval in case of multiple device to the beginning of the loop:

                        // We only bother getting the lock if the pollList is not empty
                        m_pollMutex->Lock();

                        // Get the next value to be polled
                        PollEntry pe = m_pollList.front();
                        m_pollList.pop_front();
                        ValueID  valueId = pe.m_id;

                        // If the polling interval is for the whole poll list, calculate the time before the next poll,
                        // so that all polls can take place within the user-specified interval.
                        if( !m_bIntervalBetweenPolls )
                        {
                             if( pollInterval < 100 )
                             {
                                     Log::Write( LogLevel_Info, "The pollInterval setting is only %d, which appears to be a legacy setting.  Multiplying by 1000 to  gi$
                                     pollInterval *= 1000;
                             }
                             pollInterval /= m_pollList.size();
                        }

Next, in stead of the continue statement split situation with a new else block:

                        if( pe.m_pollCounter != 1)
                        {
                                pe.m_pollCounter--;
                                m_pollList.push_back( pe );
                                m_pollMutex->Unlock();
                        }
                        else
                        {

                           // reset the poll counter to the full pollIntensity value and push it at the end of the list
                           // release the value object referenced; call GetNode to ensure the node objects are locked during this period
                           (void)GetNode( valueId.GetNodeId() );
                           Value* value = GetValue( valueId );
                           pe.m_pollCounter = value->GetPollIntensity();
                           m_pollList.push_back( pe );
                           value->Release();
                           ReleaseNodes();

keep the else block until here:

                           }
                        } // this one is new

                        // ready for next poll...insert the pollInterval delay
                        int i32 = Wait::Single( _exitEvent, pollInterval );
                        if( i32 == 0 )
 

This should do it.

Fishwaldo, 2015-04-06:
ok... seems ok. I'm working through some Security Issues right now. I'll patch this after a bit of further testing, but I can't see anyting thats obviously wrong right now. 

Cheers

killercode, 2015-05-22:
I don't know much about the openzwave polling mechanics but wouldn't it be easy to poll the battery powered devices when it signals a wake up?

I struggle a little bit with this using the NorthQ PowerReader... In fact I made some code on my openZWave implementation to manually poll this device everytime it wakes up...

Do you think that this should make any sense? a wake-up poll option on configuration?

turboproc, 2015-05-23:
Hi Diogo,

 

To provide a short explanation:

 

-          Battery powered devices behave as normal devices except that they will be sleeping most of the time. Hence, sending a message will in most case result in a lack of response. If OpenZwave does not receive a response from an device which is known to be sleeping, it will queue the message. Once the sleeping device wakes up, OpenZwave will (re)transmit all queued messages to the device.

-          Polling is performed by sending a message to a device requesting a specific value. Polling occurs at regular intervals.

-          When polling potentially sleeping devices, procedure mentioned at the first item simply applies, i.e. the initial request will usually fail, the request becomes queued, when the device wakes up, the message is send again.

-          Consequence of above is that it makes no sense to poll more frequently than the device wakes up (e.g. polling every 15 mins a device that wakes up once an hour).

-          Within my system I’ve a number of sleeping devices polled perfectly in this way.

 

Please let me know whether this helps you further.

Fishwaldo, 2015-05-25:
It actually can be a bit less "troublesome" than what TurboProc mentions:
You can set up Polling for any value on a sleeping device, and regardless of the Poll Interval, when the device is sleeping we will only every Queue up a single query... (ie, the Queue Code does some deduplication). This means if your Wakeup setting was say 1 hour, and your configured Polling for every 15 minutes, we would only every queue up a single refresh (at 15 mins) and subsequent Polling periods (30, 45, and potentially 60) would be dropped. When the device wakes up, it would clear the queue, the Polling would go out, and thus next Poll period, everything is empty, and it starts again.

closed 2015-05-25 by Fishwaldo