OpenZWave/open-zwave #509: Exception: Manager.cpp:1482

michapr, 2015-04-08:
Using Philio device (PSR04) - a programmable button switch I get this errors in OZW_log while changing the level:
(using OpenZwave v. 1.3.118 and Aeon Stick)

2015-04-08 15:14:06.522 Detail, Node027, Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x1b, 0x03, 0x20, 0x01, 0x27, 0xec
2015-04-08 15:14:06.524 Detail,
2015-04-08 15:14:06.526 Info, Node027, Received Basic set from node 27: level=39. Sending event notification.
2015-04-08 15:14:06.527 Warning, Exception: Manager.cpp:1482 - 101 - Invalid ValueID passed to GetValueLabel

Thanks,
Michael

Fishwaldo, 2015-04-08:
This is a application related issue. It means the application is passing a invalid ValueID to the OZW API. You need to work with the application developer to resolve this.

closed 2015-04-08 by Fishwaldo

michapr, 2015-04-15:
I think it is an ZWave Plus issue - can you please recheck it?
Get the same error ponly with new ZWave Plus compatible devices.