OpenZWave/open-zwave #511: NorthQ device not reporting reading

killercode, 2015-04-09:
Hi,

I've been trying to use a northQ device but I don't have energy reading at all... the device wakes up goes back to sleep and doesn't report anything in between?

Is this a misconfiguration on the xml file?

Fishwaldo, 2015-04-10:
No Logs - No Details. Nothing to diagnose this problem..... 
Sorry, we need more info....

Fishwaldo, 2015-04-10:
Check the Associations on the Device and the Configuration values..... but without the log, I can't comment... so its not that irrelevant after all...

killercode, 2015-04-10:
Ok I will send you a log
Also do you think that the zwcfg file will be handy?

killercode, 2015-04-13:
Ok, no problem at all with the lib, I have to poll the value everytime the device wakes up

closed 2015-04-16 by Fishwaldo