OpenZWave/open-zwave #520: Exception: Manager.cpp:1482 (Z-Wave Plus?)

michapr, 2015-04-15:
Do not know how to reopen a closed ticket - so have to open it again as new issue (modified).

Using Philio device (PSR04) - a programmable button switch I get this errors in OZW_log while changing the level:
(using OpenZwave v. 1.3.118 and Aeon Stick)
Same for PST02-1A.

I think it is a Z-Wave Plus issue, because of I get this error only for new Z-Wave Plus devices.

2015-04-08 15:14:06.522 Detail, Node027, Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x1b, 0x03, 0x20, 0x01, 0x27, 0xec
2015-04-08 15:14:06.524 Detail,
2015-04-08 15:14:06.526 Info, Node027, Received Basic set from node 27: level=39. Sending event notification.
2015-04-08 15:14:06.527 Warning, Exception: Manager.cpp:1482 - 101 - Invalid ValueID passed to GetValueLabel

Thanks,
Michael

Fishwaldo, 2015-04-15:
its got nothing to do with Z-Wave plus. You are passing a ValueID to Manager::GetValueLabel() that OZW doesn't know about. Check where you got that ValueID from (I assume you are monitoring ValueDeleted Notifications as well?)

Fishwaldo, 2015-04-15:
This is the code that triggers the exception. As you can see its very simple:
if( Value* value = driver->GetValue( _id ) )
{
label = value->GetLabel();
value->Release();
} else {
OZW_ERROR(OZWException::OZWEXCEPTION_INVALID_VALUEID, "Invalid ValueID passed to GetValueLabel");
}

closed 2015-04-15 by Fishwaldo

michapr, 2015-04-15:
yes, right,... I have found it in code too.
But it is very strange that this error is coming for every Gen5 device only.
And I have really a lot of different devices.... 

That's why thought maybe a general problem.