OpenZWave/open-zwave #523: Add support for TKB Home Door Contact Switch SM103

whiteduck22, 2015-04-19:
Hi there,

TKB Home provide an identical door contact switch to Everspring's SM103 so the only change I needed was to add one line to manufacturer_specific.xml:

 <Manufacturer id="0118" name="Wenzhou TKB Control System">
...
                <Product type="0002" id="0001" name="SM103 Door/Window Sensor" config="everspring/sm103.xml"/>
…

Not sure how you wish to implement this as a copy of sm103.xml under wenzhou or just refer to the everspring one.

Here is the output for the three operating modes:

Tamper:
2015-04-18 23:18:31.332 Detail, Node009,   Received: 0x01, 0x0a, 0x00, 0x04, 0x00, 0x09, 0x04, 0x71, 0x05, 0x01, 0x11, 0x98
2015-04-18 23:18:31.333 Detail, 
2015-04-18 23:18:31.333 Info, Node009, Received Alarm report: type=1, level=17
2015-04-18 23:18:31.333 Detail, Node009, Refreshed Value: old value=1, new value=1, type=uint8
2015-04-18 23:18:31.334 Detail, Node009, Changes to this value are not verified
2015-04-18 23:18:31.334 Detail, Node009, Refreshed Value: old value=17, new value=17, type=uint8
2015-04-18 23:18:31.334 Detail, Node009, Changes to this value are not verified

On (Open)
2015-04-18 23:18:51.969 Detail, Node009,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x09, 0x03, 0x20, 0x01, 0x63, 0xba
2015-04-18 23:18:51.970 Detail, 
2015-04-18 23:18:51.970 Info, Node009, Received Basic set from node 9: level=99. Treating it as a Basic report.
2015-04-18 23:18:51.971 Detail, 
2015-04-18 23:18:51.971 Detail, Node009, Queuing (WakeUp) SensorBinaryCmd_Get (Node=9): 0x01, 0x09, 0x00, 0x13, 0x09, 0x02, 0x30, 0x02, 0x25, 0x92, 0x6b
2015-04-18 23:18:51.971 Detail, Node009, Refreshed Value: old value=false, new value=true, type=bool
2015-04-18 23:18:51.971 Detail, Node009, Changes to this value are not verified

Off (Closed)
2015-04-18 23:19:14.146 Detail, Node009,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x09, 0x03, 0x20, 0x01, 0x00, 0xd9
2015-04-18 23:19:14.146 Detail, 
2015-04-18 23:19:14.146 Info, Node009, Received Basic set from node 9: level=0. Treating it as a Basic report.
2015-04-18 23:19:14.147 Detail, 
2015-04-18 23:19:14.147 Detail, Node009, Queuing (WakeUp) SensorBinaryCmd_Get (Node=9): 0x01, 0x09, 0x00, 0x13, 0x09, 0x02, 0x30, 0x02, 0x25, 0x93, 0x6a
2015-04-18 23:19:14.147 Detail, Node009, Refreshed Value: old value=true, new value=false, type=bool
2015-04-18 23:19:14.147 Detail, Node009, Changes to this value are not verified

Thanks a lot guys

Best regards
Ian

Fishwaldo, 2015-04-21:
I added it as a new config file under the wenzhou in case the everspring one changes in the future. Thanks

closed 2015-04-21 by Fishwaldo

whiteduck22, 2015-04-21:
Fantastic thanks for this!

---- On Tue, 21 Apr 2015 09:30:48 +0100 Justin Hammond &lt;notifications@github.com&gt; wrote ---- 

I added it as a new config file under the wenzhou in case the everspring one changes in the future. Thanks