OpenZWave/open-zwave #530: Compiling OpenZWave gives errors in FreeNAS 9.3 jail

wmn79, 2015-04-22:
When I try to compile in a FreeNAS 9.3 jail I get multiple errors:

First error is:
/usr/home/domoticz/open-zwave/cpp/hidapi/libusb/hid.c:48:19: error: iconv.h: No such file or directory
/usr/home/domoticz/open-zwave/cpp/hidapi/libusb/hid.c:48:19: error: iconv.h: No such file or directory

I solved that by doing: cp /usr/local/include/iconv.h /home/domoticz/open-zwave/cpp/hidapi/libusb/iconv.h

Second time the error is:
/usr/home/domoticz/open-zwave/cpp/hidapi/libusb/hid.c:266: error: static declaration of 'libusb_get_string_descriptor' follows non-static declaration
/usr/include/libusb.h:406: error: previous declaration of 'libusb_get_string_descriptor' was here
cc1: warnings being treated as errors
/usr/home/domoticz/open-zwave/cpp/hidapi/libusb/hid.c: In function 'get_usb_string':
/usr/home/domoticz/open-zwave/cpp/hidapi/libusb/hid.c:376: warning: passing argument 2 of 'libiconv' from incompatible pointer type
/usr/home/domoticz/open-zwave/cpp/build/support.mk:112: recipe for target '/usr/home/domoticz/open-zwave/.lib/hid.o' failed
gmake[1]: *** [/usr/home/domoticz/open-zwave/.lib/hid.o] Error 1
gmake[1]: Leaving directory '/usr/home/domoticz/open-zwave/cpp/build'
Makefile:20: recipe for target 'all' failed
gmake: *** [all] Error 2

I have fixed this with the a patch I made which can be found here: https://dl.dropboxusercontent.com/u/9086479/hid.patch

Now OpenZWave seems to compiles correct again.

The version of the installed packages in the jail are:
apr-1.5.1.1.5.4
boost-all-1.55.0 
boost-docs-1.55.0
boost-jam-1.55.0
boost-libs-1.55.0_4
boost-python-libs-1.55.0
ca_root_nss-3.18
cmake-3.1.3
cmake-modules-3.1.3
curl-7.41.0
cvsps-2.1_1
db5-5.3.28_2
expat-2.1.0_2
gdbm-1.11_2
gettext-0.19.4
gettext-runtime-0.19.4
gettext-tools-0.19.4
git-2.3.5
gmake-4.1_1
icu-53.1
indexinfo-0.2.3
libarchive-3.1.2_2,1
libexecinfo-1.1_3
libffi-3.2.1
libiconv-1.14_8
libxml2-2.9.2_2
lzo2-2.09
nano-2.4.1
p5-Authen-SASL-2.16_1
p5-Digest-HMAC-1.03_1
p5-Error-0.17023
p5-GSSAPI-0.28_1
p5-IO-Socket-IP-0.37
p5-IO-Socket-SSL-2.012
p5-MIME-Base64-3.15
p5-Mozilla-CA-20141217
p5-Net-SMTP-SSL-1.02
p5-Net-SSLeay-1.68
p5-Socket-2.018
perl5-5.18.4_13
pkg-1.5.1
python27-2.7.9_1
serf-1.3.8
sqlite3-3.8.9_1
subversion-1.8.13

wmn79, 2015-04-23:
Don't think that removing the following (starting in row 338 of the hid.c file) is really fixing the issue:

#ifdef __FreeBSD__
	const char *inptr;
#else
	char *inptr;
#endif
	char *outptr; 

Changing: 
#if defined(__FreeBSD__) && __FreeBSD_version < 1000000
to 
#if defined(__FreeBSD__) && __FreeBSD_version < 10

in row 255 of hid.c does seem to fix this error but that was already reported in issue 462.

When I now try to compile domoticz I get a lot off errors regarding the built hid.c:
......
[100%] Building CXX object CMakeFiles/domoticz.dir/json/json_writer.cpp.o
Linking CXX executable domoticz
lua/src/liblua.a(loslib.c.o): In function `os_tmpname':
loslib.c:(.text+0x48): warning: warning: tmpnam() possibly used unsafely; consider using mkstemp()
/home/domoticz/open-zwave-read-only/libopenzwave.a(hid.o): In function `hid_get_feature_report':
hid.c:(.text+0xc2): undefined reference to `libusb_control_transfer'
/home/domoticz/open-zwave-read-only/libopenzwave.a(hid.o): In function `hid_send_feature_report':
hid.c:(.text+0x137): undefined reference to `libusb_control_transfer'
/home/domoticz/open-zwave-read-only/libopenzwave.a(hid.o): In function `hid_write':
hid.c:(.text+0x19e): undefined reference to `libusb_interrupt_transfer'
hid.c:(.text+0x1ea): undefined reference to `libusb_control_transfer'
/home/domoticz/open-zwave-read-only/libopenzwave.a(hid.o): In function `read_thread':
hid.c:(.text+0x22c): undefined reference to `libusb_alloc_transfer'
hid.c:(.text+0x259): undefined reference to `libusb_fill_interrupt_transfer'
hid.c:(.text+0x262): undefined reference to `libusb_submit_transfer'
hid.c:(.text+0x27f): undefined reference to `libusb_handle_events'
hid.c:(.text+0x2a1): undefined reference to `libusb_cancel_transfer'
hid.c:(.text+0x2b1): undefined reference to `libusb_handle_events'
/home/domoticz/open-zwave-read-only/libopenzwave.a(hid.o): In function `hid_exit':
hid.c:(.text+0x2f1): undefined reference to `libusb_exit'
/home/domoticz/open-zwave-read-only/libopenzwave.a(hid.o): In function `hid_init':
hid.c:(.text+0x32f): undefined reference to `libusb_init'
/home/domoticz/open-zwave-read-only/libopenzwave.a(hid.o): In function `get_usb_string':
hid.c:(.text+0x663): undefined reference to `libusb_get_string_descriptor'
hid.c:(.text+0x6b7): undefined reference to `libusb_get_string_descriptor'
hid.c:(.text+0x6e9): undefined reference to `libusb_get_string_descriptor'
/home/domoticz/open-zwave-read-only/libopenzwave.a(hid.o): In function `hid_enumerate':
hid.c:(.text+0x8d0): undefined reference to `libusb_get_device_list'
hid.c:(.text+0x960): undefined reference to `libusb_get_device_descriptor'
hid.c:(.text+0x991): undefined reference to `libusb_get_active_config_descriptor'
hid.c:(.text+0xa8b): undefined reference to `libusb_get_device_address'
hid.c:(.text+0xa95): undefined reference to `libusb_get_bus_number'
hid.c:(.text+0xadb): undefined reference to `libusb_open'
hid.c:(.text+0xb18): undefined reference to `libusb_close'
hid.c:(.text+0xb7e): undefined reference to `libusb_free_config_descriptor'
hid.c:(.text+0xbe2): undefined reference to `libusb_free_device_list'
hid.c:(.text+0xc08): undefined reference to `libusb_get_config_descriptor'
/home/domoticz/open-zwave-read-only/libopenzwave.a(hid.o): In function `hid_open_path':
hid.c:(.text+0xca2): undefined reference to `libusb_get_device_list'
hid.c:(.text+0xd18): undefined reference to `libusb_get_device_descriptor'
hid.c:(.text+0xd27): undefined reference to `libusb_get_active_config_descriptor'
hid.c:(.text+0xdaf): undefined reference to `libusb_get_device_address'
hid.c:(.text+0xdbb): undefined reference to `libusb_get_bus_number'
hid.c:(.text+0xe14): undefined reference to `libusb_open'
hid.c:(.text+0xe28): undefined reference to `libusb_claim_interface'
hid.c:(.text+0xf3c): undefined reference to `libusb_free_config_descriptor'
hid.c:(.text+0xf61): undefined reference to `libusb_close'
hid.c:(.text+0xf7d): undefined reference to `libusb_free_device_list'
/home/domoticz/open-zwave-read-only/libopenzwave.a(hid.o): In function `read_callback':
hid.c:(.text+0x10b4): undefined reference to `libusb_submit_transfer'
/home/domoticz/open-zwave-read-only/libopenzwave.a(hid.o): In function `hid_close':
hid.c:(.text+0x11c0): undefined reference to `libusb_cancel_transfer'
hid.c:(.text+0x11e1): undefined reference to `libusb_free_transfer'
hid.c:(.text+0x11ed): undefined reference to `libusb_release_interface'
hid.c:(.text+0x11f6): undefined reference to `libusb_close'
*** [domoticz] Error code 1
1 error
*** [CMakeFiles/domoticz.dir/all] Error code 2
1 error
*** [all] Error code 2
1 error

Fishwaldo, 2015-04-30:
these errors are because of missing libraries etc on your box. In this case, domoticz have bundled OZW with their app, so you need to work with them on the fixes. 
the Version check for FreeBSD is also correct, but its because you have missing dependancies its failing.

closed 2015-04-30 by Fishwaldo

wmn79, 2015-04-30:
Ok thanks, is there an overview somewhere where I can find which libraries and dependencies am missing? 

Btw the OZW is not bundled with the domoticz app, you need to compile OZW first before compiling domoticz itself. During compiling it checks whether the OZW is there.

Fishwaldo, 2015-04-30:
then the errors you posted above are because Domoticz isn't linking correctly. 
(OZW doesn't use cmake). 
the missing library above is libusb.

wmn79, 2015-04-30:
Ok thank you, when I compiled with 
#if defined(FreeBSD) && FreeBSD_version < 903000

and without the -Werror flag it seems to compile both OpenZWave and domoticz without errors.

Now I need to figure out a way to access my ZWave stick in the jail. Do you maybe have some suggestions how to do that? It tried to make a symbolic link from both ttyu1 as from ugen0.4 into the jail as ttyUSB0 but with the ttyu1 symbolic link I get the following in MinOZW:
Starting MinOZW with OpenZWave Version 1.3.154
2015-04-29 12:10:08.219 Error, Cannot find a path to the configuration files at ../../../config/, Using config/ instead...
2015-04-29 12:10:08.219 Info, Reading config/options.xml for Options
2015-04-29 12:10:08.219 Warning, Failed to Parse options.xml: Failed to open file
2015-04-29 12:10:08.220 Always, OpenZwave Version 1.3.154 Starting Up
2015-04-29 12:10:08.220 Info, mgr,     Added driver for controller /dev/ttyUSB0
2015-04-29 12:10:08.220 Info,   Opening controller /dev/ttyUSB0
2015-04-29 12:10:08.220 Info, Trying to open serial port /dev/ttyUSB0 (attempt 1)

With the symbolic link to ugen0.4 I get the following:
Starting MinOZW with OpenZWave Version 1.3.154
2015-04-29 12:08:17.832 Error, Cannot find a path to the configuration files at ../../../config/, Using config/ instead...
2015-04-29 12:08:17.832 Info, Reading config/options.xml for Options
2015-04-29 12:08:17.832 Warning, Failed to Parse options.xml: Failed to open file
2015-04-29 12:08:17.833 Always, OpenZwave Version 1.3.154 Starting Up
2015-04-29 12:08:17.833 Info, mgr,     Added driver for controller /dev/ttyUSB0
2015-04-29 12:08:17.833 Info,   Opening controller /dev/ttyUSB0
2015-04-29 12:08:17.833 Info, Trying to open serial port /dev/ttyUSB0 (attempt 1)
2015-04-29 12:08:17.833 Error, ERROR: Failed to set serial port parameters
2015-04-29 12:08:17.833 Error, ERROR: Failed to open serial port /dev/ttyUSB0
2015-04-29 12:08:17.833 Warning, WARNING: Failed to init the controller (attempt 0)
2015-04-29 12:08:22.834 Info,   Opening controller /dev/ttyUSB0
2015-04-29 12:08:22.834 Info, Trying to open serial port /dev/ttyUSB0 (attempt 1)
2015-04-29 12:08:22.834 Error, ERROR: Failed to set serial port parameters
2015-04-29 12:08:22.835 Error, ERROR: Failed to open serial port /dev/ttyUSB0
2015-04-29 12:08:22.835 Warning, WARNING: Failed to init the controller (attempt 1)
2015-04-29 12:08:27.835 Info,   Opening controller /dev/ttyUSB0
2015-04-29 12:08:27.835 Info, Trying to open serial port /dev/ttyUSB0 (attempt 1)
2015-04-29 12:08:27.835 Error, ERROR: Failed to set serial port parameters
2015-04-29 12:08:27.835 Error, ERROR: Failed to open serial port /dev/ttyUSB0
2015-04-29 12:08:27.836 Warning, WARNING: Failed to init the controller (attempt 2)

Can you please give a hint on which of the two should be the correct way? So I now which path to continue with. Thanks again.