OpenZWave/open-zwave #533: Additonal type for Remotec ZXT-120

rayslinky, 2015-04-26:
manufacturer_specific.xml lists the Remotec ZXT-120 as type="0101" id="8377". I recently received one that is reporting type="0100" id="8377". I manually added it to my manufacturer_specific.xml to mimic the 0101 and it works as expected.

killercode, 2015-04-30:
My remotec ZXT-120 also has different bitmask configuration, took me a long to understand how to configure...

rayslinky, 2015-04-30:
You're right -- they are a bit different. :/

Fishwaldo, 2015-05-02:
Then we would need a new config file for the updated device in that case.

killercode, 2015-05-22:
Yeah.... but the problem is that it seems that is a fake chinese remotec ZXT-120... Which means that it has the same manufacturer id and device id with different settings.... I don't know how this can be solved...

Fishwaldo, 2015-05-25:
Then I'm not going to update for "Fakes"... :)

closed 2015-05-25 by Fishwaldo

antoniolobefaro, 2015-05-27:
Hi guys, 
I'm new about zwave and remotec but I need to use these in a project thtat is about to end . I'm able to see the remotec in zwave network using DomotiGa (I see the temperature releved) but how can I set the THERMOSTAT_MODE or THERMOSTAT_SETPOINT or THERMOSTAT_FAN_MODE? 
Do you have examples?