OpenZWave/open-zwave #535: NorthQ Power Reader issue

manolo75, 2015-04-29:
Hi,

I bought a few months ago a NorthQ NQ9021 Power Reader device and I run it with domoticz. 

Since, the data in the charts are wrong and when I look in the domoticz logs here is what I see :

subtype = ELEC2 - OWL CM119, CM160
Sequence nbr = 0
ID = 3585
Count = 1
Instant usage = 0 Watt
total usage = 1698734.00 Wh
Signal level = 12
Battery = OK
2015-04-29 21:40:57.647 (Aeon Stick S2) Energy (Compteur électrique)

The instant usage is always equal to 0. I really do't know what to do because all the data is wrong and my device is useless. I can provide charts and logs on demand.

Regards,

Manu

Fishwaldo, 2015-04-30:
We would need to see the OZW log file to diagnose. Please post a copy to the Google Groups List.

manolo75, 2015-04-30:
There is not much in the OZW log file sadly :

pi@raspberrypi ~/dev-domoticz/Config $ tail -f OZW_Log.txt
2015-04-30 07:29:17.237 Always, OpenZwave Version 1.3.154 Starting Up

That's all. I continue to check if I can have OZW telling more.

manolo75, 2015-04-30:
Here is my startup command for domoticz : 

DAEMON_ARGS="-daemon -www 8084 -log /var/log/domoticz.txt -verbose 1 -loglevel 0"

I can't find anything concerning OZW and a eventually debug mode. I only get a huge log message for domoticz, with the message I said in it.

manolo75, 2015-04-30:
Here is a log that just appears but concerns another device : 

2015-04-30 09:20:02.340 Error, Node005, ERROR: Dropping command, expected response not received after 3 attempt(s)

If there is a way to increase logs in OZW, please let me know.

manolo75, 2015-04-30:
Debug activated for my ZWave Controller. Here is the log :

2015-04-30 12:00:07.232 Info, Node014, Sending (Poll) message (Callback ID=0x93, Expected Reply=0x04) - MeterCmd_Get (Node=14): 0x01, 0x0a, 0x00, 0x13, 0x0e, 0x03, 0x32, 0x01, 0x00, 0x25, 0x93, 0x6e
2015-04-30 12:00:07.256 Detail, Node014, Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2015-04-30 12:00:07.257 Detail, Node014, ZW_SEND_DATA delivered to Z-Wave stack
2015-04-30 12:00:07.261 Detail, Node014, Received: 0x01, 0x05, 0x00, 0x13, 0x93, 0x00, 0x7a
2015-04-30 12:00:07.262 Detail, Node014, ZW_SEND_DATA Request with callback ID 0x93 received (expected 0x93)
2015-04-30 12:00:07.262 Info, Node014, Request RTT 29 Average Request RTT 24
2015-04-30 12:00:07.266 Detail, Node014, Received: 0x01, 0x14, 0x00, 0x04, 0x00, 0x0e, 0x0e, 0x32, 0x02, 0x21, 0x64, 0x00, 0x19, 0xec, 0xc8, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xa7
2015-04-30 12:00:07.267 Detail,
2015-04-30 12:00:07.270 Info, Node014, Response RTT 37 Average Response RTT 33

manolo75, 2015-04-30:
Following :
2015-04-30 12:03:07.324 Info, Node014, Response RTT 29 Average Response RTT 31
2015-04-30 12:03:07.324 Detail, Node014, Refreshed Value: old value=false, new value=false, type=bool
2015-04-30 12:03:07.324 Detail, Node014, Changes to this value are not verified
2015-04-30 12:03:07.325 Info, Node014, Received Meter report from node 14: Energy=1699.018kWh
2015-04-30 12:03:07.325 Detail, Node014, Refreshed Value: old value=1699.016, new value=1699.018, type=string
2015-04-30 12:03:07.326 Detail, Node014, Changes to this value are not verified
2015-04-30 12:03:07.326 Detail, Node014, Expected reply and command class was received
2015-04-30 12:03:07.326 Detail, Node014, Message transaction complete
2015-04-30 12:03:07.327 Detail,
2015-04-30 12:03:07.327 Detail, Node014, Removing current message

manolo75, 2015-04-30:
Some more if needed : 

2015-04-30 12:12:07.378 Detail, Node014, Polling: COMMAND_CLASS_METER index = 0 instance = 1 (poll queue has 0 messages)
2015-04-30 12:12:07.379 Detail, Node014, Queuing (Poll) MeterCmd_Get (Node=14): 0x01, 0x0a, 0x00, 0x13, 0x0e, 0x03, 0x32, 0x01, 0x00, 0x25, 0x98, 0x65
2015-04-30 12:12:07.380 Detail,
2015-04-30 12:12:07.380 Info, Node014, Sending (Poll) message (Callback ID=0x98, Expected Reply=0x04) - MeterCmd_Get (Node=14): 0x01, 0x0a, 0x00, 0x13, 0x0e, 0x03, 0x32, 0x01, 0x00, 0x25, 0x98, 0x65
2015-04-30 12:12:07.387 Detail, Node014, Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2015-04-30 12:12:07.388 Detail, Node014, ZW_SEND_DATA delivered to Z-Wave stack
2015-04-30 12:12:07.400 Detail, Node014, Received: 0x01, 0x05, 0x00, 0x13, 0x98, 0x00, 0x71
2015-04-30 12:12:07.400 Detail, Node014, ZW_SEND_DATA Request with callback ID 0x98 received (expected 0x98)
2015-04-30 12:12:07.401 Info, Node014, Request RTT 19 Average Request RTT 19
2015-04-30 12:21:07.502 Detail,
2015-04-30 12:21:07.503 Info, Node014, Response RTT 32 Average Response RTT 31
2015-04-30 12:21:07.503 Detail, Node014, Refreshed Value: old value=false, new value=false, type=bool
2015-04-30 12:21:07.504 Detail, Node014, Changes to this value are not verified
2015-04-30 12:21:07.504 Info, Node014, Received Meter report from node 14: Energy=1699.024kWh
2015-04-30 12:21:07.505 Detail, Node014, Refreshed Value: old value=1699.022, new value=1699.024, type=string
2015-04-30 12:21:07.505 Detail, Node014, Changes to this value are not verified
2015-04-30 12:21:07.506 Detail, Node014, Expected reply and command class was received
2015-04-30 12:21:07.506 Detail, Node014, Message transaction complete
2015-04-30 12:21:07.506 Detail,
2015-04-30 12:21:07.506 Detail, Node014, Removing current message

Fishwaldo, 2015-04-30:
as I asked, please send the complete log file to the google groups list. Snippets don't help.

openzwave-dev-watcher, 2015-05-01:
Here is my OZW log file.

Regards,

Manu

2015-05-01 06:40:05.689 Always, OpenZwave Version 1.3.154 Starting Up
2015-05-01 06:40:05.720 Info, mgr,     Added driver for controller /dev/ttyUSB22
2015-05-01 06:40:05.723 Info,   Opening controller /dev/ttyUSB22
2015-05-01 06:40:05.723 Info, Trying to open serial port /dev/ttyUSB22 (attempt 1)
2015-05-01 06:40:05.772 Info, Serial port /dev/ttyUSB22 opened (attempt 1)
2015-05-01 06:40:05.774 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_VERSION: 0x01, 0x03, 0x00, 0x15, 0xe9
2015-05-01 06:40:05.774 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_MEMORY_GET_ID: 0x01, 0x03, 0x00, 0x20, 0xdc
2015-05-01 06:40:05.774 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_CONTROLLER_CAPABILITIES: 0x01, 0x03, 0x00, 0x05, 0xf9
2015-05-01 06:40:05.775 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_GET_CAPABILITIES: 0x01, 0x03, 0x00, 0x07, 0xfb
2015-05-01 06:40:05.775 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_SUC_NODE_ID: 0x01, 0x03, 0x00, 0x56, 0xaa
2015-05-01 06:40:05.776 Detail, 
2015-05-01 06:40:05.776 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x15) - FUNC_ID_ZW_GET_VERSION: 0x01, 0x03, 0x00, 0x15, 0xe9
2015-05-01 06:40:05.784 Detail, contrlr,   Received: 0x01, 0x10, 0x01, 0x15, 0x5a, 0x2d, 0x57, 0x61, 0x76, 0x65, 0x20, 0x32, 0x2e, 0x37, 0x38, 0x00, 0x01, 0x9b
2015-05-01 06:40:05.785 Detail, 
2015-05-01 06:40:05.786 Info, contrlr, Received reply to FUNC_ID_ZW_GET_VERSION:
2015-05-01 06:40:05.786 Info, contrlr,     Static Controller library, version Z-Wave 2.78
2015-05-01 06:40:05.786 Detail, contrlr,   Expected reply was received
2015-05-01 06:40:05.786 Detail, contrlr,   Message transaction complete
2015-05-01 06:40:05.787 Detail, 
2015-05-01 06:40:05.787 Detail, contrlr, Removing current message
2015-05-01 06:40:05.787 Detail, 
2015-05-01 06:40:05.788 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x20) - FUNC_ID_ZW_MEMORY_GET_ID: 0x01, 0x03, 0x00, 0x20, 0xdc
2015-05-01 06:40:05.811 Detail, contrlr,   Received: 0x01, 0x08, 0x01, 0x20, 0x01, 0x84, 0xe2, 0x8d, 0x01, 0x3d
2015-05-01 06:40:05.812 Detail, 
2015-05-01 06:40:05.812 Info, contrlr, Received reply to FUNC_ID_ZW_MEMORY_GET_ID. Home ID = 0x0184e28d.  Our node ID = 1
2015-05-01 06:40:05.812 Detail, contrlr,   Expected reply was received
2015-05-01 06:40:05.813 Detail, contrlr,   Message transaction complete
2015-05-01 06:40:05.813 Detail, 
2015-05-01 06:40:05.813 Detail, contrlr, Removing current message
2015-05-01 06:40:05.814 Detail, 
2015-05-01 06:40:05.814 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x05) - FUNC_ID_ZW_GET_CONTROLLER_CAPABILITIES: 0x01, 0x03, 0x00, 0x05, 0xf9
2015-05-01 06:40:05.818 Detail, contrlr,   Received: 0x01, 0x04, 0x01, 0x05, 0x1c, 0xe3
2015-05-01 06:40:05.819 Detail, 
2015-05-01 06:40:05.819 Info, contrlr, Received reply to FUNC_ID_ZW_GET_CONTROLLER_CAPABILITIES:
2015-05-01 06:40:05.819 Info, contrlr,     There is a SUC ID Server (SIS) in this network.
2015-05-01 06:40:05.820 Info, contrlr,     The PC controller is an inclusion static update controller (SUC) and was the original primary before the SIS was added.
2015-05-01 06:40:05.820 Detail, contrlr,   Expected reply was received
2015-05-01 06:40:05.820 Detail, contrlr,   Message transaction complete
2015-05-01 06:40:05.821 Detail, 
2015-05-01 06:40:05.821 Detail, contrlr, Removing current message
2015-05-01 06:40:05.821 Detail, 
2015-05-01 06:40:05.822 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x07) - FUNC_ID_SERIAL_API_GET_CAPABILITIES: 0x01, 0x03, 0x00, 0x07, 0xfb
2015-05-01 06:40:05.831 Detail, contrlr,   Received: 0x01, 0x2b, 0x01, 0x07, 0x03, 0x07, 0x00, 0x86, 0x00, 0x02, 0x00, 0x01, 0xfe, 0x80, 0xfe, 0x88, 0x0f, 0x00, 0x00, 0x00, 0xfb, 0x97, 0x3f, 0x82, 0x07, 0x00, 0x00, 0x80, 0x00, 0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x82
2015-05-01 06:40:05.832 Detail, 
2015-05-01 06:40:05.843 Info, contrlr,  Received reply to FUNC_ID_SERIAL_API_GET_CAPABILITIES
2015-05-01 06:40:05.843 Info, contrlr,     Serial API Version:   3.7
2015-05-01 06:40:05.843 Info, contrlr,     Manufacturer ID:      0x0086
2015-05-01 06:40:05.844 Info, contrlr,     Product Type:         0x0002
2015-05-01 06:40:05.844 Info, contrlr,     Product ID:           0x0001
2015-05-01 06:40:05.844 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_RANDOM: 0x01, 0x04, 0x00, 0x1c, 0x20, 0xc7
2015-05-01 06:40:05.845 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_GET_INIT_DATA: 0x01, 0x03, 0x00, 0x02, 0xfe
2015-05-01 06:40:05.845 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_SET_TIMEOUTS: 0x01, 0x05, 0x00, 0x06, 0x64, 0x0f, 0x97
2015-05-01 06:40:05.845 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_APPL_NODE_INFORMATION: 0x01, 0x07, 0x00, 0x03, 0x01, 0x02, 0x01, 0x00, 0xf9
2015-05-01 06:40:05.846 Detail, contrlr,   Expected reply was received
2015-05-01 06:40:05.846 Detail, contrlr,   Message transaction complete
2015-05-01 06:40:05.846 Detail, 
2015-05-01 06:40:05.846 Detail, contrlr, Removing current message
2015-05-01 06:40:05.847 Detail, 
2015-05-01 06:40:05.847 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x56) - FUNC_ID_ZW_GET_SUC_NODE_ID: 0x01, 0x03, 0x00, 0x56, 0xaa
2015-05-01 06:40:05.864 Detail, contrlr,   Received: 0x01, 0x04, 0x01, 0x56, 0x01, 0xad
2015-05-01 06:40:05.865 Detail, 
[... cut ...]
2015-05-01 07:05:19.410 Detail, Node005,   Message transaction complete
2015-05-01 07:05:19.410 Detail, 
2015-05-01 07:05:19.411 Detail, Node005, Removing current message
2015-05-01 07:06:19.437 Detail, Node005,   Received: 0x01, 0x0c, 0x00, 0x04, 0x00, 0x05, 0x06, 0x31, 0x05, 0x01, 0x22, 0x00, 0xc8, 0x2b
2015-05-01 07:06:19.438 Detail, 
2015-05-01 07:06:19.439 Info, Node005, Received SensorMultiLevel report from node 5, instance 1, Temperature: value=20.0C
2015-05-01 07:06:19.439 Detail, Node005, Refreshed Value: old value=20.0, new value=20.0

Fishwaldo, 2015-05-02:
The device doesn't support instant power readings as its just counting pulses from your electricity meter. 

And you need to calibrate the device according to manual to make sure the total value is accurate. Nothing wrong with how OZW is dealing with it.

closed 2015-05-02 by Fishwaldo

manolo75, 2015-05-02:
Ok thanks. Then I will change the device for another model.

Regards,

Manu