OpenZWave/open-zwave #558: Enhance the Alarm CC version 2

Fishwaldo, 2015-05-14:
It doesn't handle the different Alarm Types very well for version 2, and based on Issue #552 and Issue #481 there are some extra fields that we currently ignore as well.

we also need to find all the different alarm events that are possible for the alarm types. 

Possibly change the implementation to a Read only ValueList for the v2 Alarm Type and Alarm Events. That should keep backwards compatibility I think.

gizmocuz, 2015-05-14:
According the philio PST02 manual, and it specifies the values 0x16/0x17
(0x16 = door open, 0x17 = door closed)

When i was debugging i set a breakpoint and notices the 0x03 (temper) and 0x08 (pir)

Fishwaldo, 2015-05-15:
Can you post a Alarm Report Recieved Line (the full string). I want to see the differences between the version 1 fields and version 2 fields in your case.

gizmocuz, 2015-05-15:
you mean the raw byte values?

Fishwaldo, 2015-05-15:
yep... the Recieved Line just before it starts printing out the Alarm Report message. Like this sample:
```
2015-05-16 00:38:43.207 Detail, Node050,   Received: 0x01, 0x28, 0x00, 0x04, 0x00, 0x32, 0x22, 0x98, 0x81, 0x49, 0x3b, 0x8c, 0xf8, 0xa4, 0x34, 0x45, 0xdc, 0x2c, 0x7a, 0x1c, 0xab, 0x81, 0xdd, 0xcd, 0x2d, 0x50, 0x24, 0x2e, 0x65, 0xf0, 0xd0, 0x6f, 0xaa, 0x00, 0x93, 0xa6, 0x83, 0x61, 0x67, 0xde, 0x3a, 0x06
```

gizmocuz, 2015-05-15:
Seems i am not able to attach zip/txt files, here is the log:

PIR triggered:

2015-05-15 18:49:17.842 Detail, Node003,   Received: 0x01, 0x08, 0x00, 0x04, 0x00, 0x03, 0x02, 0x98, 0x40, 0x2a
2015-05-15 18:49:17.845 Info, Node003, Received SecurityCmd_NonceGet from node 3
2015-05-15 18:49:17.845 Info, NONCES: 0x22, 0x11, 0x74, 0x12, 0x8c, 0xd1, 0xa2, 0x5b
2015-05-15 18:49:17.847 Info, NONCES: 0x51, 0x64, 0x92, 0x33, 0xd9, 0xf7, 0xa9, 0xea
2015-05-15 18:49:17.847 Info, NONCES: 0xfa, 0x81, 0x73, 0x41, 0x8e, 0x6f, 0x7a, 0xed
2015-05-15 18:49:17.847 Info, NONCES: 0xa2, 0x2c, 0x73, 0x37, 0x93, 0x68, 0xb7, 0x77
2015-05-15 18:49:17.850 Info, NONCES: 0x36, 0x2b, 0xeb, 0xde, 0xf4, 0x06, 0x32, 0x4f
2015-05-15 18:49:17.850 Info, NONCES: 0x0a, 0xf1, 0xd0, 0x0c, 0x36, 0xca, 0xb1, 0xba
2015-05-15 18:49:17.850 Info, NONCES: 0x96, 0xcb, 0x97, 0xdc, 0x03, 0x13, 0xfd, 0xc1
2015-05-15 18:49:17.852 Info, NONCES: 0x36, 0x34, 0xf5, 0x0e, 0x4d, 0xe7, 0x06, 0x3a
2015-05-15 18:49:17.852 Info, Node003, Sending (Send) message (Callback ID=0x01, Expected Reply=0x00) - Nonce_Report - 0x01, 0x11, 0x00, 0x13, 0x03, 0x0a, 0x98, 0x80, 0x96, 0xcb, 0x97, 0xdc, 0x03, 0x13, 0xfd, 0xc1, 0x05, 0x01, 0xd2:
2015-05-15 18:49:17.862 Detail,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2015-05-15 18:49:17.862 Detail,   ZW_SEND_DATA delivered to Z-Wave stack
2015-05-15 18:49:17.877 Detail,   Received: 0x01, 0x05, 0x00, 0x13, 0x01, 0x00, 0xe8
2015-05-15 18:49:17.880 Detail,   ZW_SEND_DATA Request with callback ID 0x01 received (expected 0x01)
2015-05-15 18:49:17.905 Detail, Node003,   Received: 0x01, 0x34, 0x00, 0x04, 0x00, 0x03, 0x2e, 0x98, 0xc1, 0x3c, 0x36, 0xbb, 0xad, 0xcc, 0x61, 0xb2, 0x48, 0x5d, 0x6a, 0x11, 0xe6, 0x93, 0x4e, 0x83, 0xe1, 0xde, 0x61, 0xa8, 0xa3, 0xff, 0x19, 0xcf, 0x00, 0x25, 0x05, 0xb0, 0x71, 0x88, 0x5a, 0xe6, 0x66, 0x92, 0x06, 0x49, 0x96, 0x87, 0x2a, 0x40, 0x3e, 0x22, 0x6d, 0x4b, 0x9b, 0xa6
2015-05-15 18:49:17.907 Info, Raw: 0x98, 0xc1, 0x3c, 0x36, 0xbb, 0xad, 0xcc, 0x61, 0xb2, 0x48, 0x5d, 0x6a, 0x11, 0xe6, 0x93, 0x4e, 0x83, 0xe1, 0xde, 0x61, 0xa8, 0xa3, 0xff, 0x19, 0xcf, 0x00, 0x25, 0x05, 0xb0, 0x71, 0x88, 0x5a, 0xe6, 0x66, 0x92, 0x06, 0x49, 0x96, 0x87, 0x2a, 0x40, 0x3e, 0x22, 0x6d, 0x4b, 0x9b, 0xa6
2015-05-15 18:49:17.907 Detail, Node003, Decrypted Packet: 0x11, 0x8f, 0x01, 0x04, 0x03, 0x80, 0x03, 0x64, 0x09, 0x71, 0x05, 0x00, 0x00, 0x00, 0xff, 0x07, 0x08, 0x00, 0x05, 0x31, 0x05, 0x03, 0x01, 0x09, 0x06, 0x31, 0x05
2015-05-15 18:49:17.907 Detail, 
2015-05-15 18:49:17.910 Info, Node003, Received encapsulated multi-command from node 3
2015-05-15 18:49:17.910 Info, Node003, Received Battery report from node 3: level=100
2015-05-15 18:49:17.912 Detail, Node003, Refreshed Value: old value=100, new value=100, type=uint8
2015-05-15 18:49:17.912 Detail, Node003, Changes to this value are not verified
2015-05-15 18:49:17.912 Info, Node003, Received Alarm report: type=0, level=0
2015-05-15 18:49:17.915 Detail, Node003, Refreshed Value: old value=0, new value=0, type=uint8
2015-05-15 18:49:17.915 Detail, Node003, Changes to this value are not verified
2015-05-15 18:49:17.915 Detail, Node003, Refreshed Value: old value=0, new value=0, type=uint8
2015-05-15 18:49:17.917 Detail, Node003, Changes to this value are not verified
2015-05-15 18:49:17.917 Detail, Node003, Refreshed Value: old value=255, new value=255, type=uint8
2015-05-15 18:49:17.917 Detail, Node003, Changes to this value are not verified
2015-05-15 18:49:17.920 Detail, Node003, Refreshed Value: old value=8, new value=8, type=uint8
2015-05-15 18:49:17.920 Detail, Node003, Changes to this value are not verified
2015-05-15 18:49:17.920 Info, Node003, Received SensorMultiLevel report from node 3, instance 1, Luminance: value=9%
2015-05-15 18:49:17.922 Detail, Node003, Refreshed Value: old value=6, new value=9, type=string
2015-05-15 18:49:17.922 Detail, Node003, Changes to this value are not verified
2015-05-15 18:49:17.922 Warning, Node003, sensorType Value was greater than range. Dropping
2015-05-15 18:49:17.925 Info, Node003, End of encapsulated multi-command from node 3
2015-05-15 18:49:17.925 Detail, Node003, Notification: ValueChanged
2015-05-15 18:49:17.945 Info, mgr,     Manager::WriteConfig completed for driver with home ID of 0xd3cf129c
2015-05-15 18:49:17.947 Detail, Node003, Notification: ValueChanged
2015-05-15 18:49:17.992 Detail, Node003, Notification: ValueChanged
2015-05-15 18:49:18.042 Detail, Node003, Notification: ValueChanged
2015-05-15 18:49:18.042 Detail, Node003, Notification: ValueChanged
2015-05-15 18:49:18.085 Detail, Node003, Notification: ValueChanged

Door contact open:

2015-05-15 18:50:21.767 Detail, Node003,   Received: 0x01, 0x08, 0x00, 0x04, 0x00, 0x03, 0x02, 0x98, 0x40, 0x2a
2015-05-15 18:50:21.767 Info, Node003, Received SecurityCmd_NonceGet from node 3
2015-05-15 18:50:21.769 Info, NONCES: 0xcf, 0x44, 0x42, 0x83, 0x70, 0x32, 0xc5, 0x74
2015-05-15 18:50:21.769 Info, NONCES: 0xfd, 0x66, 0x9e, 0x76, 0x9e, 0x30, 0xd6, 0xb8
2015-05-15 18:50:21.769 Info, NONCES: 0x9c, 0xf0, 0x50, 0x45, 0xfc, 0xe5, 0x30, 0xd4
2015-05-15 18:50:21.772 Info, NONCES: 0xa2, 0x2c, 0x73, 0x37, 0x93, 0x68, 0xb7, 0x77
2015-05-15 18:50:21.772 Info, NONCES: 0x36, 0x2b, 0xeb, 0xde, 0xf4, 0x06, 0x32, 0x4f
2015-05-15 18:50:21.774 Info, NONCES: 0x0a, 0xf1, 0xd0, 0x0c, 0x36, 0xca, 0xb1, 0xba
2015-05-15 18:50:21.774 Info, NONCES: 0x96, 0xcb, 0x97, 0xdc, 0x03, 0x13, 0xfd, 0xc1
2015-05-15 18:50:21.774 Info, NONCES: 0x11, 0x1f, 0xed, 0x16, 0xab, 0x21, 0xc4, 0x14
[... cut ...]
2015-05-15 18:50:46.426 Warning, Node003, sensorType Value was greater than range. Dropping
2015-05-15 18:50:46.426 Info, Node003, End of encapsulated multi-command from node 3
2015-05-15 18:50:46.428 Detail, Node003, Notification: ValueChanged
2015-05-15 18:50:46.431 Detail, Node003, Notification: ValueChanged
2015-05-15 18:50:46.481 Detail, Node003, Notification: ValueChanged
2015-05-15 18:50:46.523 Detail, Node003, Notification: ValueChanged
2015-05-15 18:50:46.523 Detail, Node003, Notification: ValueChanged
2015-05-15 18:50:46.563 Detail, Node003, Notification: ValueChanged

Fishwaldo, 2015-06-07:
Thanks. I'll look into this soon.

xs4, 2015-06-24:
it looks like the AlarmIndex_SourceNodeId is using the wrong byte. 255 looks like the alarm status. 
Pull request #612 addresses this.

Fishwaldo, 2016-06-28:
I've just re-factored the Alarm CC on the Dev Branch. Its changing the way the ValueID's are exposed compared to to 1.4 version. Please see https://github.com/OpenZWave/open-zwave/wiki/Alarm-Command-Class for details on the new implementation.

closed 2016-06-28 by Fishwaldo