OpenZWave/open-zwave #566: Fibaro 2x1,5kw relay not recognized

daniellast, 2015-05-18:
Fibaro has released an upgrade version of the FGS221 2x1.5kw relay. The new product name is the FGS222 which has almost simular parameters settings as the FGS221.

Can there please be support for this new device in an upgraded version?

Many thanks in advance.

Regards,
Daniel

Fishwaldo, 2015-05-19:
We would need to see the OZW_Log.txt files or the new product type's and ID's to be able to merge. Please send to the mailing list....

daniellast, 2015-05-19:
Hi Justin,
The log I've reviewed it does only contain the line
2015-05-19 16:15:37.730 Always, OpenZwave Version 1.3.200 Starting Up
Looking at the device description it shows FIBARO System Unknown: type=0202, id=1002
type = 0202ID = 1002
I can include the device but only use it with default parameter settings. I've tried editing the file manufacturer_specific.xml making an extra line for this device and using config file fgs221.xml but doens't seem to work.
Regards,Daniel
Date: Tue, 19 May 2015 09:37:09 -0700

Fishwaldo, 2015-05-21:
Closed in PR #567

closed 2015-05-21 by Fishwaldo