OpenZWave/open-zwave #568: Build error on Mac 10.10.3

Antonasg, 2015-05-20:
When building I get the following error:

```
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C /Users/myUserName/Downloads/openzwave-1.2.919/cpp/build/ -
Building Manager.o
/Users/myUserName/Downloads/openzwave-1.2.919/cpp/src/Manager.cpp:2104:11: error: reference cannot be bound to dereferenced null pointer in
      well-defined C++ code; comparison may be assumed to always evaluate to true [-Werror,-Wtautological-undefined-compare]
                                        if( &item != NULL )
                                             ^~~~    ~~~~
1 error generated.
make[1]: *** [/Users/myUserName/Downloads/openzwave-1.2.919/.lib/Manager.o] Error 1
make: *** [all] Error 2
```

stephbu, 2015-06-02:
Root cause is more aggressive static analysis in CLang 3.5 etc. v1.3 release also whining with same errors.  Suppressing the former also showed a similar complaint in vers.cpp too.

```
/Users/stephbu/dev/github.com/openzwave/open-zwave-1.3/cpp/src/vers.cpp:4:30: 
error: expected expression
uint16_t ozw_vers_revision = ;
```

Suppressing former with

```
#pragma clang diagnostic ignored "-Wtautological-compare" 
```
and the latter with

```
uint16_t ozw_vers_revision = 0;
```

Seemed to work ok.

Fishwaldo, 2015-06-07:
I fixed the compile error in the first comment. 

For the issue in the second comment with ozw_vers_revision being assigned "nothing" thats a problem with the build script. It shouldn't be empty. Do you get any errors/warnings from Make when building (try removing cpp/src/vers.cpp before doing make - Its auto generated).

monigrip, 2015-08-03:
I'm getting the same error (mac 10.10.4), How can this be fixed?

/Applications/Xcode.app/Contents/Developer/usr/bin/make -C /Users/myUserName/Downloads/openzwave-1.2.919/cpp/build/ -
Building Manager.o
/Users/myUserName/Downloads/openzwave-1.2.919/cpp/src/Manager.cpp:2104:11: error: reference cannot be bound to dereferenced null pointer in
      well-defined C++ code; comparison may be assumed to always evaluate to true [-Werror,-Wtautological-undefined-compare]
                                        if( &item != NULL )
                                             ^~~~    ~~~~
1 error generated.
make[1]: *** [/Users/myUserName/Downloads/openzwave-1.2.919/.lib/Manager.o] Error 1
make: *** [all] Error 2

Fishwaldo, 2015-08-03:
I just pushed hopefully a fix. Please update and try again

monigrip, 2015-08-03:
awesome, I'll try later. many thanks.

doodahdayz, 2015-08-03:
I saw this error and the looked at the code.  I pulled the version just pushed and it gives the same error still when building on OS X.  The real issue here that the compiler has is that you are testing the address of a constant variable that is in memory to see if it si NULL.  It cannot ever be NULL.  Its the address of something that is in memory.

So what the compiler wants you to do is either remove the “&” from in front of item and test to see if this points to NULL or just remove the test for NULL from the offending if statement.

Fishwaldo, 2015-08-03:
I forgot to actually commit the fix after the refactor.

Fishwaldo, 2015-08-03:
The previous version was returning a reference, not a pointer, hence why we were testing the reference for NULL. It is undefined (but works on GCC and Windows) to return a reference pointing at NULL... its just that llvm didn't like it...

doodahdayz, 2015-08-03:
Ok.. Pulled that version and it now builds on OS X.  Pops out some warnings from building MinOZW about some placeholder code assigning a variable to itself, but builds fine.  I’m curious now if OZWCP will build on OS X.  Has anyone successfully built that lately on a Mac?

Fishwaldo, 2015-08-03:
I build regularly on Mac.

doodahdayz, 2015-08-03:
I pulled the latest OZWCP to mine… it doesn’t build on OS X 10.11  …..  I had to add a flag to the compiler options to get past one error about an used field in a structure and then the link statement is incorrect to pick up the openzwave library when it links… changing those it appears to then build fine on Mac.

I took the latest open-wave build to my Raspberry Pi and it wouldn’t compile PowerLevel.cpp now….  It doesn’t like that the powerLevel variable wasn’t initialed….     setting it to a default enumeration of PowerLevel_Normal when its declared in these two functions then lets it compile on the Raspberry Pi Linux.

On the Pi the compiler spits out the following:
Building Powerlevel.o
/home/pi/open-zwave-master/cpp/src/command_classes/Powerlevel.cpp: In member function ‘bool OpenZWave::Powerlevel::Set(uint8)’:
/home/pi/open-zwave-master/cpp/src/command_classes/Powerlevel.cpp:355:2: error: ‘powerLevel’ may be used uninitialized in this function [-Werror=uninitialized]
/home/pi/open-zwave-master/cpp/src/command_classes/Powerlevel.cpp: In member function ‘bool OpenZWave::Powerlevel::Test(uint8)’:
/home/pi/open-zwave-master/cpp/src/command_classes/Powerlevel.cpp:436:33: error: ‘powerLevel’ may be used uninitialized in this function [-Werror=uninitialized]
cc1plus: all warnings being treated as errors
/home/pi/open-zwave-master/cpp/build/support.mk:106: recipe for target '/home/pi/open-zwave-master/.lib/Powerlevel.o' failed
make[1]: *** [/home/pi/open-zwave-master/.lib/Powerlevel.o] Error 1
make[1]: Leaving directory '/home/pi/open-zwave-master/cpp/build'
Makefile:20: recipe for target 'all' failed
make: *** [all] Error 2

It

Fishwaldo, 2015-08-03:
Could you please make a pr for the ozwcp fixes. 

The power level fix is already done

closed 2015-08-03 by Fishwaldo

doodahdayz, 2015-08-03:
I have no clue what you mean….  I’ve never worked with git or any of this stuff you young guns use these days.  I’m basically just old and useless and just hacking around trying to keep a few brain cells still moving.

The changes I had to make for OZWCP was to add this to the compiler flags:  “-Wno-unused-private-field”
and then also I had to remove the line that was in the MAC section which was redefining LIBZWAVE to point to the wrong place.  So instead of the comment in the makefile telling Mac users to uncomment the next 5 lines… it should only be 4 lines… Leave LIBZWAVE the same as for Linux build and delete that line from the section for Mac build.

I also removed the -pthread option in the LIBS definition for Mac…  it was being ignored and just generating a warning that it was being ignored.

monigrip, 2015-08-04:
Hi,

I'm old too, and don't have a clue either way. Just fumbling around using google searches and learning something along the way. I've managed to install the open-zwave, but now I've errors installing domoticz, which installed fine before (without z-wave), I didn't get much time last night to have a proper look, but you guys might be able to help and point me in the right direction.

Cheers,

Undefined symbols for architecture x86_64:

  "OpenZWave::Manager::AddAssociation(unsigned int, unsigned char, unsigned char, unsigned char)", referenced from:

      COpenZWave::UpdateValue(OpenZWave::ValueID const&) in OpenZWave.cpp.o

      COpenZWave::AddNodeToGroup(int, int, int) in OpenZWave.cpp.o

      MyNode::updateGroup(unsigned char, unsigned char, char*) in ozwcp.cpp.o

     (maybe you meant: __ZN9OpenZWave7Manager14AddAssociationEjhhhh)

  "OpenZWave::Manager::RemoveAssociation(unsigned int, unsigned char, unsigned char, unsigned char)", referenced from:

      COpenZWave::RemoveNodeFromGroup(int, int, int) in OpenZWave.cpp.o

      MyNode::updateGroup(unsigned char, unsigned char, char*) in ozwcp.cpp.o

     (maybe you meant: __ZN9OpenZWave7Manager17RemoveAssociationEjhhhh)

ld: symbol(s) not found for architecture x86_64

clang: error: linker command failed with exit code 1 (use -v to see invocation)

make[2]: *** [domoticz] Error 1

make[1]: *** [CMakeFiles/domoticz.dir/all] Error 2

make: *** [all] Error 2

Fishwaldo, 2015-08-04:
You probably have to take this up with domoticz, as they bundle OZW with their build scripts.

monigrip, 2015-08-04:
Thought that might be the case, thanks, will do.

greggwon, 2015-08-06:
There are lots of places where size_t needs to be used instead of uin32_t on Mac OS X.  I will try to put together a pull request sometime soon.

Fishwaldo, 2015-08-06:
Honestly why?  We are never going to process a packet or file that big and it has the potential to break the API if you touch public functions. I won't merge anything that breaks the API unless it's a bug fix until we get to the stage where we are considering moving to 2.x version.

greggwon, 2015-08-06:
I guess it's up to you.  I can just maintain my changes as a fork and deal with any of the issues that create compilation problems.

rbaindourov, 2016-06-16:
I really recommend fixing any build issues.
I have Ubuntu Desktop / Raspbian PI3 / and Macbook Pro as part of my work environment.
I am not able to build on Mac because of linker errors:

https://github.com/OpenZWave/open-zwave-control-panel/issues/26

Being able to compile on Mac is important.
Even urgent, in my opinion, since my Mac is the magic company laptop.
I need to be able to demo and develop on the go, as part of my job description.