OpenZWave/open-zwave #575: Basic report for ZME_064435

spyjo, 2015-05-23:
I had a problem with basic report on this device, when I push a button, I see the basic event in the openzwave log, but the change was not reported in the software (jeedom / python-openzwave)

So I added this line in the config file of the device config/zwave.me/ZME_064435.xml :

```
<CommandClass id="32" setasreport="true"/>
```

I copied this line from other config files..

Now it's working for me. May be the config file need to be patched for everyone, unless someone tells this was already working for him without this line..

Thanks

nechry, 2015-05-24:
I don't have this device to validate, but yes is, for some device, the right way to got report. So we can add this xml element in specific device configuration.

spyjo, 2015-05-25:
Thank you for making the pull request, I was thinking to make one but i'm not a github expert ;)

Fishwaldo, 2015-05-28:
Closed by PR #578

closed 2015-05-28 by Fishwaldo