OpenZWave/open-zwave #584: Vision zp3102 (New configuration file)

gizmocuz, 2015-05-29:
I received a PIR sensor to test (it was not working correctly).

I finally got the sensor working correctly, and needed to modify the configuration file.

Is it possible to replace the current one with the attached one?

Most problems where in values being requested that where not answered back, but the status was reporting correctly

i have attached the new configuration file in mailen list at:

https://groups.google.com/forum/#!topic/openzwave/tNk-XtNenYI

With kind regards,
Rob

Fishwaldo, 2015-06-10:
As commented, can you post of log file while doing inclusion on this device? so many unsupported CC's seems dubious. I just want to confirm before merging.

openzwave-dev-watcher, 2015-06-10:
dont have the device here anymore.
Without these lines, the sensor was not working, until we manually wake it

still reports the values correctly

Erik-NA, 2015-12-13:
Please check issue #698 (which should have been fixed but isn't?) and try to use that config file.

Fishwaldo, 2016-10-07:
Please reopen if this is still a issue

closed 2016-10-07 by Fishwaldo