OpenZWave/open-zwave #585: Node 0/255 issues

gizmocuz, 2015-05-31:
Using the current master branch:

I noticed twice now that i received a "Type_NodeAdded" with nodeID = 0
Don't know what is causing it, and once when i included a node, it was added as Node '0' with no product ID

Then when i looked in configuration file (zwcfg_xxxx.xml), the first entry was ... Node 0

This was also my problem that i never received:

- Type_AwakeNodesQueried
- Type_AllNodesQueried
- Type_AllNodesQueriedSomeDead

it looks like when this happens, it is never returning any of the above notifications

(Took long to figure that one out)

Then (just now) i received the notification "Type_NodeQueriesComplete", with nodeID = 255

Is 'NodeID=0' or 'NodeID=255' used somewhere in the code for 'special' messages/nodes ?

If not, can i propose to start checking the m_nodes from 1 to 255 (not till?)

there is one place in the code that this is done (beginning from 1)

void Driver::SaveButtons

Node 0 and Node 255 should (i believe) not be possible.

Fishwaldo, 2015-06-07:
Anyway you can get a Log of when these Node 0 and Node 255 are found. I've seen some sporadic Nodes but only in logging methods....... 

It would be better to find out where its starting from.

nechry, 2015-07-14:
Hello Justin,

We got from time to time this issues a node 0 appear in network. I found a trace wend is added:
```
2015-07-14 09:29:45.243 Info, Add Device
2015-07-14 09:29:45.245 Info, contrlr, Sending (Command) message (Callback ID=0xd5, Expected Reply=0x4a) - AddDevice: 0x01, 0x05, 0x00, 0x4a, 0x81, 0xd5, 0xe4
2015-07-14 09:29:45.742 Info, contrlr, FUNC_ID_ZW_ADD_NODE_TO_NETWORK:
2015-07-14 09:29:45.742 Info, contrlr, ADD_NODE_STATUS_FAILED
2015-07-14 09:29:45.743 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x4a) - Add Node Stop: 0x01, 0x04, 0x00, 0x4a, 0x05, 0xb4
2015-07-14 09:29:45.797 Info, contrlr, FUNC_ID_ZW_ADD_NODE_TO_NETWORK:
2015-07-14 09:29:45.797 Info, contrlr, ADD_NODE_STATUS_DONE
2015-07-14 09:29:45.797 Info, Initilizing Node. New Node: true (true)
2015-07-14 09:29:45.799 Info, Sending (Query) message (Callback ID=0x00, Expected Reply=0x41) - Get Node Protocol Info (Node=0): 0x01, 0x04, 0x00, 0x41, 0x00, 0xba
2015-07-14 09:29:45.803 Info, Received reply to FUNC_ID_ZW_GET_NODE_PROTOCOL_INFO
2015-07-14 09:29:45.803 Info,   Protocol Info for Node 0 reports node nonexistent
```
Sometime if we try a node inclusion, the processes start, but also failed quickly by it self, but ozw still add a fake node with id 0.

to remove this node we have to mark node 0 as failed and then use remove failed node command. other wise the network become totally unstable

If the inclusion failed as has show in this example, the only way to add again a node is to stop the network and restart. also if we start the network with a node 0 inside the network is really unstable and take more time to become ready.

So I hope you have more information here to investigate this issue. I suspected this behavior has started after the security inclusion have merged to master branch. previously I don't remember got this node 0. but yes, sometime the inclusion don't started, or failed.

Fishwaldo, 2015-07-15:
Thanks, that gives me a good lead where things go wrong. 

Im currently travelling, but will try to push a fix soon for that.

nechry, 2015-07-15:
also, if you have any clue why the controller stop by it's self the inclusion process's? I know it's a wide question...

Fishwaldo, 2015-07-20:
any error during inclusion (network traffic, busy SUC/SIS etc etc etc will cause a inclusion to fail. in this case, you got a ADD_NODE_STATUS_FAILED message, and then we should exit from inclusion. The problem was when we were exiting inclusion mode, we didn't check if there was a failure of not. I'm about to push a fix for that.

Fishwaldo, 2015-07-20:
Please post a message if it reoccurs.

closed 2015-07-20 by Fishwaldo

nechry, 2015-07-20:
Thank we will try