OpenZWave/open-zwave #586: Building error driver.cpp

Sybon876, 2015-06-02:
On my arch linux system I have installed gcc 5.1.0-4 and by building open-zwave I get the following error: 

make -C /home/seb/install/open-zwave-read-only/cpp/build/ -
make[1]: Entering directory '/home/seb/install/open-zwave-read-only/cpp/build'
Building OpenZWave Version 1.3-293-ga4ea66b-dirty
Building Driver.o
/home/seb/install/open-zwave-read-only/cpp/src/Driver.cpp: In member function ‘void OpenZWave::Driver::UpdateNodeRoutes(uint8, bool)’:
/home/seb/install/open-zwave-read-only/cpp/src/Driver.cpp:6345:17: error: array subscript is above array bounds [-Werror=array-bounds]
      if( nodes[k] == associations[j] )
                 ^
cc1plus: all warnings being treated as errors
/home/seb/install/open-zwave-read-only/cpp/build/support.mk:108: recipe for target '/home/seb/install/open-zwave-read-only/.lib/Driver.o' failed
make[1]: *** [/home/seb/install/open-zwave-read-only/.lib/Driver.o] Error 1
make[1]: Leaving directory '/home/seb/install/open-zwave-read-only/cpp/build'
Makefile:20: recipe for target 'all' failed
make: *** [all] Error 2

By removing the complete "Driver::UpdateNodeRoutes" from the driver.cpp file I was able to build driver.cpp (didn't complete the complete oz because missing of "Driver::UpdateNodeRoutes"). Also by manipulating the "numNodes" value of the for loops I was able to build but still didn't found out what is causing the error.

mach327, 2015-06-06:
Same version and same problem here. It looks like it might be a GCC bug.
[Possibly Related GCC Bug report](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=59124)

I temporarily bypassed this while building python-openzwave by adding `&& k < sizeof(nodes)` to the for loop condition above where GCC complains. Another solution would be to remove `-Werror` from open-zwaves compile flags.

     for( k = 0; k < numNodes && k < sizeof(nodes); k++ )
     {
        if( nodes[k] == associations[j] ) //Line that GCC complains about, 6157

closed 2015-06-07 by Fishwaldo

Sybon876, 2015-06-11:
Worked for me, thanks.