OpenZWave/open-zwave #590: Building libopenzwave with or without libudev

gkroone, 2015-06-05:
I am working on packaging libudev on synology, which uses udev, but doesn't come with libudev. I managed to compile libudev for it, but it comes at a cost of additional compile time and package maintenance as there's quite some unique dependency required to build it, just for openzwave. We have the option of patching out the udev dependency, but I'd like to understand what the downside would be for doing that. Would you be able to comment on the benefits of using a hid device over a normal usb serial device, and if using a normal usb serial device will be sustainable in the future? Do we lack support for certain device types if not using libudev or expect stability issues? I see that for openzwave the libudev is not required? I saw some commit messages on making the libudev dependency platform dependent? So do you now offer a configuration parameter to remove the depenency on libudev?

Fishwaldo, 2015-06-07:
without udev, you can't talk to HID based controllers. There is only one old HID controller, so I think you would be safe not supporting them. Any further questions, please use the mailing list.

closed 2015-06-07 by Fishwaldo

gizmocuz, 2016-01-31:
I would like to open this old thread.

Having issues with this as well.

"There is only one old HID controller"

How about we drop support for this HID device completely ? The one (few) users using this OLD controller sure have saved some money now to buy a new controller (with zwave+ support) ?

Is it OK if i make a pull request for this ?