OpenZWave/open-zwave #599: Internal errors and dropped message: Log analyzer said to report this

ash7777, 2015-06-11:
I modified MinOZW to unlock my Shlage BE469 door lock, and the log analyzer said to report some issues in the log file that resulted. The unlock worked, but apparently not perfectly.

The analysis link is:
http://www.openzwave.com/log-analyzer/public/c9f5da6a-5a32-47ec-804f-b9c1b30618f9/

Let me know if you need any additional information.

Thanks!

Fishwaldo, 2015-06-15:
without knowing what you changed:

Warning 58 - a whole bunch of messages were transmitted in quick succession without gettting a chance to ACK. Did you modify any timeouts or queue code?

Error 4 - Again, seems the Queues got all messed up.

Can you tell us what you changed? that way I can figure out if your change is buggy, or its a OZW bug?

ash7777, 2015-06-15:
I have attached the modified program I used (comments haven't been 
updated from the original program). It has a variable "doThisOne" that I 
used so I could change the variable I wanted (I didn't set the variable, 
just directly changed the next line to execute). Perhaps OZW was 
confused because I was stepping in a debugger, which can cause 
abnormally long delays during execution?

// Home Automation Core.cpp : Defines the exported functions for the DLL application.
//

#include "stdafx.h"
#include "Home Automation Core.h"
#include "Manager.h"
#include "Options.h"
#include "pthread.h"
#include "Notification.h"

using namespace OpenZWave;

static pthread_mutex_t g_criticalSection;
static pthread_cond_t  initCond = PTHREAD_COND_INITIALIZER;
static pthread_mutex_t initMutex = PTHREAD_MUTEX_INITIALIZER;

typedef struct
{
	uint32			m_homeId;
	uint8			m_nodeId;
	bool			m_polled;
	list<ValueID>	m_values;
}NodeInfo;

static list<NodeInfo*> g_nodes;
bool temp = false;
static uint32 g_homeId = 0;
static bool   g_initFailed = false;

// This is an example of an exported variable
HOMEAUTOMATIONCORE_API int nHomeAutomationCore=0;

// This is an example of an exported function.
HOMEAUTOMATIONCORE_API int fnHomeAutomationCore(void)
{
	return 42;
}

NodeInfo* GetNodeInfo(const Notification* _notification)
{
	const uint32 homeId = _notification->GetHomeId();
	const uint8 nodeId = _notification->GetNodeId();
	for (list<NodeInfo*>::iterator it = g_nodes.begin(); it != g_nodes.end(); ++it)
	{
		NodeInfo* nodeInfo = *it;
		if (nodeInfo->m_homeId == homeId && nodeInfo->m_nodeId == nodeId)
			return nodeInfo;
	}

	return NULL;
}

void OnNotification(Notification const* _notification, void* _context)
{
	// Must do this inside a critical section to avoid conflicts with the main thread
	pthread_mutex_lock(&g_criticalSection);

	switch (_notification->GetType())
	{
		case Notification::Type_ValueAdded:
			{
				NodeInfo* nodeInfo = GetNodeInfo(_notification);
				if (nodeInfo)
[... 204 lines cut ...]
		// We must hold the critical section for as short a time as possible, to avoid
		// stalling the OpenZWave drivers.
		// At this point, the program just waits for 3 minutes (to demonstrate polling),
		// then exits
		for (int i = 0; i < 60 * 3; i++)
		{
			pthread_mutex_lock(&g_criticalSection);
			// but NodeInfo list and similar data should be inside critical section
			pthread_mutex_unlock(&g_criticalSection);
			Sleep(1000);
		}

		Driver::DriverData data;
		Manager::Get()->GetDriverStatistics(g_homeId, &data);
		printf("SOF: %d ACK Waiting: %d Read Aborts: %d Bad Checksums: %d\n", data.m_SOFCnt, data.m_ACKWaiting, data.m_readAborts, data.m_badChecksum);
		printf("Reads: %d Writes: %d CAN: %d NAK: %d ACK: %d Out of Frame: %d\n", data.m_readCnt, data.m_writeCnt, data.m_CANCnt, data.m_NAKCnt, data.m_ACKCnt, data.m_OOFCnt);
		printf("Dropped: %d Retries: %d\n", data.m_dropped, data.m_retries);
	}

	Manager::Get()->RemoveDriver(port);
	Manager::Get()->RemoveWatcher(OnNotification, NULL);
	Manager::Destroy();
	Options::Destroy();
	pthread_mutex_destroy(&g_criticalSection);
}

Fishwaldo, 2015-07-03:
I cant see anything wrong with this... but yeah, as OZW is multithreaded, a debugger might be screwing with things there.

closed 2015-07-03 by Fishwaldo