OpenZWave/open-zwave #601: Issue at starting Network

sarakha63, 2015-06-14:
hello ,
I'm using the last version of OZW. i also change my controller for a gen5 aeon stick.
Since those two changes, i experiment some strange behaviours :
Often the network drops and at the restart i have some kinf of infinite loops on some notifications making the queue grow to thousands of element.
Generally this kind of loops happens on group associations notifications (or some other notifications). Did you already heard this kind of behaviour ?
Is it related to an issue with the controller or some kind of network start issue on some network. I'm not really sure it is an OZW issue. But maybe you are aware of what could cause this kind of behaviour and help me focus on the cause.

Here is some examples of logs showing this issue

 The first one shows the beginning of a loop for node 20 (in this case this is a Nodon smartplus this kind of behaviour happened also on a fibaro RGBW
```
 2015-06-06 22:28:44.789 Info, Node020, Number of association groups reported for                               node 20 is 8.
2015-06-06 22:28:44.789 Info, Node020, Get Associations for group 1 of node 20
2015-06-06 22:28:44.790 Info, Node020, Sending (Send) message (Callback ID=0x3f,                               Expected Reply=0x04) - AssociationCmd_Get (Node=20): 0x01, 0x0a, 0x00, 0x13, 0x                              14, 0x03, 0x85, 0x02, 0x01, 0x25, 0x3f, 0x6d
2015-06-06 22:28:44.839 Info, Node020, Request RTT 48 Average Request RTT 47
2015-06-06 22:28:47.664 Info, Node020, Response RTT 2874 Average Response RTT 28                              74
2015-06-06 22:28:47.664 Info, Node020, Received Association report from node 20,                               group 1, containing 1 associations
2015-06-06 22:28:47.665 Info, Node020,   The group contains:
2015-06-06 22:28:47.665 Info, Node020,     Node 1
2015-06-06 22:28:47.665 Info, Node020, Get Associations for group 2 of node 20
2015-06-06 22:28:47.666 Info, Node020, Sending (Send) message (Callback ID=0x40,                               Expected Reply=0x04) - AssociationCmd_Get (Node=20): 0x01, 0x0a, 0x00, 0x13, 0x                              14, 0x03, 0x85, 0x02, 0x02, 0x25, 0x40, 0x11
2015-06-06 22:28:47.713 Info, Node020, Response RTT 46 Average Response RTT 1460
2015-06-06 22:28:47.713 Info, Node020, Received Association report from node 20,                               group 1, containing 1 associations
2015-06-06 22:28:47.713 Info, Node020,   The group contains:
2015-06-06 22:28:47.714 Info, Node020,     Node 1
2015-06-06 22:28:47.714 Info, Node020, Get Associations for group 2 of node 20
2015-06-06 22:28:47.820 Info, Node020, Request RTT 153 Average Request RTT 100
2015-06-06 22:28:47.860 Info, Node020, Response RTT 193 Average Response RTT 826
2015-06-06 22:28:47.860 Info, Node020, Received Association report from node 20,                               group 1, containing 1 associations
2015-06-06 22:28:47.860 Info, Node020,   The group contains:
2015-06-06 22:28:47.861 Info, Node020,     Node 1
2015-06-06 22:28:47.861 Info, Node020, Get Associations for group 2 of node 20
2015-06-06 22:28:47.861 Info, Node020, Sending (Send) message (Callback ID=0x41,                               Expected Reply=0x04) - AssociationCmd_Get (Node=20): 0x01, 0x0a, 0x00, 0x13, 0x                              14, 0x03, 0x85, 0x02, 0x02, 0x25, 0x41, 0x10
2015-06-06 22:28:47.899 Info, Node020, Response RTT 37 Average Response RTT 431
2015-06-06 22:28:47.899 Info, Node020, Received Association report from node 20,                               group 1, containing 1 associations
2015-06-06 22:28:47.899 Info, Node020,   The group contains:
2015-06-06 22:28:47.899 Info, Node020,     Node 1
2015-06-06 22:28:47.900 Info, Node020, Get Associations for group 2 of node 20
2015-06-06 22:28:48.109 Info, Node020, Response RTT 248 Average Response RTT 339
2015-06-06 22:28:48.109 Info, Node020, Received Association report from node 20,                               group 1, containing 1 associations
2015-06-06 22:28:48.109 Info, Node020,   The group contains:
2015-06-06 22:28:48.109 Info, Node020,     Node 1
2015-06-06 22:28:48.109 Info, Node020, Get Associations for group 2 of node 20
2015-06-06 22:28:48.151 Info, Node020, Request RTT 290 Average Request RTT 195
2015-06-06 22:28:49.498 Info, Node020, Response RTT 1637 Average Response RTT 98                              8
2015-06-06 22:28:49.498 Info, Node020, Received Association report from node 20,                               group 2, containing 0 associations
2015-06-06 22:28:49.498 Info, Node020, Get Associations for group 3 of node 20
2015-06-06 22:28:49.499 Info, Node020, Sending (Send) message (Callback ID=0x42,                               Expected Reply=0x04) - AssociationCmd_Get (Node=20): 0x01, 0x0a, 0x00, 0x13, 0x                              14, 0x03, 0x85, 0x02, 0x02, 0x25, 0x42, 0x13
2015-06-06 22:28:49.561 Info, Node020, Request RTT 61 Average Request RTT 128
2015-06-06 22:28:49.815 Info, Node020, Response RTT 315 Average Response RTT 651
2015-06-06 22:28:49.815 Info, Node020, Received Association report from node 20,                               group 2, containing 0 associations
2015-06-06 22:28:49.815 Info, Node020, Get Associations for group 3 of node 20
2015-06-06 22:28:49.816 Info, Node020, Sending (Send) message (Callback ID=0x43,                               Expected Reply=0x04) - AssociationCmd_Get (Node=20): 0x01, 0x0a, 0x00, 0x13, 0x                              14, 0x03, 0x85, 0x02, 0x02, 0x25, 0x43, 0x12
2015-06-06 22:28:50.797 Info, Node020, Request RTT 981 Average Request RTT 554
2015-06-06 22:28:54.242 Info, Node005, Received SensorBinary report: Sensor:40 S                              tate=On
2015-06-06 22:28:54.473 Info, Node005, Received SensorBinary report: Sensor:40 S                              tate=On
2015-06-06 22:28:54.684 Info, Node005, Received SensorBinary report: Sensor:40 S                              tate=On
2015-06-06 22:28:55.433 Info, Node005, Received Basic set from node 5: level=255                              . Treating it as a Basic report.
2015-06-06 22:28:55.817 Error, Node020, ERROR: Dropping command, expected respon                              se not received after 1 attempt(s)
2015-06-06 22:28:55.818 Info, Node020, Sending (Send) message (Callback ID=0x44,                               Expected Reply=0x04) - AssociationCmd_Get (Node=20): 0x01, 0x0a, 0x00, 0x13, 0x                              14, 0x03, 0x85, 0x02, 0x02, 0x25, 0x44, 0x15
2015-06-06 22:28:55.973 Info, Node005, Received Basic set from node 5: level=255                              . Treating it as a Basic report.
2015-06-06 22:28:55.976 Info, Node005, Received Basic set from node 5: level=255                              . Treating it as a Basic report.
2015-06-06 22:28:56.016 Info, Node020, Request RTT 198 Average Request RTT 376
2015-06-06 22:28:56.335 Info, Node005, Received Basic set from node 5: level=255                              . Treating it as a Basic report.
2015-06-06 22:28:58.259 Info, Node020, Response RTT 2441 Average Response RTT 15                              46
2015-06-06 22:28:58.259 Info, Node020, Received Association report from node 20,                               group 2, containing 0 associations
2015-06-06 22:28:58.259 Info, Node020, Get Associations for group 3 of node 20
2015-06-06 22:28:58.260 Info, Node020, Sending (Send) message (Callback ID=0x45,                               Expected Reply=0x04) - AssociationCmd_Get (Node=20): 0x01, 0x0a, 0x00, 0x13, 0x                              14, 0x03, 0x85, 0x02, 0x03, 0x25, 0x45, 0x15
2015-06-06 22:28:58.309 Info, Node020, Response RTT 49 Average Response RTT 797
2015-06-06 22:28:58.309 Info, Node020, Received Association report from node 20,                               group 2, containing 0 associations
[... 158 lines cut ...]
2015-06-06 22:29:24.292 Info, Node020,   The group contains:
2015-06-06 22:29:24.292 Info, Node020,     Node 1
2015-06-06 22:29:24.292 Info, Node020, Get Associations for group 5 of node 20
2015-06-06 22:29:24.293 Info, Node020, Sending (Send) message (Callback ID=0x53,                               Expected Reply=0x04) - AssociationCmd_Get (Node=20): 0x01, 0x0a, 0x00, 0x13, 0x                              14, 0x03, 0x85, 0x02, 0x04, 0x25, 0x53, 0x04
2015-06-06 22:29:24.328 Info, Node020, Response RTT 34 Average Response RTT 641
[... cut ...]
2015-06-14 16:09:57.818 Info, Node028, Received SensorMultiLevel report from node 28, instance 1, Luminance: value=53lux
2015-06-14 16:09:57.911 Info, Node028, Received SensorMultiLevel report from node 28, instance 1, Luminance: value=54lux
```

And this happened a lot .

Don't focus on the nodes as when this happens it happens almost on all the nodes of the network. And another information very often all the nodes (even the plugged one )  gives timeouts on the network starting process

Hope ou will have some clues

Fishwaldo, 2015-06-15:
Question - Did you recently update from Github? Make sure you do a "make clean && make" when you update as it looks like some sort of corruption.

Otherwise, I'm at a loss for whats going on.

sarakha63, 2015-06-18:
Hello,

Yes i was up to date from a scratch install to be sure.

I will try to investiguate and give you more elements if i can

Fishwaldo, 2015-07-03:
any update on this?

sarakha63, 2015-07-05:
Hello Justin.

I think we can close this issue. As some improvements were made on the way the requests are made on OZW start in Jeedom end i never had this issue since. Before i used to encounter it once a day at least. So this was probably not an OZW issue

closed 2015-07-05 by sarakha63