OpenZWave/open-zwave #605: RefreshNodeInfo manager method don't trigger the Fetching of data for a node

nechry, 2015-06-18:
In some case a node don't report the ProtocolInfo during the initial query. early if I found a node without all ProtocolInfo  information I use manager RefreshNodeInfo  command to Fetch of data for a node. 
This feature seem no more working, the query state return but no more data are got. I check in history and I found during the Security CC work https://github.com/OpenZWave/open-zwave/commit/e4fb138b622ad3b9fc561c7328c45c114f23cfaf 
you add the test to stop the RefreshNodeInfo if ProtocolInfo is received previously.

me be this flag must reset if I want force the fetch ? or is something else.

thank for your answer

Fishwaldo, 2015-06-22:
Hi,
Yep - I can fix this up easily enough, but I'm interested to understand why you don't get a full protocolinfo in the first place? That sounds like a issue as well.

nechry, 2015-06-22:
Hi Justin, I don't have a recipe to got this issues. Most of time i got everything, but some time if I fulle remove the cfg and start my network I got this issues, I mean no type and product id's. Sometimes it's occurred durring a inclusions.

nechry, 2015-06-25:
Hi Justin, we got more and more the same case as I report with lot of users.

You told is easy to fix, if you can release this "grale" patch we will got happy users

sarakha63, 2015-06-25:
Hello Justin, effectively this patch could be very usefull. But effectively we have more and more users with issue on inclusions with no ids.

Fishwaldo, 2015-06-25:
It will be a couple of days before I can get to it. 

But I'd still like to see logs. This shouldn't happen in the first place.

Fishwaldo, 2015-06-30:
I can't see how this ever worked before the Security CC was rewritten based on what you said. we have always had a flag that would not request ProtocolInfo if it was already set in AdvanceQueries. I need to see logs of this before and after. When I test on the OZWCP, its working as intended right now. 

Or, we could fix the reason why you are not getting full protocolinfo messages in the first place, which seems to be the real cause of your issues.

nechry, 2015-06-30:
Ok me be I don't point the right line in code, but yes every time I didn't got all product type and id I use the method. Previously via the beginCommand system of course. Now is deprecated I use the direct method. Recently I included lot of device without got the issue but as I remember is not something new to don't get the full interview at this query stage. Went I got this case, of course all others releated stage for groups, parameters and values also empty. 

I have lot off interview to execute until friday, I pass all my nodes form a aeon-labs v2 stick to a gen5. So I expect got the issue at least one time.

nechry, 2015-06-30:
Completely information: sometimes it's also occurred if I remove the zwcfg.xml file and force a full repopulate of all network data. Not for the entire nodes just few of then and not just the battery operated, but mostly. Also if I stop the network before reaching topology loaded, and I restarted the network with the uncompleted wzcfg.xml file the protocol info is passed without perform a device query. As I understand off course.

Fishwaldo, 2015-07-03:
ehhhh. RefreshNodeInfo never went through the beginControllerCommand functions. Its always been standalone :-1: 

send me one of these uncompleted zwcfg_*.xml files.... the query_stage for each node should be Probe or Probe1 if its dead from the restart... 

Were any of these nodes Security enabled? There was a bug I just fixed that if a Security Device failed one NONCE Requests, it would still pass through the various QueryStages and appear "alive" rather than dead.

nechry, 2015-07-03:
Hi Justin,

Yes the RefreshNodeInformation previously is standalone. Is just to explain before the cc security they always worked. 

So in my network I have some aeon-labs gen5 in my network, and yes at least one time a smart switch fall in this case as I remember (I hope, my memory is better for device) . But it occurring with many no secured device too. From my side I didn't got this issue in my last including process, but some user still report this issue. Is more complicated to got there log file, so I will wait to got the issue from my side to send you a full details. But trust me it's occurred many time and the RefreshNodeInformation previously always (90% of time) fix partial node discovery in cfg file.

Nechry

Fishwaldo, 2015-09-15:
any update on the logs or zwcfg_*.xml file?

nechry, 2015-09-15:
Hi, sorry yes it's occure from time to time but I add a work around, I add a helper to remove the node information in xml file and restart ozw. It's not cool beacose it's mean we must do the nodes discovery again, if the refresh node info like before can do it as before, it's better. 
So I have a developement instance, I will try to got the case and give you everithing you nead to investigate this issue

nechry, 2015-10-06:
hello I perform many tests this week-end and every time I try the RefreshNodeInfo I got the expected result so a device form no product type and id to a fully detected device. I don't know if something else restore this behavior with latest update, but from my point of view this issue can be close now.

closed 2015-10-06 by nechry