OpenZWave/open-zwave #608: Minimote Works but some issues

sarakha63, 2015-06-21:
Hello,
 I'm from Jeedom team and i have some strange beahviour with the minimote from Aeotec.

Everythings works fine once included param 250 set to scene.

But even if ti works i have strange issues.

1) the device does'nt accept group association with the controller or another node
2) the device doesn't report it's battery level (i'm almost sure it used to do it before)
3) a very well known issue but maybe related to the fact that the minimote doesnt respond to request on some class (when param 250 is changed it always show as same value even if i can surely say the device accepted the change)
4) This one is the most annoying. i have a keyfob gen 5 and a minimote on my network .Both are configured as remotes (not as primary conroller). If i make an exclusion of the minimote, the keyfob is the one excluded from my network, an i must reexclude the minimote to make it be excluded. And unfotunately i have to reinclude the keyfob.

Thanks for your answers

Fishwaldo, 2015-06-22:
It sounds like you have some sort of network node id collision etc happening on your network based on issue 4. I'd fix that and you might find the others are also fixed.

I'd try to do a factory reset on your Controller and re-include everything. Then we can look at the other issues if they remain. 

Please use our mailing list at https://groups.google.com/forum/#!forum/openzwave for these types of questions/support. The Issue tracker is only for reporting bugs.

closed 2015-06-22 by Fishwaldo

openzwave-dev-watcher, 2015-06-22:
Hello Justin,

Here am i on the google group discussion.

My controller was reseted a lot of times. I don't have anymore issue with 
the loop at network start. (Some changes were made on the hammering at 
network start on Jeedom)

But considering the minimote, all the 4 points mentionned are recurrent.

Any clue on how to investiguate ?

Greetings