OpenZWave/open-zwave #619: Header file references

rushidesai, 2015-07-06:
I tried installing the library using `make install`, but internal references in the header files don't see to be setup correctly. For instance, value_classes/ValueID.h includes "Defs.h" instead of <openzwave/Defs.h>.

Fishwaldo, 2015-07-07:
thats correct. Your include line for GCC should be -I/usr/local/include/openzwave (or whatever prefix you used).

closed 2015-07-07 by Fishwaldo

rushidesai, 2015-07-07:
Requiring -I/usr/local/include/openzwave beats the purpose for installing it on a system. Headers should be addressable natively, eg>

#include <openzwave/Manager.h>

Also including the /usr/local/include/openzwave path will conflict generic header names like "Manager.h".

Fishwaldo, 2015-07-08:
Almost all source packages default to a /usr/local/ prefix, and I don't know any distribution that by default includes any /usr/local/include directory, so that point is mute. :-1: 

As we maintain API compatibility, and its been this way since the start of the project, fixing it would break builds all over the place (many OSS HomeAutomation projects tell users to clone directly from our repo. If I change stuff that breaks the builds, it breaks all those applications as well.

Anyway, its a 1 line change on most builds, whats the issue?

julienw, 2016-02-10:
Actually there are 2 different issues here. Let me try to explain.

1. the main Openz