OpenZWave/open-zwave #620: Aeon Bulb

michapr, 2015-07-08:
it seems Aeon Bulb is not supported by Open Zwave.
RGB can not be set and dimming (white) does not work stable.

Tested with Domoticz  (OZW1.3-399-ga6703f1).

Thanks!
Michael

ualex73, 2015-07-08:
Your new issue is lacking usefull information. We expect minimum the OZW_Log.txt file.

michapr, 2015-07-08:
Yes, of course you are right, sorry...
While rechecking the OZW log have seen, that there maybe a problem in Domoticz - I have no lines in log while changing colors ... :-S

The described delay maybe because of missing configuration (there seems not to be an XML for this device).
It can be a default value.

Anway there should be additional "warm white" and "cold white" device - this is missing too.
Not sure -- is it OZW related or not - if yes - you need the log while inclusion, I think?

If this all is not important - then close the issue - would be nice to have a device XML.

Thanks!
Michael

ualex73, 2015-07-08:
Checked the color.cpp code, and it should support the cold/warm white color. It looks to detect the capabilities at the start. Without logging, not much we can say.

Fishwaldo, 2015-07-08:
There is no need for a config file for the device. It works fine without it. You can test using the OZW Control Panel to confirm.

Fishwaldo, 2015-07-08:
Btw, the capabilities ValueID will give a hint on what it supports and the Units value of the Color ValueID will also update based on the detected support.

closed 2015-07-08 by Fishwaldo

michapr, 2015-07-08:
hmm.... I tried the OZW panel - dimming is working fine.
But change color - no sucess.... Have tried different values - like
0x02 0xFF 0x03 0x80 0x04 0xFF for color. No change, still white....
(was 0x02 0x00 0x03 0x80 0x04 0x00)
I do not see any line about this change in OZW.log....

Thanks for your help!
Michael

Fishwaldo, 2015-07-08:
Wrong format. It's a RGB value. Eg:

#FF0088 or #000000FF (for warm white) or #00000000FF for cold white. 

(Of course you can mix colors etc)

If your seeing stuff like 0x02 etc I assume you had used the original patch that was supplied? Make sure you delete the zwcfg file. 

You should also see a value list with predefined colors.

Fishwaldo, 2015-07-08:
By the way if you had previously set a warm white or cold white, then try todo colors, you have to turn off warmwhite/cold white:
#rrggbb0000

michapr, 2015-07-08:
Sorry,… maybe stupid question….

 

> I assume you had used the original patch that was supplied?

 

What patch?

Using Version: 1.3-399-ga6703f1

 

The values I see – I see in “current values” – Color (0x02 ….)

Really here write #FF0088 ?

It is another format….

 

And where use #rrggbb0000  ??

 

I’m a little bit confused, sorry.

 

Want to find out – is it a Domoticz issue or not, this could help.

 

Thank you!

Michael

Fishwaldo, 2015-07-08:
Can you send a screenshot?

 

 

What patch?

Using Version: 1.3-399-ga6703f1

 

The values I see – I see in “current values” – Color (0x02 ….)

Really here write #FF0088 ?

It is another format….

 

And where use #rrggbb0000 ??

 

I’m a little bit confused, sorry.

 

Want to find out – is it a Domoticz issue or not, this could help.

 

Thank you!

Michael

 

 

Von: Justin Hammond [mailto:notifications@github.com <mailto:notifications@github.com> ] 
Gesendet: Mittwoch, 8. Juli 2015 16:32
An: OpenZWave/open-zwave
Cc: Michael P.
Betreff: Re: [open-zwave] Aeon Bulb (#620)

 

By the way if you had previously set a warm white or cold white, then try todo colors, you have to turn off warmwhite/cold white:
#rrggbb0000

michapr, 2015-07-08:
Michael

Fishwaldo, 2015-07-08:
Empty? Try emailing directly to the mailing list.

michapr, 2015-07-08:
Outlook „feature“ – have copied from snipping tool direct into outlook message…

 

Now as png attachments.

 

Thanks,

Michael