OpenZWave/open-zwave #623: segfault

wzator, 2015-07-15:
segfault at 4 ip b7695160 sp b5b68c30 error 4 in libopenzwave.so.1.3[b7603000+fa000]
dump:
   92153:       83 80 c4 00 00 00 01    addl   $0x1,0xc4(%eax)
   9215a:       8b 86 f4 04 00 00       mov    0x4f4(%esi),%eax
   92160:       80 78 04 00             cmpb   $0x0,0x4(%eax)

function:
00092080 OpenZWave::Driver::HandleSendDataRequest(unsigned char*, bool):
_ZN9OpenZWave6Driver21HandleSendDataRequestEPhb():

log:
2015-07-15 19:12:07.516 Detail, Node011,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2015-07-15 19:12:07.516 Detail, Node011,   ZW_SEND_DATA delivered to Z-Wave stack
2015-07-15 19:12:17.186 Error, Node011, ERROR: Dropping command, expected response not received after 1 attempt(s)
2015-07-15 19:12:17.186 Detail, Node011, Removing current message
2015-07-15 19:12:17.186 Detail, Node011, Notification: Notification - TimeOut
2015-07-15 19:12:17.526 Detail,   Received: 0x01, 0x05, 0x00, 0x13, 0x0a, 0x01, 0xe2
2015-07-15 19:12:17.526 Detail,   ZW_SEND_DATA Request with callback ID 0x0a received (expected 0x00)

Fishwaldo, 2015-07-20:
is this crash reproducebale (meaning it always crashes?)

I'd need a full gdb backtrace if so.

wzator, 2015-07-22:
sorry no gdb code, not always, after 3-4 days...
i think the problem is Node 11 (sometimes it can't connect to z-wave network) - Danfoss Z-Wave Living Connect (old version).

this never happened before secure class.

segfault:
------------

Index=1
Label=Heating 1
Units=C
Type=Decimal (raw value=18.00)
2015-07-15 19:12:07.140 Detail, Node011,   Received: 0x01, 0x0a, 0x00, 0x04, 0x00, 0x0b, 0x04, 0x46, 0x08, 0x00, 0x7f, 0xcf
2015-07-15 19:12:07.141 Detail,
2015-07-15 19:12:07.141 Info, Node011, Received climate control schedule override report:
2015-07-15 19:12:07.141 Info, Node011,   Override State: None:
2015-07-15 19:12:07.141 Detail, Node011, Refreshed Value: old value=0, new value=0, type=int32
2015-07-15 19:12:07.141 Detail, Node011, Changes to this value are not verified
2015-07-15 19:12:07.141 Detail, Node011, Refreshed Value: old value=127, new value=127, type=uint8
2015-07-15 19:12:07.141 Detail, Node011, Changes to this value are not verified
2015-07-15 19:12:07.141 Detail, Node011, Notification: ValueChanged

!!!!NOTIFICATION!!!!! : 2
ValueChanged: HomeId=23195741 Node=11
Genre=1
CommandClassId=70
Instance=1
Index=8
Label=Override State
Units=
Type=List (raw value=None)
2015-07-15 19:12:07.149 Detail, Node011, Notification: ValueChanged

!!!!NOTIFICATION!!!!! : 2
ValueChanged: HomeId=23195741 Node=11
Genre=1
CommandClassId=70
Instance=1
Index=9
Label=Override Setback
Units=
Type=Byte (raw value=127)

2015-07-15 19:12:07.185 Info, Node011, Received Wakeup Notification from node 11
2015-07-15 19:12:07.185 Detail, Node011, Queuing (WakeUp) WakeUpCmd_NoMoreInformation (Node=11): 0x01, 0x09, 0x00, 0x13, 0x0b, 0x02, 0x84, 0x08, 0x25, 0x0a,
0x4f
2015-07-15 19:12:07.185 Detail,
2015-07-15 19:12:07.185 Info, Node011, Sending (WakeUp) message (Callback ID=0x0a, Expected Reply=0x13) - WakeUpCmd_NoMoreInformation (Node=11): 0x01, 0x09,
0x00, 0x13, 0x0b, 0x02, 0x84, 0x08, 0x25, 0x0a, 0x4f
2015-07-15 19:12:07.516 Detail, Node011,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2015-07-15 19:12:07.516 Detail, Node011,   ZW_SEND_DATA delivered to Z-Wave stack
2015-07-15 19:12:17.186 Error, Node011, ERROR: Dropping command, expected response not received after 1 attempt(s)
2015-07-15 19:12:17.186 Detail, Node011, Removing current message
2015-07-15 19:12:17.186 Detail, Node011, Notification: Notification - TimeOut

!!!!NOTIFICATION!!!!! : 26
STATUS: 1 = 3
2015-07-15 19:12:17.526 Detail,   Received: 0x01, 0x05, 0x00, 0x13, 0x0a, 0x01, 0xe2
2015-07-15 19:12:17.526 Detail,   ZW_SEND_DATA Request with callback ID 0x0a received (expected 0x00)

no answer example
---------------------------

Wakeup log:
2015-07-15 23:31:00
2015-07-15 23:35:59
2015-07-15 23:50:55
2015-07-15 23:55:54
--dead?--
2015-07-16 07:24:55 (battery out/in)

after that no communication with node 11:

!!!!NOTIFICATION!!!!! : 2
ValueChanged: HomeId=23195741 Node=11
Genre=1
CommandClassId=70
Instance=1
Index=8
Label=Override State
Units=
Type=List (raw value=None)
2015-07-16 00:00:55.134 Detail, Node011, Notification: ValueChanged

!!!!NOTIFICATION!!!!! : 2
ValueChanged: HomeId=23195741 Node=11
Genre=1
CommandClassId=70
Instance=1
Index=9
Label=Override Setback
Units=
Type=Byte (raw value=127)
2015-07-16 00:00:58.484 Detail, Node011,   Received: 0x01, 0x05, 0x00, 0x13, 0x9a, 0x00, 0x73
2015-07-16 00:00:58.484 Detail, Node011,   ZW_SEND_DATA Request with callback ID 0x9a received (expected 0x9a)
2015-07-16 00:00:58.484 Info, Node011, Request RTT 4373 Average Request RTT 2301
2015-07-16 00:00:58.484 Detail,   Expected callbackId was received
2015-07-16 00:00:59.802 Detail, Node011,   Received: 0x01, 0x05, 0x00, 0x13, 0x9a, 0x00, 0x73
2015-07-16 00:00:59.802 Detail, Node011,   ZW_SEND_DATA Request with callback ID 0x9a received (expected 0x00)
2015-07-16 00:00:59.802 Warning, Node011, WARNING: Unexpected Callback ID received
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
WTF?

2015-07-16 00:01:04.111 Error, Node011, ERROR: Dropping command, expected response not received after 1 attempt(s)
2015-07-16 00:01:04.112 Detail, Node011, Removing current message
2015-07-16 00:01:04.112 Detail, Node011, Notification: Notification - TimeOut

-----
log after 23:59:00 

TIMER: [TV] Nothing to do
TIMER: [STATE] Nothing to do
parValue: 1
parValue: 0, 519
onState: 251
MUTEX_LOCK: LoadSMS
MUTEX_LOCK_OK: LoadSMS

TIMER GOING
TIMER: [ZONESPOWER] Nothing to do
THERMONODE 11 SET VALUE 18
^^^^^^ no answer

2015-07-16 00:06:44.108 Info, Node011, Value::Set - COMMAND_CLASS_THERMOSTAT_SETPOINT - Heating 1 - 1 - 1 - 18
2015-07-16 00:06:44.108 Detail,
2015-07-16 00:06:44.108 Detail, Node011, Queuing (WakeUp) ThermostatSetpointCmd_Set (Node=11): 0x01, 0x0d, 0x00, 0x13, 0x0b, 0x06, 0x43, 0x01, 0x01, 0x42, 0x
07, 0x08, 0x25, 0xa9, 0x6e
2015-07-16 00:06:44.108 Detail,
2015-07-16 00:06:44.108 Detail, Node011, Queuing (WakeUp) ThermostatSetpointCmd_Get (Node=11): 0x01, 0x0a, 0x00, 0x13, 0x0b, 0x03, 0x43, 0x02, 0x01, 0x25, 0x
aa, 0x21
Command class: 67

sarakha63, 2015-07-27:
This seems related to the other issue as the common point is the thermostat class. It also happens on remotec zx120.
I guess it. Happens each time there is a thermostat class.hope you will find the issue justin. As far as i know it happens on remotec z120, hrt4, heat it and danfoss,

Fishwaldo, 2015-07-28:
Ok. Please test the latest commit and report the status on issue #626

closed 2015-07-28 by Fishwaldo