OpenZWave/open-zwave #626: Secure SRT321 or Horstmann THR4 raise a Segmentation fault durring query stage

nechry, 2015-07-17:
hello,

We find a new issues, and we are sure they started with the security CC merge to master branch.

Secure SRT321 or Horstmann HRT4 Thermostat raise a Segmentation fault during query stage. we got the error if we try to include or if we restart network (with the Thermostat included) and we trigger the Thermostat to send the NIF.

I got also same report from users with the Thermostat Secure SSR303.
From my side I have a Thermostat Secure SCS317, but no Segmentation fault. To reproduce the issues I receive a Secure SRT321 and no way to got this device work with current version. I try a 1.3.337 and a 1.3.333 same issues.

This is the log from openzwave control panel.
```
2015-07-17 09:54:46.065 Always, OpenZwave Version 1.3.333 Starting Up
webserver starting port 8000
post: key=dev data=/dev/ttyUSB21 size=13
post: key=fn data=open size=4
post: key=usb data=false size=5
2015-07-17 09:55:34.056 Info, Setting Up Provided Network Key for Secure Communications
2015-07-17 09:55:34.057 Warning, Failed - Network Key Not Set
2015-07-17 09:55:34.059 Info, mgr,     Added driver for controller /dev/ttyUSB21
post: key=usb data= size=0
2015-07-17 09:55:34.068 Info,   Opening controller /dev/ttyUSB21
2015-07-17 09:55:34.069 Info, Trying to open serial port /dev/ttyUSB21 (attempt 1)
2015-07-17 09:55:34.073 Info, Serial port /dev/ttyUSB21 opened (attempt 1)
2015-07-17 09:55:34.075 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_VERSION: 0x01, 0x03, 0x00, 0x15, 0xe9
2015-07-17 09:55:34.077 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_MEMORY_GET_ID: 0x01, 0x03, 0x00, 0x20, 0xdc
2015-07-17 09:55:34.079 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_CONTROLLER_CAPABILITIES: 0x01, 0x03, 0x00, 0x05, 0xf9
2015-07-17 09:55:34.081 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_GET_CAPABILITIES: 0x01, 0x03, 0x00, 0x07, 0xfb
2015-07-17 09:55:34.082 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_SUC_NODE_ID: 0x01, 0x03, 0x00, 0x56, 0xaa
2015-07-17 09:55:34.085 Detail, 
2015-07-17 09:55:34.087 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x15) - FUNC_ID_ZW_GET_VERSION: 0x01, 0x03, 0x00, 0x15, 0xe9
2015-07-17 09:55:34.093 Detail, contrlr,   Received: 0x01, 0x10, 0x01, 0x15, 0x5a, 0x2d, 0x57, 0x61, 0x76, 0x65, 0x20, 0x33, 0x2e, 0x39, 0x35, 0x00, 0x01, 0x99
2015-07-17 09:55:34.094 Detail, 
2015-07-17 09:55:34.096 Info, contrlr, Received reply to FUNC_ID_ZW_GET_VERSION:
2015-07-17 09:55:34.098 Info, contrlr,     Static Controller library, version Z-Wave 3.95
2015-07-17 09:55:34.099 Detail, Node045,   Expected reply was received
2015-07-17 09:55:34.102 Detail, Node045,   Message transaction complete
2015-07-17 09:55:34.104 Detail, 
2015-07-17 09:55:34.106 Detail, contrlr, Removing current message
2015-07-17 09:55:34.108 Detail, 
2015-07-17 09:55:34.110 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x20) - FUNC_ID_ZW_MEMORY_GET_ID: 0x01, 0x03, 0x00, 0x20, 0xdc
2015-07-17 09:55:34.115 Detail, contrlr,   Received: 0x01, 0x08, 0x01, 0x20, 0xf5, 0x8b, 0xa8, 0x56, 0x01, 0x57
2015-07-17 09:55:34.117 Detail, 
2015-07-17 09:55:34.118 Info, contrlr, Received reply to FUNC_ID_ZW_MEMORY_GET_ID. Home ID = 0xf58ba856.  Our node ID = 1
2015-07-17 09:55:34.125 Detail, Node139,   Expected reply was received
2015-07-17 09:55:34.126 Detail, Node139,   Message transaction complete
2015-07-17 09:55:34.128 Detail, 
2015-07-17 09:55:34.129 Detail, contrlr, Removing current message
2015-07-17 09:55:34.131 Detail, 
2015-07-17 09:55:34.133 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x05) - FUNC_ID_ZW_GET_CONTROLLER_CAPABILITIES: 0x01, 0x03, 0x00, 0x05, 0xf9
2015-07-17 09:55:34.137 Detail, contrlr,   Received: 0x01, 0x04, 0x01, 0x05, 0x08, 0xf7
2015-07-17 09:55:34.139 Detail, 
2015-07-17 09:55:34.140 Info, contrlr, Received reply to FUNC_ID_ZW_GET_CONTROLLER_CAPABILITIES:
2015-07-17 09:55:34.142 Info, contrlr,     There is no SUC ID Server (SIS) in this network.
2015-07-17 09:55:34.143 Info, contrlr,     The PC controller is a primary controller.
2015-07-17 09:55:34.145 Detail, Node247,   Expected reply was received
2015-07-17 09:55:34.146 Detail, Node247,   Message transaction complete
2015-07-17 09:55:34.148 Detail, 
2015-07-17 09:55:34.149 Detail, contrlr, Removing current message
2015-07-17 09:55:34.151 Detail, 
2015-07-17 09:55:34.153 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x07) - FUNC_ID_SERIAL_API_GET_CAPABILITIES: 0x01, 0x03, 0x00, 0x07, 0xfb
2015-07-17 09:55:34.160 Detail, contrlr,   Received: 0x01, 0x2b, 0x01, 0x07, 0x01, 0x00, 0x00, 0x86, 0x00, 0x01, 0x00, 0x5c, 0xfe, 0x81, 0xff, 0x88, 0x4f, 0x1f, 0x00, 0x00, 0xfb, 0x9f, 0x7d, 0xa0, 0x67, 0x00, 0x00, 0x80, 0x00, 0x80, 0x86, 0x00, 0x00, 0x00, 0xe8, 0x73, 0x00, 0x00, 0x0e, 0x00, 0x00, 0x60, 0x00, 0x00, 0xfd
2015-07-17 09:55:34.162 Detail, 
2015-07-17 09:55:34.163 Info, contrlr,  Received reply to FUNC_ID_SERIAL_API_GET_CAPABILITIES
2015-07-17 09:55:34.165 Info, contrlr,     Serial API Version:   1.0
2015-07-17 09:55:34.166 Info, contrlr,     Manufacturer ID:      0x0086
2015-07-17 09:55:34.169 Info, contrlr,     Product Type:         0x0001
2015-07-17 09:55:34.170 Info, contrlr,     Product ID:           0x005c
2015-07-17 09:55:34.173 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_RANDOM: 0x01, 0x04, 0x00, 0x1c, 0x20, 0xc7
2015-07-17 09:55:34.174 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_GET_INIT_DATA: 0x01, 0x03, 0x00, 0x02, 0xfe
2015-07-17 09:55:34.175 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_SET_TIMEOUTS: 0x01, 0x05, 0x00, 0x06, 0x64, 0x0f, 0x97
[... 269 lines cut ...]
2015-07-17 09:56:03.779 Detail, Node005, Notification: ValueAdded
2015-07-17 09:56:03.779 Info, Notification: Value Added Home 0xf58ba856 Node 5 Genre system Class VERSION Instance 1 Index 1 Type string
2015-07-17 09:56:03.781 Detail, Node005, Notification: ValueAdded
2015-07-17 09:56:03.781 Info, Notification: Value Added Home 0xf58ba856 Node 5 Genre system Class VERSION Instance 1 Index 2 Type string
2015-07-17 09:56:03.783 Detail, Node005, Notification: ValueAdded
2015-07-17 09:56:03.784 Info, Notification: Value Added Home 0xf58ba856 Node 5 Genre user Class BATTERY Instance 1 Index 0 Type byte
2015-07-17 09:56:03.786 Detail, Node005, Notification: ValueAdded
2015-07-17 09:56:03.787 Info, Notification: Value Added Home 0xf58ba856 Node 5 Genre user Class THERMOSTAT MODE Instance 1 Index 0 Type list
2015-07-17 09:56:03.788 Detail, Node005, Notification: ValueAdded
2015-07-17 09:56:03.789 Info, Notification: Value Added Home 0xf58ba856 Node 5 Genre user Class SWITCH BINARY Instance 1 Index 0 Type bool
2015-07-17 09:56:03.790 Detail, Node005, Notification: Notification - Node Awake
2015-07-17 09:56:03.792 Info, Notification: Notification home f58ba856 node 5 Awake
2015-07-17 09:56:03.794 Detail, Node005, Query Stage Complete (Probe)
2015-07-17 09:56:03.795 Detail, Node005, AdvanceQueries queryPending=0 queryRetries=0 queryStage=WakeUp live=1
2015-07-17 09:56:03.797 Detail, Node005, QueryStage_WakeUp
2015-07-17 09:56:03.798 Detail, Node005, Queuing (WakeUp) WakeUpCmd_IntervalGet (Node=5): 0x01, 0x09, 0x00, 0x13, 0x05, 0x02, 0x84, 0x05, 0x25, 0x0c, 0x4a
2015-07-17 09:56:03.800 Detail, Node005, Queuing (Query) Query Stage Complete (WakeUp)
2015-07-17 09:56:03.802 Detail, 
2015-07-17 09:56:03.803 Info, Node005, Sending (WakeUp) message (Callback ID=0x0c, Expected Reply=0x04) - WakeUpCmd_IntervalGet (Node=5): 0x01, 0x09, 0x00, 0x13, 0x05, 0x02, 0x84, 0x05, 0x25, 0x0c, 0x4a
2015-07-17 09:56:03.817 Detail, Node005,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2015-07-17 09:56:03.819 Detail, Node005,   ZW_SEND_DATA delivered to Z-Wave stack
Segmentation fault
```

Let me know if you need other informations

Fishwaldo, 2015-07-17:
I'll need a GDB backtrack at least. Logs are not very helpful for segfaults.

nechry, 2015-07-17:
Oki Justin, I'm on a raspberry pi I have to figure how to got your GDB backtrack
I can also confirm the issues still in OpenZwave Version 1.3.401

nechry, 2015-07-17:
I try to use gdb, the control panel wont start
```
pi@raspberrypi ~/open-zwave-control-panel $ gdb ./ozwcp -p 8000 
GNU gdb (GDB) 7.4.1-debian
Copyright (C) 2012 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>...
Reading symbols from /home/pi/open-zwave-control-panel/ozwcp...done.
Attaching to program: /home/pi/open-zwave-control-panel/ozwcp, process 8000
ptrace: No such process.
(gdb) run
Starting program: /home/pi/open-zwave-control-panel/ozwcp 

Program received signal SIGILL, Illegal instruction.
0xb6fe8acc in ?? () from /lib/ld-linux-armhf.so.3
(gdb) 
```

I'm newbie with this tool, me be I go wrong

Fishwaldo, 2015-07-18:
Commandline is gdb ./ozwcp then when GDB is running:
Run -p 8080

nechry, 2015-07-18:
same...
```
pi@raspberrypi ~/zwave-control-panel $ gdb ./ozwcp
GNU gdb (GDB) 7.4.1-debian
Copyright (C) 2012 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>...
Reading symbols from /home/pi/zwave-control-panel/ozwcp...done.
(gdb) run -p 8000
Starting program: /home/pi/zwave-control-panel/ozwcp -p 8000

Program received signal SIGILL, Illegal instruction.
0xb6fe8acc in ?? () from /lib/ld-linux-armhf.so.3
```

ualex73, 2015-07-18:
can you type in:
bt full

(backtrace full) - should give some information where something goes wrong.

nechry, 2015-07-18:
(gdb) bt full
#0  0xb6fe8acc in ?? () from /lib/ld-linux-armhf.so.3
No symbol table info available.
#1  0x00000000 in ?? ()
No symbol table info available.

Fishwaldo, 2015-07-20:
that appears to be a GDB issue. Have you updated your rpi to the very latest?
https://github.com/raspberrypi/linux/issues/766

the other option is to start ozwcp as normal, then attach gdb with the command line "gdb -p <pid of ozwcp>" then trigger the error, and hopefully get a meaningful backtrace.

nechry, 2015-07-20:
Hi Justin, 

so it's not just because I'm not good, I check, but don't found this thread. Now it's my turn out of office until end of week. I will try to update my rpi.

I will also try the other way, launch in background the ozwcp and attach the gdb.

mainly all zwave Thermostat device have a issues, I hope I will give you all expected information to close this issue. You don't have any Thermostat device?

Fishwaldo, 2015-07-20:
I live in the tropics where the temperature doesn't dip below 30! so, no, no Thermostat. 

if all Thermostat have issues, it sounds like something else. Not the Security Class (maybe the new ZWPlus classes etc?)

nechry, 2015-07-20:
I will fugure to report everything you need. The only thing I know it' working until middle of mai before the big merge. I also look nothing changes in any thermostat cc. So yes somewhere else

nechry, 2015-07-20:
By the way, thermostat are also use with climate control for cooling system!

Fishwaldo, 2015-07-21:
The aircon's over here are usually split system wall mounted units. they don't support interfacing with traditional thermostats. Most come with IR remotes and newer ones are WIFI enabled. So no luck there.

nechry, 2015-07-21:
Ok, we got also another issue with the remotec zxt-120 use as zwave bridge with ir device like cooling system, but is my next fight

nechry, 2015-07-26:
start openzwave control panel
 ```
gdb ./ozwcp 
run -p 8000
```
init control panel
and start inclusion
```
2015-07-26 14:20:19.176 Info, ADD_NODE_STATUS_PROTOCOL_DONE
2015-07-26 14:20:19.178 Detail, contrlr, Queuing (Command) Add Node Stop: 0x01, 0x04, 0x00, 0x4a, 0x05, 0xb4
2015-07-26 14:20:19.179 Detail, Notification: ControllerCommand - InProgress
2015-07-26 14:20:19.181 Info, Notification: ControllerCommand home f58ba856 Event 6 Error 0
2015-07-26 14:20:19.182 Detail, 
2015-07-26 14:20:19.184 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x4a) - Add Node Stop: 0x01, 0x04, 0x00, 0x4a, 0x05, 0xb4
2015-07-26 14:20:19.281 Detail, contrlr,   Received: 0x01, 0x07, 0x00, 0x4a, 0x3d, 0x06, 0x0b, 0x00, 0x82
2015-07-26 14:20:19.283 Detail, 
2015-07-26 14:20:19.285 Info, contrlr, FUNC_ID_ZW_ADD_NODE_TO_NETWORK:
2015-07-26 14:20:19.287 Info, contrlr, ADD_NODE_STATUS_DONE
2015-07-26 14:20:19.289 Detail, Node011, AdvanceQueries queryPending=0 queryRetries=0 queryStage=None live=1
2015-07-26 14:20:19.290 Detail, Node011, QueryStage_ProtocolInfo
2015-07-26 14:20:19.292 Detail, Node011, Queuing (Query) Get Node Protocol Info (Node=11): 0x01, 0x04, 0x00, 0x41, 0x0b, 0xb1
2015-07-26 14:20:19.293 Detail, Node011, Queuing (Query) Query Stage Complete (ProtocolInfo)
2015-07-26 14:20:19.294 Info, Node011, Initilizing Node. New Node: true (true)
2015-07-26 14:20:19.297 Detail, Node006,   Expected reply was received
2015-07-26 14:20:19.299 Detail, Node006,   Message transaction complete
2015-07-26 14:20:19.301 Detail, 
2015-07-26 14:20:19.302 Detail, contrlr, Removing current message
2015-07-26 14:20:19.304 Detail, Node011, Notification: NodeAdded
2015-07-26 14:20:19.306 Info, Notification: Node Added Home f58ba856 Node 11 Genre basic Class NO OPERATION Instance 0 Index 0 Type bool
2015-07-26 14:20:19.307 Detail, Notification: ControllerCommand - Completed
2015-07-26 14:20:19.308 Info, Notification: ControllerCommand home f58ba856 Event 7 Error 0
2015-07-26 14:20:19.310 Detail, 
2015-07-26 14:20:19.312 Info, Node011, Sending (Query) message (Callback ID=0x00, Expected Reply=0x41) - Get Node Protocol Info (Node=11): 0x01, 0x04, 0x00, 0x41, 0x0b, 0xb1
2015-07-26 14:20:19.317 Detail, Node011,   Received: 0x01, 0x09, 0x01, 0x41, 0x12, 0x96, 0x00, 0x01, 0x08, 0x00, 0x3b
2015-07-26 14:20:19.319 Detail, 
2015-07-26 14:20:19.321 Info, Node011, Received reply to FUNC_ID_ZW_GET_NODE_PROTOCOL_INFO
2015-07-26 14:20:19.323 Info, Node011,   Protocol Info for Node 11:
2015-07-26 14:20:19.325 Info, Node011,     Listening     = false
2015-07-26 14:20:19.327 Info, Node011,     Frequent      = false
2015-07-26 14:20:19.328 Info, Node011,     Beaming       = true
2015-07-26 14:20:19.330 Info, Node011,     Routing       = false
2015-07-26 14:20:19.331 Info, Node011,     Max Baud Rate = 40000
2015-07-26 14:20:19.333 Info, Node011,     Version       = 3
2015-07-26 14:20:19.334 Info, Node011,     Security      = false
2015-07-26 14:20:19.363 Info, Node011,   Basic device class    (0x01) - Controller
2015-07-26 14:20:19.365 Info, Node011,   Generic device Class  (0x08) - Thermostat
2015-07-26 14:20:19.366 Info, Node011,   No specific device class defined
2015-07-26 14:20:19.369 Info, Node011,     COMMAND_CLASS_BASIC is not mapped
2015-07-26 14:20:19.371 Info, Node011,   Mandatory Command Classes for Node 11:
2015-07-26 14:20:19.373 Info, Node011,     COMMAND_CLASS_BASIC
2015-07-26 14:20:19.375 Info, Node011,     COMMAND_CLASS_WAKE_UP
2015-07-26 14:20:19.376 Info, Node011,   Mandatory Command Classes controlled by Node 11:
2015-07-26 14:20:19.377 Info, Node011,     None
2015-07-26 14:20:19.379 Detail, Node150,   Expected reply was received
2015-07-26 14:20:19.380 Detail, Node150,   Message transaction complete
2015-07-26 14:20:19.381 Detail, 
2015-07-26 14:20:19.383 Detail, Node011, Removing current message
2015-07-26 14:20:19.385 Detail, Node011, Notification: NodeProtocolInfo
2015-07-26 14:20:19.386 Info, Notification: Node Protocol Info Home f58ba856 Node 11 Genre basic Class NO OPERATION Instance 0 Index 0 Type bool
2015-07-26 14:20:19.388 Detail, Node011, Notification: ValueAdded
2015-07-26 14:20:19.389 Info, Notification: Value Added Home 0xf58ba856 Node 11 Genre system Class WAKE UP Instance 1 Index 0 Type int
2015-07-26 14:20:19.391 Detail, Node011, Notification: ValueAdded
2015-07-26 14:20:19.392 Info, Notification: Value Added Home 0xf58ba856 Node 11 Genre basic Class BASIC Instance 1 Index 0 Type byte
2015-07-26 14:20:19.400 Detail, Node011, Query Stage Complete (ProtocolInfo)
2015-07-26 14:20:19.401 Detail, Node011, AdvanceQueries queryPending=0 queryRetries=0 queryStage=Probe live=1
2015-07-26 14:20:19.403 Detail, Node011, QueryStage_Probe
2015-07-26 14:20:19.404 Info, Node011, NoOperation::Set - Routing=true
2015-07-26 14:20:19.407 Detail, Node011, Queuing (NoOp) NoOperation_Set (Node=11): 0x01, 0x09, 0x00, 0x13, 0x0b, 0x02, 0x00, 0x00, 0x25, 0x3e, 0xf7
2015-07-26 14:20:19.409 Detail, Node011, Queuing (Query) Query Stage Complete (Probe)
2015-07-26 14:20:19.410 Detail, 
[... 324 lines cut ...]
#7  MHD_connection_handle_idle (connection=0xedf00) at connection.c:2466
        daemon = 0xee380
        timeout = <optimized out>
        end = <optimized out>
        line = <optimized out>
#8  0x000b5420 in MHD_handle_connection (data=0xedf00) at daemon.c:987
        con = 0xedf00
        num_ready = <optimized out>
        rs = {fds_bits = {512, 0 <repeats 31 times>}}
[... cut ...]
        now = <optimized out>
        p = {{fd = 0, events = 0, revents = 0}}
#9  0xb6e18c00 in start_thread () from /lib/arm-linux-gnueabihf/libpthread.so.0
No symbol table info available.
#10 0xb6dad728 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
No symbol table info available.
#11 0xb6dad728 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
No symbol table info available.
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
Warning: the current language does not match this frame.
```

nechry, 2015-07-26:
if I restart with node, previously added

and hit the send NIF button on Thermostat:
```
2015-07-26 14:28:55.080 Detail, Node011,   Received: 0x01, 0x14, 0x00, 0x49, 0x84, 0x0b, 0x0e, 0x01, 0x08, 0x00, 0x72, 0x86, 0x80, 0x84, 0x31, 0x43, 0x85, 0x70, 0xef, 0x40, 0x25, 0xd7
2015-07-26 14:28:55.082 Detail, 
2015-07-26 14:28:55.084 Info, Node011, UPDATE_STATE_NODE_INFO_RECEIVED from node 11
2015-07-26 14:28:55.085 Info, Node011,   Optional command classes for node 11:
2015-07-26 14:28:55.087 Info, Node011,     COMMAND_CLASS_MANUFACTURER_SPECIFIC
2015-07-26 14:28:55.089 Info, Node011,     COMMAND_CLASS_VERSION
2015-07-26 14:28:55.090 Info, Node011,     COMMAND_CLASS_BATTERY
2015-07-26 14:28:55.091 Info, Node011,     COMMAND_CLASS_WAKE_UP (Existing)
2015-07-26 14:28:55.093 Info, Node011,     COMMAND_CLASS_SENSOR_MULTILEVEL
2015-07-26 14:28:55.095 Info, Node011,     COMMAND_CLASS_THERMOSTAT_SETPOINT
2015-07-26 14:28:55.097 Info, Node011,     COMMAND_CLASS_ASSOCIATION
2015-07-26 14:28:55.099 Info, Node011,     COMMAND_CLASS_CONFIGURATION
2015-07-26 14:28:55.101 Info, Node011,   Optional command classes controlled by node 11:
2015-07-26 14:28:55.102 Info, Node011,     COMMAND_CLASS_THERMOSTAT_MODE
2015-07-26 14:28:55.104 Info, Node011,     COMMAND_CLASS_SWITCH_BINARY
2015-07-26 14:28:55.105 Detail, Node011, AdvanceQueries queryPending=1 queryRetries=0 queryStage=Probe live=1
2015-07-26 14:28:55.108 Info, Node011,   Node 11 has been marked as awake
2015-07-26 14:28:55.109 Detail, Node011, Queuing (Query) Query Stage Complete (Probe)
2015-07-26 14:28:55.111 Detail, Node011, Notification: ValueAdded
2015-07-26 14:28:55.112 Info, Notification: Value Added Home 0xf58ba856 Node 11 Genre system Class VERSION Instance 1 Index 0 Type string
2015-07-26 14:28:55.114 Detail, Node011, Notification: ValueAdded
2015-07-26 14:28:55.115 Info, Notification: Value Added Home 0xf58ba856 Node 11 Genre system Class VERSION Instance 1 Index 1 Type string
2015-07-26 14:28:55.118 Detail, Node011, Notification: ValueAdded
2015-07-26 14:28:55.119 Info, Notification: Value Added Home 0xf58ba856 Node 11 Genre system Class VERSION Instance 1 Index 2 Type string
2015-07-26 14:28:55.121 Detail, Node011, Notification: ValueAdded
2015-07-26 14:28:55.122 Info, Notification: Value Added Home 0xf58ba856 Node 11 Genre user Class BATTERY Instance 1 Index 0 Type byte
2015-07-26 14:28:55.124 Detail, Node011, Notification: ValueAdded
2015-07-26 14:28:55.125 Info, Notification: Value Added Home 0xf58ba856 Node 11 Genre user Class THERMOSTAT MODE Instance 1 Index 0 Type list
2015-07-26 14:28:55.127 Detail, Node011, Notification: ValueAdded
2015-07-26 14:28:55.128 Info, Notification: Value Added Home 0xf58ba856 Node 11 Genre user Class SWITCH BINARY Instance 1 Index 0 Type bool
2015-07-26 14:28:55.130 Detail, Node011, Notification: Notification - Node Awake
2015-07-26 14:28:55.131 Info, Notification: Notification home f58ba856 node 11 Awake
2015-07-26 14:28:55.133 Detail, Node011, Query Stage Complete (Probe)
2015-07-26 14:28:55.134 Detail, Node011, AdvanceQueries queryPending=0 queryRetries=0 queryStage=WakeUp live=1
2015-07-26 14:28:55.136 Detail, Node011, QueryStage_WakeUp
2015-07-26 14:28:55.138 Detail, Node011, Queuing (WakeUp) WakeUpCmd_IntervalGet (Node=11): 0x01, 0x09, 0x00, 0x13, 0x0b, 0x02, 0x84, 0x05, 0x25, 0x3f, 0x77
2015-07-26 14:28:55.139 Detail, Node011, Queuing (Query) Query Stage Complete (WakeUp)
2015-07-26 14:28:55.141 Detail, 
2015-07-26 14:28:55.142 Info, Node011, Sending (WakeUp) message (Callback ID=0x3f, Expected Reply=0x04) - WakeUpCmd_IntervalGet (Node=11): 0x01, 0x09, 0x00, 0x13, 0x0b, 0x02, 0x84, 0x05, 0x25, 0x3f, 0x77

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0xb64e1460 (LWP 2214)]
0x0003ba8c in OpenZWave::Manager::GetValueListSelection(OpenZWave::ValueID const&, std::string*) ()
(gdb) 
```
and the backtrace full:
```
(gdb) bt full
#0  0x0003ba8c in OpenZWave::Manager::GetValueListSelection(OpenZWave::ValueID const&, std::string*) ()
No symbol table info available.
#1  0x0001e248 in Webserver::web_get_values (this=0xeda78, i=11, ep=0xb3b00e38) at webserver.cpp:249
        strs = {<std::_Vector_base<std::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::basic_string<char, std::char_traits<char>, std::allocator<char> > > >> = {
            _M_impl = {<std::allocator<std::basic_string<char, std::char_traits<char>, std::allocator<char> > >> = {<__gnu_cxx::new_allocator<std::basic_string<char, std::char_traits<char>, std::allocator<char> > >> = {<No data fields>}, <No data fields>}, _M_start = 0x0, _M_finish = 0x0, 
              _M_end_of_storage = 0x0}}, <No data fields>}
        str = {static npos = <optimized out>, _M_dataplus = {<std::allocator<char>> = {<__gnu_cxx::new_allocator<char>> = {<No data fields>}, <No data fields>}, 
            _M_p = 0xb6f9bd30 ""}}
        valueElement = 0xb3b02b08
        vals = 0x105480
        id = {m_id = 189792260, m_id1 = 16777216, m_homeId = 4119570518}
        str = {static npos = <optimized out>, _M_dataplus = {<std::allocator<char>> = {<__gnu_cxx::new_allocator<char>> = {<No data fields>}, <No data fields>}, 
            _M_p = 0xb6f9bd30 ""}}
        j = 6
        idcnt = 8
#2  0x000214ec in Webserver::SendPollResponse (this=0xeda78, conn=0xedf00) at webserver.cpp:776
        listening = false
        flirs = false
[... 34 lines cut ...]
        timeout = <optimized out>
        end = <optimized out>
[... cut ...]
        now = <optimized out>
        p = {{fd = 0, events = 0, revents = 0}}
#9  0xb6e18c00 in start_thread () from /lib/arm-linux-gnueabihf/libpthread.so.0
No symbol table info available.
#10 0xb6dad728 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
No symbol table info available.
#11 0xb6dad728 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
No symbol table info available.
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) 
```

let me know if you need any more details

Fishwaldo, 2015-07-26:
Looks like the same issue as #381

nechry, 2015-07-26:
Hello, but the 381 as be reported before june 2015. We get this issue with 75% of thermostats device after the merge of june. But yes is possible they have a root case in common.

Actually this issue block a big percentage of jeedom users 1/20 have a thermostat. The only work around is to exclude the node or use version before juin 2015, but they also loose the gen5 compatibility... If they use previous version.

Tell me everything it's can be usefull for you to help (us) with this issues.

Fishwaldo, 2015-07-27:
The bug has always been there for the longest time. It could crash on previous versions, but I suspect that the security rewrite has just altered the memory layout enough to trigger the segfault. 

I'll try to look into it soon.

xs4, 2015-07-28:
@nechry i can try to provide a patch, if you can test it.

xs4, 2015-07-28:
You can find the patch here xs4/open-zwave#1 , let me know if it solves your issue.

Fishwaldo, 2015-07-28:
XS4's patch (already merged into master) and some additional work should fix this up. Can you test and revert?

nechry, 2015-07-28:
Hi we will do some tests tonight.
Thank

sarakha63, 2015-07-28:
Hello justin i continue here as you asked.

i will try with remotec as soon as i get home

sarakha63, 2015-07-28:
Hello justin , let's wait the test from nechry for the hrt4

But i can confirm that i succesfully included my remotec with the last version

Good Job ;)

(BTW ; we are preparing some PR for confs files as promised ;) )

nechry, 2015-07-28:
Hello Justin, 

I confirm, like sarakha63, the latest version with XS4 patch work! 
So my Secure SRT321 (Horstmann THR4) include without error. Restart the network and wakeup the device and no more segmentation fault.

So is a great news! I give my entire gratitude for you help with gdb and for the resolution of this issues. 

I let you close this issue. Thank again for everything you do days after days on openzwave!

closed 2015-07-29 by Fishwaldo