OpenZWave/open-zwave #629: Issue excluding remotes

sarakha63, 2015-07-20:
Hello Justin,

I have 4 remote controls (keyfibs, minimote...) they are not used as secondary controllers but as scene remotes. I wanted to exclude the fourth one and i kept redoing it 4 times to exclude it (the 3 first times excluding the 3 other remotes)

I read the log 

2015-07-20 19:26:03.914 Info, REMOVE_NODE_STATUS_REMOVING_CONTROLLER
2015-07-20 19:26:03.914 Info, Alternative controller lookup found more then one match. Using the first one found.
2015-07-20 19:26:03.914 Info, Alternative controller lookup found more then one match. Using the first one found.

So i guess it is wanted. But is it possible to avoid this problem. Imagine someone having 10 remotes and who wants to exclude the last one included. he will be forced to exclude all the other nines.

Do you have any idea on implementing an othe way for excluding remotes ?

And again thank you for all your work

Fishwaldo, 2015-07-21:
You have obviously included the remotes as controllers, or done a Transfer/Replication to them at one stage. 

As far as what you are hitting, older controllers (I know the Aeotec Z-Stick S2 for sure) don't return the Node Number when doing a controller removal, so thats the only way. Newer controllers seem to return the Node Number of the Controller when doing a removal, and this wont happen. 

This is pretty much a protocol issue. Not much OZW can do to influence it.

closed 2015-07-21 by Fishwaldo

sarakha63, 2015-07-21:
Hello Justin,

I think i don't understand. The keyfobs, minimote and panic button can only be included one way. It's after inclusion that you configure them to act as controllers, scene activators, group activators etc....

Do you have any advice, on how to avoid this issue, if theres is any way or trick to do so ?

Because for all the four of them i only included them without any special things (no transfr replication or so)

One of them is the last keyfob gen 5.

I understand quite well the issue and the trick you did as there are no node number returned. But did you experiment a way to include them and avoid the issue. If there are not, i will live with it ;) It's not a big issue after all