OpenZWave/open-zwave #632: No Battery Information On SES301 Temp Sensor

ado464, 2015-07-28:
It is found in openzwave 1.3 r410 as an unknown device with type = "000d" and id="0001", said manufacturer was Horstmann so i added it to the manufacturer.xml file like this:

<Product type="000d" id="0001" name="SES 301 Temperature Sensor"/>

It reports temperature, but i can not get the battery information out of it even tho it has "BATTERY" in supported classes, i have also assosiated both group1 & 2 to node1 for reports but still no battery.

Info from pepper1:
http://www.pepper1.net/zwavedb/device/309

Help would be appreciated on how to go on next, thanks a lot guys.

ualex73, 2015-07-28:
Most likely you need to poll the battery value yourself, devices don't
always send it by itself.

ado464, 2015-07-28:
Thanks for the quick reply alex, i would love to know how to poll it myself if you could please explain. I had it running on razberry as well and the battery came up by itself, that is why i got confused when it didn't show up in openzwave.

Thanks again

Fishwaldo, 2015-07-31:
see Manager::EnablePoll (http://www.openzwave.com/dev/classOpenZWave_1_1Manager.html#a50d795cb20a0bea55ecfd4a02c9777f3)

closed 2015-07-31 by Fishwaldo