OpenZWave/open-zwave #633: Adding support for Widom Energy Driven switch

robybob64, 2015-07-28:
Energy driven switch is a new widom device that provides voltage, current, power and power factor measurement. More over it may directly control a slave switch in order to activate loads when events occur. 

<Product type="1214" id="0304" />
(first draft XML are ready to be submitted here)

robybob64, 2015-07-28:
Here you are 
Please insert the following code within widom Manufacturer Id (manufacturer_specific.xml):

    
    <Product type="1214" id="0304" name="UME304 Energy Driven Switch" config="widom/UME304C_S.xml"/>

Then file UME304C_S.xml, first draft, is the following: 

    <?xml version="1.0" encoding="UTF-8"?>
    <Product xmlns='http://code.google.com/p/open-zwave/'>

    

        
        <CommandClass id="112">
                <Value type="list" genre="config"  index="1" label="Device status 1 click" size="1" min="1" max="4" value="1">
                        <Help>Device status when the external switch receives 1 click</Help>
                        <Item label="Toggle (default)" value="1" />
                        <Item label="On" value="2" />
                        <Item label="Off" value="3" />
                        <Item label="Ignore" value="4" />
                </Value>

                <Value type="list" genre="config"  index="20" label="Device status Basic Set" size="1" min="1" max="4" value="1">
                        <Help>Device status when ZWave command Basic Set is received</Help>
                        <Item label="As Received (default)" value="1" />
                        <Item label="Ignore if On" value="2" />
                        <Item label="Ignore if Off" value="3" />
                        <Item label="Ignore" value="4" />
                </Value>

                <Value type="short" genre="config"  index="30" label="Overcurrent level" size="2" min="0" max="4500" value="4500" units="centiAmperes">
                        <Help>C VERSION. Sets the current level above which the time spent above that level is measured. For S VERSION. Sets to 1250 the default and max current level above which the time spent above that level is measured  (Hundredths of Amperes)</Help>
                </Value>

                <Value type="short" genre="config"  index="31" label="Overcurrent time" size="2" min="0" max="10800" value="10" units="seconds">
                        <Help>Sets the time beyond which, in the case of overcurrent at a level higher than that established in parameter 30, an OverCurrent event occurs (seconds)</Help>
                </Value>

                <Value type="list" genre="config"  index="32" label="Next state" size="1" min="0" max="3" value="1">
                        <Help>Defines the next state of the device if an overcurrent event has occurred.</Help>
                        <Item label="Ignore" value="1" />
                        <Item label="On" value="2" />
                        <Item label="Off" value="3" />
                        <Item label="Toggle" value="4" />
                </Value>

                <Value type="int" genre="config"  index="33" label="Meter reset" size="4" min="0" max="4294967296" value="1">
                        <Help>Provides total operating time from last meter reset. When set to 0 it resets the cumulated values of energy and total operating time.</Help>
                </Value>

                <Value type="list" genre="config"  index="34" label="Energy flow" size="1" min="1" max="2" value="1">
                        <Help>Determines whether energy events are related to energy consumed or energy produced</Help>
                        <Item label="Consumed" value="1" />
                        <Item label="Produced" value="2" />
                </Value>

                <Value type="short" genre="config"  index="35" label="UP Power Level" size="2" min="0" max="11250" value="11250" units="Watts">
                        <Help>C VERSION. Sets the level of instantaneous power in Watts beyond which time of permanence above this level is calculated. For S VERSION. Sets to 3000 the default and max level of instantaneous power in Watts beyond which time of permanence above this level is calculated.</Help>
                </Value>

                <Value type="short" genre="config"  index="36" label="UP Power Time" size="2" min="0" max="10800" value="10" units="seconds">
                        <Help>Sets the time in seconds beyond which, if instantaneous power remains at levels exceeding threshold defined by parameter 35, a UP Power event occurs.</Help>
                </Value>

                <Value type="list" genre="config"  index="37" label="UP Power State" size="1" min="0" max="3" value="0">
                        <Help>Defines the next state of the device if an overcurrent event has occurred.</Help>
                        <Item label="Ignore" value="0" />
                        <Item label="On" value="1" />
                        <Item label="Off" value="2" />
[... 115 lines cut ...]
                        <Item label="switch" value="2" />
                        <Item label="Automatic recognition" value="4" />
                </Value>

                <Value type="list" genre="config"  index="63" label="Load control" size="1" min="1" max="4" value="1">
                        <Help>Defines the load control mode.(default 1)</Help>
                        <Item label="Direct control" value="1" />
                        <Item label="Contactor (normally open)" value="2" />
                        <Item label="Contactor (normally closed)" value="3" />
                        <Item label="As external indicator" value="4" />
                </Value>
        </CommandClass>

        
        <CommandClass id="133">
         <Associations num_groups="3">
[... cut ...]
                        <Group index="2" max_associations="8" label="Devices controlled by the Up Power event" auto="false" />
                        <Group index="3" max_associations="8" label="Devices controlled by the Down Power event" auto="false"/>
         </Associations>
        </CommandClass>

</Product>

Please note, there are 2 versions with the same productId (C and S), the previous XML has been confugured for C version. In the case you have S version please manually change param 30 and 35 according instructions.

Fishwaldo, 2015-07-31:
Next time you cat put it all in one Pull Request.

closed 2015-07-31 by Fishwaldo