OpenZWave/open-zwave #642: Building open-zwave

Beddie, 2015-08-03:
Building ValueButton.o
/install/open-zwave-read-only/cpp/src/command_classes/Powerlevel.cpp: In member function ‘bool OpenZWave::Powerlevel::Set(uint8)’:
/install/open-zwave-read-only/cpp/src/command_classes/Powerlevel.cpp:355:2: error: ‘powerLevel’ may be used uninitialized in this function [-Werror=uninitialized]
/install/open-zwave-read-only/cpp/src/command_classes/Powerlevel.cpp: In member function ‘bool OpenZWave::Powerlevel::Test(uint8)’:
/install/open-zwave-read-only/cpp/src/command_classes/Powerlevel.cpp:436:33: error: ‘powerLevel’ may be used uninitialized in this function [-Werror=uninitialized]
cc1plus: all warnings being treated as errors
/install/open-zwave-read-only/cpp/build/support.mk:106: recept voor doel '/install/open-zwave-read-only/.lib/Powerlevel.o' is mislukt
make[1]: *** [/install/open-zwave-read-only/.lib/Powerlevel.o] Fout 1
make[1]: *** Wachten op onvoltooide taken...
make[1]: Map '/install/open-zwave-read-only/cpp/build' wordt verlaten
Makefile:20: recept voor doel 'all' is mislukt
make: *** [all] Error 2

Fishwaldo, 2015-08-03:
Already fixed.

closed 2015-08-03 by Fishwaldo

Beddie, 2015-08-03:
Great!

Beddie, 2015-08-03:
Not quite yet, what am i doing wrong?

Building FileOpsImpl.o
Building TimeStampImpl.o
/install/open-zwave-read-only/cpp/src/command_classes/Powerlevel.cpp: In member function ‘bool OpenZWave::Powerlevel::Set(uint8)’:
/install/open-zwave-read-only/cpp/src/command_classes/Powerlevel.cpp:355:2: error: ‘powerLevel’ may be used uninitialized in this function [-Werror=uninitialized]
/install/open-zwave-read-only/cpp/src/command_classes/Powerlevel.cpp: In member function ‘bool OpenZWave::Powerlevel::Test(uint8)’:
/install/open-zwave-read-only/cpp/src/command_classes/Powerlevel.cpp:436:33: error: ‘powerLevel’ may be used uninitialized in this function [-Werror=uninitialized]
cc1plus: all warnings being treated as errors
/install/open-zwave-read-only/cpp/build/support.mk:106: recept voor doel '/install/open-zwave-read-only/.lib/Powerlevel.o' is mislukt
make[1]: *** [/install/open-zwave-read-only/.lib/Powerlevel.o] Fout 1
make[1]: *** Wachten op onvoltooide taken...
make[1]: Map '/install/open-zwave-read-only/cpp/build' wordt verlaten
Makefile:20: recept voor doel 'all' is mislukt
make: *** [all] Error 2

Fishwaldo, 2015-08-03:
Confirm line 331 looks like this:
PowerLevelEnum powerLevel = PowerLevel_Normal;

Otherwise you didn't pull the latest version