OpenZWave/open-zwave #644: Device Configuration support version

nechry, 2015-08-07:
Hi.

The Fibaro FGS221 have different parameter definition according to the version
for paramter index 3:

in version 1.4 - 1.7
Parameter no. 3 - Automatic turning off relay after set time
default value 0 Available configuration parameters:
0 - Auto OFF disabled for both relays
1 - Auto OFF enabled for relay no. 1. disabled for relay no. 2.
2 - Auto OFF disabled for relay no. 1. enabled for relay no. 2.
3 - Auto OFF enabled for both relays

in version 1.9 - 1.10 and version 2.1 - 2.3
Parameter No. 3 - Auto off relay after specified time, with the possibility of manual override - immediate Off after button push. Default value: 0 Available settings:
0 - manual override disabled. After single button push the relay turns on and automatically turns off after specified time.
1 - manual override enabled. After single button push the relay turns on and automatically turns off after specified time. Another button push turns the relay off immediately

the actual ozw config file is target  version 1.9 - 1.10 and version 2.1 - 2.3

We have this issue with other fibaro device like fgk001 see my https://github.com/OpenZWave/open-zwave/pull/500

A way to define device configuration according to the version can be a good enhancement for ozw librarie.

In some case the product type and product id can be a part of the solution but I don't have a matching key with version.

thanks

Fishwaldo, 2015-08-07:
yes - Its planned, but will require some rework of the core code as version info comes from the device only "after" we have loaded the configuration. Its probably a 1.6 timeline

nechry, 2015-08-07:
yes the only solution here is to setup a hybrid parameter 3, without list value and a complex help.

nechry, 2016-03-02:
Yesterday I receive the new Fibaro FGMS-001 is a zwave+ version 3.02.
Fibaro keep old Product Type et ID, the associations groups are in number of 5, previously 3. and me be some parameters are also different, I did not yet find the doc for v3.02. 

We must find a way to apply a specific configuration file to correctly support this device.