OpenZWave/open-zwave #647: Ezmotion issue basic Set and multi encapsulated

sarakha63, 2015-08-13:
Hi Justin,

I have both Ezmotions (The first one with IDs [0x001e] [0x0002] [0x0001] and the Ezmotion + version with ids  [0x001e] [0x0002] [0x0002])

The second one works perfectly with <CommandClass id="32" setasreport="true"/>

But the first one as some issues :

in fact setting <CommandClass id="32" setasreport="true"/> doesn't work as OZW doesn't give any 48 CC only 49 CC. 

The only way i made it work was using this in the config file

<CommandClass id="32" create_vars="true" ignoremapping="true" setasreport="true">
    <Instance index="1"/>
        <Value type="bool" genre="user" instance="1" index="0" label="Sensor" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="False" />
            <SensorMap index="0" type="12" />
	</CommandClass>

This is not clean but it works pretty well. But after an OZW restart the command CC 32 disappears with  some errors in the log : "Invalid ValueID passed to GetValueAsBool and so on) and the command is removed.

I think this is expected.

Do you have any idea to make this device work (i can send you one if you want, i own 5 of them)

I think the initial problem is only related to bad multiencapasulated mappings.

I also tried 
 setting an endpoint to CC 48 it worked but it caused some dropping command issues in the log and the device stops from responding.

Do you have any idea (do you want one device ?). 

Thank you

PS :  we also noticed some lots of strange behaviour with lots of device with multi encapsulated and multi-instances). But we will mail you some informations (its is systematic on some Qubino devices)

Thank you for reading

Fishwaldo, 2015-08-18:
I'd need logs without any of the modifications you made to see whats going on. Manually adding CommandClasses shouldn't be necessary unless its really buggy. Please send logs to the mailing list

Fishwaldo, 2015-09-15:
Any update on the logs?

sarakha63, 2015-10-18:
Now that i have a special plateform with ozwcp. I will.give you the exact log you need

sarakha63, 2015-11-02:
'''
2015-11-02 12:10:49.252 Info, FUNC_ID_ZW_ADD_NODE_TO_NETWORK:
2015-11-02 12:10:49.253 Info, ADD_NODE_STATUS_ADDING_SLAVE
2015-11-02 12:10:49.253 Info, Adding node ID 24 - Non-Secure
2015-11-02 12:10:49.254 Detail, Notification: ControllerCommand - InProgress
2015-11-02 12:10:49.254 Info, Notification: ControllerCommand home 0184d0d0 Event 6 Error 0
2015-11-02 12:10:49.770 Detail, Received: 0x01, 0x07, 0x00, 0x4a, 0x0c, 0x05, 0x18, 0x00, 0xa3
2015-11-02 12:10:49.771 Detail, 
2015-11-02 12:10:49.771 Info, FUNC_ID_ZW_ADD_NODE_TO_NETWORK:
2015-11-02 12:10:49.772 Info, ADD_NODE_STATUS_PROTOCOL_DONE
2015-11-02 12:10:49.772 Detail, contrlr, Queuing (Command) Add Node Stop: 0x01, 0x04, 0x00, 0x4a, 0x05, 0xb4
2015-11-02 12:10:49.773 Detail, Notification: ControllerCommand - InProgress
2015-11-02 12:10:49.773 Info, Notification: ControllerCommand home 0184d0d0 Event 6 Error 0
2015-11-02 12:10:49.774 Detail, 
2015-11-02 12:10:49.774 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x4a) - Add Node Stop: 0x01, 0x04, 0x00, 0x4a, 0x05, 0xb4
2015-11-02 12:10:49.826 Detail, contrlr, Received: 0x01, 0x07, 0x00, 0x4a, 0x0c, 0x06, 0x18, 0x00, 0xa0
2015-11-02 12:10:49.827 Detail, 
2015-11-02 12:10:49.827 Info, contrlr, FUNC_ID_ZW_ADD_NODE_TO_NETWORK:
2015-11-02 12:10:49.828 Info, contrlr, ADD_NODE_STATUS_DONE
2015-11-02 12:10:49.828 Info, Node024, Network Key Not Set - Secure Option is not required
2015-11-02 12:10:49.829 Info, Node024, Basic device class (0x04) - Routing Slave
2015-11-02 12:10:49.829 Info, Node024, Generic device Class (0x21) - Multilevel Sensor
2015-11-02 12:10:49.830 Info, Node024, Specific device class (0x01) - Routing Multilevel Sensor
2015-11-02 12:10:49.830 Info, Node024, COMMAND_CLASS_BASIC will be mapped to COMMAND_CLASS_SENSOR_MULTILEVEL
2015-11-02 12:10:49.831 Info, Node024, Mandatory Command Classes for Node 24:
2015-11-02 12:10:49.831 Info, Node024, COMMAND_CLASS_SENSOR_MULTILEVEL
2015-11-02 12:10:49.831 Info, Node024, Mandatory Command Classes controlled by Node 24:
2015-11-02 12:10:49.832 Info, Node024, COMMAND_CLASS_BASIC
2015-11-02 12:10:49.832 Info, Node024, Optional command classes for node 24:
2015-11-02 12:10:49.833 Info, Node024, COMMAND_CLASS_MULTI_INSTANCE/CHANNEL
2015-11-02 12:10:49.833 Info, Node024, COMMAND_CLASS_SENSOR_MULTILEVEL (Existing)
2015-11-02 12:10:49.834 Info, Node024, COMMAND_CLASS_CONFIGURATION
2015-11-02 12:10:49.834 Info, Node024, COMMAND_CLASS_WAKE_UP
2015-11-02 12:10:49.835 Info, Node024, COMMAND_CLASS_ASSOCIATION
2015-11-02 12:10:49.835 Info, Node024, COMMAND_CLASS_BATTERY
2015-11-02 12:10:49.836 Info, Node024, COMMAND_CLASS_MANUFACTURER_SPECIFIC
2015-11-02 12:10:49.836 Info, Node024, COMMAND_CLASS_NODE_NAMING
2015-11-02 12:10:49.837 Info, Node024, COMMAND_CLASS_VERSION
2015-11-02 12:10:49.837 Info, Node024, Optional command classes controlled by node 24:
2015-11-02 12:10:49.838 Info, Node024, COMMAND_CLASS_BASIC (Existing)
2015-11-02 12:10:49.838 Info, Node024, None
2015-11-02 12:10:49.838 Detail, Node024, AdvanceQueries queryPending=0 queryRetries=0 queryStage=None live=1
2015-11-02 12:10:49.839 Detail, Node024, QueryStage_ProtocolInfo
2015-11-02 12:10:49.839 Detail, Node024, Queuing (Query) Get Node Protocol Info (Node=24): 0x01, 0x04, 0x00, 0x41, 0x18, 0xa2
2015-11-02 12:10:49.839 Detail, Node024, Queuing (Query) Query Stage Complete (ProtocolInfo)
2015-11-02 12:10:49.839 Info, Node024, Initilizing Node. New Node: true (true)
2015-11-02 12:10:49.840 Detail, Node006, Expected reply was received
2015-11-02 12:10:49.840 Detail, Node006, Message transaction complete
2015-11-02 12:10:49.840 Detail, 
2015-11-02 12:10:49.841 Detail, contrlr, Removing current message
2015-11-02 12:10:49.841 Detail, Node024, Notification: NodeAdded
2015-11-02 12:10:49.842 Info, Notification: Node Added Home 0184d0d0 Node 24 Genre basic Class NO OPERATION Instance 0 Index 0 Type bool
2015-11-02 12:10:49.842 Detail, Node024, Notification: NodeProtocolInfo
2015-11-02 12:10:49.842 Info, Notification: Node Protocol Info Home 0184d0d0 Node 24 Genre basic Class NO OPERATION Instance 0 Index 0 Type bool
2015-11-02 12:10:49.843 Detail, Node024, Notification: ValueAdded
2015-11-02 12:10:49.843 Info, Notification: Value Added Home 0x0184d0d0 Node 24 Genre system Class WAKE UP Instance 1 Index 0 Type int
2015-11-02 12:10:49.843 Detail, Node024, Notification: ValueAdded
2015-11-02 12:10:49.843 Info, Notification: Value Added Home 0x0184d0d0 Node 24 Genre user Class BATTERY Instance 1 Index 0 Type byte
2015-11-02 12:10:49.844 Detail, Node024, Notification: ValueAdded
2015-11-02 12:10:49.844 Info, Notification: Value Added Home 0x0184d0d0 Node 24 Genre system Class VERSION Instance 1 Index 0 Type string
2015-11-02 12:10:49.844 Detail, Node024, Notification: ValueAdded
2015-11-02 12:10:49.845 Info, Notification: Value Added Home 0x0184d0d0 Node 24 Genre system Class VERSION Instance 1 Index 1 Type string
2015-11-02 12:10:49.845 Detail, Node024, Notification: ValueAdded
2015-11-02 12:10:49.845 Info, Notification: Value Added Home 0x0184d0d0 Node 24 Genre system Class VERSION Instance 1 Index 2 Type string
2015-11-02 12:10:49.846 Detail, Notification: ControllerCommand - Completed
2015-11-02 12:10:49.846 Info, Notification: ControllerCommand home 0184d0d0 Event 7 Error 0
2015-11-02 12:10:49.847 Detail, 
[... cut ...]
0x31 with a GENERAL CC instead of The Sensor (0x30) cc so even with basic seasreport it doesn't work

The only way i made it work is with this in the config file

<CommandClass id="32" setasreport="true" ignoremapping="true"/>
  <CommandClass id="32">
       <Instance index="1" />
      <Value type="bool" genre="user" instance="1" index="0" label="Sensor" read_only="true" write_only="false"  min="0" max="0" value="false" />
      <SensorMap index="0" type="12" />
  </CommandClass>  
  But this is not persistant the 0x20 CC is removed when OZW restarts

Fishwaldo, 2015-11-04:
Did the device really go to sleep at 2015-11-02 12:10:50.882?

Could you redo the log and wake up the device if it goes to sleep. Its strange though, as it sends the response at 2015-11-02 12:10:50.925

Looking at it, the controller reports no ACK at 882, this means that the stick sent the message out to the network, but it didn't ACK the message back to the controller, but at 925, it sends the response to the "unacked" message. 

Because of this, we never send out the MultiInstace Capabilities for Instance 3+ (unless a WAKEUP message comes in later), hence things wont be "setup" correctly.

Can you confirm the above behavior is consistent (ie, we always get the No ACK - Device Asleep flow?)

nechry, 2016-01-29:
normally this config 
https://github.com/jeedom/plugin-openzwave/blob/master/ressources/openzwave/config/homeseer/hsm100.xml
will resolve this issue.

nechry, 2016-02-29:
you can close after PR:
https://github.com/sarakha63/open-zwave/commit/707b7333b067688de28dd200964a852bc1d4653d
will be apply.

closed 2016-10-06 by nechry