OpenZWave/open-zwave #653: Recalibration parameter for HEM G1

ado464, 2015-08-27:
To start recalibration of these aeotec G1 hem meters a parameter 252 is needed to unlock the meter, followed by the 253 parameter to set the new settings.

Parameter 252 [4 byte dec]: 2982335232 (or 0xB1C2D300) will unlock the HEM G1 for calibration.

What you send into parameter 253 will determine the calibration. The calculation is:
WATT = Vrms*Currnt*TotalPhase

Once you do this calculation, you should have parameter 253 ready to be sent in after parameter 252.

Steps:

1) Send in Parameter 252 [4 byte hex] 0xB1C2D300, after this is sent, you will have 1 minute to send in Parameter 253.

2) Send in Parameter 253 [4 byte dec]: WATT

3) Once the HEM G1 starts calibration, the LED will begin to blink fast to show that it is calibrating. Any messages that the HEM has received from any device or your gateway will not be responded to during the calibration.

We need this implemented in the xml file for configuration.

Best regards,
Ado

nechry, 2016-06-08:
I don't see any param 252 in documentation:
https://aeotec.freshdesk.com/helpdesk/attachments/6009584509

Fishwaldo, 2016-06-22:
@ado464 Can you let us know where you got param 252 from?

ado464, 2016-06-22:
Hello :) i got it straight from Aeon Labs :)

nechry, 2016-06-23:
Hi @ado464 I got answer for Aeon-labs.

This procedure is not recommended. So to avoid any issues from un-experimented users I prefer don't add this param to the existing configuration file.

As I propose, stop your network manually edit your zwcfgxxx.xml file for add the definition for param 252
```
<Value type="list" index="252" genre="config" label="Unlock Command" size="4" value="0" write_only="true">
    <Help>Send Unlock Command</Help>
    <Item label="Normal" value="0"/>
    <Item label="Unlock" value="2982335232"/>
</Value>
```

save and re-start your network the parameter will be present and normally you will enable to perform calibration step.

ado464, 2016-06-23:
Yeah we have a contact at Aeon labs, a head technician named Chris. We have been working with Aeon labs a while now and they recommended we do a recalibration because the meter arrived in poor condition with hay-wire values. We have had a number of runnings with aeon labs on firmware version on meters, some worked, some worked but counted x3 as much, and some didn't work at all. Currently the firmware 1.08 seems stable (on 3 clamp 3 phase meters), so if anyone is ordering in, order it with firmware 1.08 :)

nechry, 2016-06-23:
I got also the answer from Chris! I don't know if is the same

I will close the issue

closed 2016-06-23 by nechry

Fishwaldo, 2016-06-23:
@ado464: it would be great if you could post that info at http://discourse.db.openzwave.com/t/aeotec-home-energy-meter/80 so people in the future know about this option when they visit http://www.openzwave.com/device-database/0086%3a0009%3a0002