OpenZWave/open-zwave #657: Setting Kwikset lock code - Code format?

darkwolfxp, 2015-09-10:
Hi,

I am having some issues setting up new codes for my Kwikset 916. I have added it and I can issue lock/unlock commands.

However, in the Open Zwave Control Panel, the codes seem to be in hexadecimal. I have tried to convert my pin codes to hexadecimal and submit but they seem to be invalid.
I also tried to add just the raw pin code, it gets converted to Hexadecimal by the lock but doesn't seem to be converting correctly either.

Does anyone know what "type" of value needs to be inserted?

Kind Regards
Junio

Fishwaldo, 2015-09-10:
The codes are usually stored as Ascii - So you need to convert to Ascii representation first. 
(eg - 1 Decimal = 0x31, a 0x61)

closed 2015-09-10 by Fishwaldo

ronluna, 2016-07-17:
Hello, I'm having a similar problem when trying to set a pin into the lock. 
I've been sending the pin in ascii mode but somehow openzwave change the ascii code to something else and the pin code that gets created is different from the desired original pin.
Example:

If I try to Pin 111 using ascii code 0x31 0x31 0x31 0x31 0x50 0x7f 0x00 0x00 0x00 0x00
after submission and polling the lock it returns 0x00 0x92 0x05 0x60 0x50 0x7f 0x00 0x00 0x00 0x00 which is totally different from what it was originally sent.

Any ideas?