OpenZWave/open-zwave #658: Ignoring configuration parameter when set

m0rphi, 2015-09-10:
Hi All!

I am trying to set some config parameters with openzwave (my system specs are below) and the **Fibaro FGMS-001 Motion-Sensor**.
In openzwave I use the method setConfigParam. 
E.g. I am configuring the parameter **"motion alarm cancellation delay"**, which has the the param_id 6 and a value of 30 in my case.

For some reasons when I execute the **setConfigParam** once and I manually wake up the sensor, the value will not set. (the full log I uploaded to http://www.file-upload.net/download-10901523/OZW_Log.log.html )

First I try to set the the value of 10 (default was 30)
[image]

I manually wake up the sensor , get an ACK that setting the parameter was successfull
[image]

Now I am trying to set the parameter to 30:
[image]

I manually wake up the sensor again, Now the parameter will not set correctly (it will be used the "old" one)
[image]

When I am executing the setConfigParam twice in a role, the parameter will be set.
The parameter will also set, when the sensor wakes up automatically.

Any ideas why this behaviour occurs and can I fix it somehow?
My currently workaround is also to set the parameter twice, but this is not a "real" solution for that problem.

Thanks in advice for any response ;)

my system:
OS: Linux Mint 17 32 Bit `3.13.0-24-generic #47-Ubuntu SMP Fri May 2 23:31:42 UTC 2014 i686 i686 i686 GNU/Linux`
openzwave-commit: a8ef10c
HW: `Z-Wave Aeotec Z-Stick S2`

nechry, 2015-09-10:
I have sometime this behavior, but really not every time, most of time it's work at first try. In really few case, I must try many times the same parameter before is apply. If a configuration parameter always refuse a value, it's also a possible device configuration file with error (data type, size, etc)

Fishwaldo, 2015-09-15:
I can see where the problem lies. the attempt to set Param 6 to 30 gets dropped after its moved to the wakeup queue, as the code incorrectly assumes 1 attempt to send it has already occurred. 

Please try the latest commit and let us know how it goes.

m0rphi, 2015-09-15:
Thanks for response Justin!
Right now I get an error like nechry, when starting openzwave. See https://github.com/OpenZWave/open-zwave/commit/ca123d66b5cc9cb85e05d3e175a8505f92abe79e

Fishwaldo, 2015-09-15:
Should be fixed now.

In this case, it wasn't a particular message that was causing problems, it was that when a device goes to sleep we move all the messages to a "wakeup" queue, where its held till the device wakes up. 

The problem was (and it was there for years I suspect, but only reared its ugly head when I disabled OZW retries recently) is that when the device did wake up, there was a counter on the each message in the queue in terms of how many times it had been attempted. With the new no-retry option, the first message counter was already set to 1 attempt and since we don't retry anymore, it was dropped. Other messages behind it in the wakeup queue had a attempt count of 0, so they went through ok. 

In your case, the dropped message was the Configuration_Set command, hence why if you set it twice, the 2nd message would work. 

Anyway, I'm pretty confident that this will fix it. :8ball:

m0rphi, 2015-09-16:
tested, works like a charm with newest commit 0a73afbe4114e8d6f0593b55f13a48130d095bb9  :smiley:

what´s interesting, how the queues are working.
1. Structure: Is for any node an extra wakeup queue or is there one global wakeup queue for all nodes?
2. Methods: Can I access the queues with methods (for example reading the wakeup queues or maybe cancelling a message in the queue) - didn´t find any method in the documentation
3. Is there something like a timeout in the queues which deletes  messages? e.g. an config parameter couldn´t be set since 1 Week, therefore delete the config parameter message)
4. What´s happening with duplicated GET/SET Messages? (e.g. asking twice the same config parameter)

Thanks in advance for your answers!

nechry, 2015-09-16:
I also try this morning and no more segfault. Thank for the quick patch. I will also check if I got a influence for my Vision Secure devices for battery consuming they have something around the wakeup.

Fishwaldo, 2015-09-17:
@m0rphi: Its "complicated" to say the least:

1) There is a WakeUp Queue for each device that supports the Wakeup CC. (The queue is actually managed by the Wakeup CC)

At the Global Level, I think there are upto 7 Queues created (but I should drop it to 6 soon with the security rewrite - I keep forgetting to do that). These are basically setup in a priority order, so things like "polling" requests don't block actions that the user might be performing etc. It all gets a bit hairy in there. :-1: 

2) No. A lot of what goes on at the at the SerialAPI level is transactional. Removing messages etc would probably break the "protocol" in some cases. The "actual" message stored in each queue is essentially a byte stream, so unless you reimplement the serial api decoding in your app, even getting access to the message wouldn't tell you what it is. 

3) Nope. For same reason as above. The Queue system is pretty basic in that it doesn't know what the messages contain, it just knows when it gets to the top of the queue, send it out. 

4) I'm not sure what you mean there.

closed 2015-09-17 by Fishwaldo

m0rphi, 2015-09-17:
Thanks for the answers!

4) Just thought that there is somethink like a duplicate check. An example use case case: the first time the sensitivity of a motion sensor will be set to 50 (set command is going into the queue, because the device is sleeping), but afterwards it will be set to 80. Is openzwave delete the first sensitivity set because of the duplcate set afterwards?

Fishwaldo, 2015-09-17:
No. The only queue that checks for duplicates is the wakeup queue, and it must be a exact match, so in your example they are different so would be queued up. 

For "normal" queues, as i mentioned, we dont inspect existing messages and remove settings etc as it might be something the user wants todo. Eg switching a light on and off for a "disco!"  Queues have no idea if its a set/get or for configuration or switches etc. To a queue its essential just a byte stream waiting to be sent.

m0rphi, 2015-09-25:
I think I have a strange behaviour again by setting parameters with  **Fibaro FGMS-001 Motion-Sensor**.
When I want to include/reinclude a device, I am waiting for the Notification **NodeQueriesComplete**.
When this notification occured, I set some default parameters via setConfigParam.
[image]

This log looks ok, the sensor is sleeping, so the message will be send to the wakeup queue.

Now the sensor wakes up, and it seems to be the wakeup queue is emtpy (which shouldn´t):
[image]

Has someone an idea why the parameters will be not set? Thanks in advance ;)

The full log is attached here; http://www.filedropper.com/ozw

my system:
OS: Linux Mint 17 32 Bit `3.13.0-24-generic #47-Ubuntu SMP Fri May 2 23:31:42 UTC 2014 i686 i686 i686 GNU/Linux`
openzwave commit: ea9b0157f94942e699be2b6dd7f286cb9fadff07
HW: `Z-Wave Aeotec Z-Stick S2`

Fishwaldo, 2015-09-27:
This is bizarre. 

But why are you "re-including" it in your network. You should first delete it from the network, then include it again. 

Could you please retry without doing the "reinclude"?

m0rphi, 2015-10-04:
Seems to be there was an issue on threading/multiprocessing on my side, therefore it "hangs" on setting parameters. Now it´s working!