OpenZWave/open-zwave #660: Aeotec micro smart switch v2

wackoracoon, 2015-09-24:
Hello,

I use the micro smart switch v2 with domoticz with open zwave.
I have several notes/suggestion about it because for basic usage it's works well (on/off switch is ok) but parameters are wrong, in the manufacturer_specific.xml the xml associated with the device is ses.xml instead of msesv2.xml. I looked into the engineering documentation to be sure.
So I changed this line : 
```
--- manufacturer_specific.xml	2015-09-25 00:19:45.185409291 +0200
+++ manufacturer_specific.new	2015-09-25 00:20:13.565408634 +0200
@@ -64,7 +64,7 @@
                 <Product type="0003" id="0018" name="Smart Switch (2nd Edition)"/>
 		<Product type="0003" id="001a" name="Micro Switch (2nd Edition)"/>
 		<Product type="0003" id="001b" name="Micro Smart Energy Illuminator G2"/>
-		<Product type="0003" id="0012" name="Micro Smart Energy Switch G2" config="aeotec/msesv2.xml" />
+		<Product type="0003" id="0012" name="Micro Smart Energy Switch v2" config="aeotec/ses.xml" />
 		<Product type="0003" id="0013" name="Micro Smart Energy Dimmer v2" config="aeotec/msesv2.xml" />
 		<Product type="0003" id="004b" name="Smart Energy Switch Gen5" config="aeotec/ses2.xml"/>
 		<Product type="0003" id="004e" name="Heavy Duty Smart Switch Gen5" config="aeotec/hdss_gen5.xml" />
```
The other point is I made some change to fit the documentation for example "Current Overload Protection" is 15.5A (parameter 3), I also change the descriptions for parameters 101, 102 and 103 (there is no multisensor to report but voltage and current to add) and to finish I changed the <MAJ> used in first letter of world for some description to fit correct typography rule :). I can put the complete file somewhere (pastebin...) or just send it to you by mail.

It's all for suggestions, let's talk about my problems now !
I try to set the 101 parameter to 15 to get the complete report (kWh, W, amp and V) but if I put this value either the value goes to an other (12, 8 or 4) or the node goes into dead state... I don't understand why I have this behavior...

I motivated to help you to improve open zwave working with this node, if you want some logs or some tests I could make tell me !

Thanks guys !

Fishwaldo, 2015-09-27:
> changed the used in first letter of world for some description to fit

Can you submit a Pull Request for the changes to the msesv2 file?
https://github.com/OpenZWave/open-zwave/wiki/Adding-Devices
I've updated the manufacture_specific.xml file already.

wackoracoon, 2015-09-29:
Done ! I push the changes in the file, I'm not familiar with pull request so I hope you'll see what I modified.
And I still have a problem with my micro smart switch, when I send new parameters for reports (101, 102, 103) they are not apply for others values than 4,8. Sometimes 12 is apply but not always.
I don't know if it a domoticz or openzwave issue, I enable debug for openzwave and try to analyze exchanges but it's quite hard to understand because I don't know well how the protocol works.
Maybe I can open a new issue ? I can paste the exchange between my doogle and my switch. I'm also available to make tests if you want.

Fishwaldo, 2015-10-06:
I'd need to see the logs to see why you can't set those paramaters.

Fishwaldo, 2015-11-25:
The PR was invalid and I've not seen any further response. Closing this.

closed 2015-11-25 by Fishwaldo