OpenZWave/open-zwave #663: Z-Wave.Me ZME_KFOB-S 4 Button Key Fob

ben33880, 2015-10-01:
Hi,

I just acquire Z-Wave.Me ZME_KFOB-S 4 Button Key Fob and is missing in the configuration file "manufacturer_specific.xml" on line 740 should be added -> config = "zwave.me/kfob. xml ".

Moreover, in the "zwave.me/kfob.xml" file, there are parameter 11 duplicate the setting on line 61. I think it should be the parameter 21.

Thank you.

PS: sorry for my english .....
PS1 : tested on domoticz

nechry, 2015-10-01:
hello the z-wave.me zmekfob-s4 is not totally like the z-wave.me zmekfob-4

I prepare a specific config for this remote:
https://github.com/jeedom/plugin-openzwave/commit/fc5fa50e36ee632e1ecf6f1746dccaec41719fe2

the zmekfob-s4 have 5 associations groups, not 4 like the zmekfob-4
yes, the zmekfob-4 have a duplicate parameter 11
parameter 20, 24 don't exist on zmekfob-s4
And I fix some default values 

we will merge jeedom devices configuration really soon, I a PR to ozw.

ben33880, 2015-10-01:
ok cool, for information with domoticz,i use the file kfob.xml and it works perfectly.

Fishwaldo, 2015-10-06:
Closing and waiting for the Jeedom PR

closed 2015-10-06 by Fishwaldo