OpenZWave/open-zwave #666: Device don't return in sleep mode

nechry, 2015-10-06:
I fount a weird issues with some Vision Secure battery operated devices.
the zg8108 for example, is a garage door sensor. from time to time with openzwave during the startup interview he will wakeup but never return in sleep mode. If he reach this behavior he stop send any notifications like sensor..The testNode, healNode response directly, is really awake. I also measure current consummation just to be sure is not a missing sleep notification. 

the only way to restore a normal state is to remove the battery. It's like during startup interview ozw send a command and the device become incomparable. The same device with the same controller but other zwave gateway, like zway don't get this issues. we currently have many users with this issues, we actually see on Vision Secure zg8108,  zd2101 and zp3102. We also try exclusion/inclusion but same issues. the only workaround is to remove battery during openzwave startup... 

I will check to provide a log

Fishwaldo, 2015-10-07:
Related to issue #441 - But these devices are obviously buggy. They are meant to automatically go back to sleep after a timeout anyway. Closing this issue and we can track it in #441

closed 2015-10-07 by Fishwaldo

nechry, 2015-10-07:
hi Justin, I really understand your point off view about this case. It's my initial response to users with this issues. but the fact is the same device with other zwave controller don't reach this issue. I finally think openzwave has a role is this issues, but yes. the device also is part of picture to.

We discover this strange behavior this summer, but we don't know if it's related to a change in openzwave this summer or it's only time circumstance because we migrate user form zway to openzwave with all device. From my side I got a Vision Secure zg8108 at end of June, I'm pretty sure he work good until august, it's mean is not "directly" after the Security CC merge. 

So in resume, a Vision Secure zg8108 in zway will always return in sleep after the timeout, but with onpenzwave the same device with the same controller will not return every time.

thank any way for all work time!

nechry, 2016-03-02:
Justin, can we reopen this issue, or you prefer I create a new one?

Fishwaldo, 2016-03-03:
No Need. #441 will fix it.