OpenZWave/open-zwave #671: Issue with ZME_WALLC-S controller and the standard XML configuration file

rgroothuis, 2015-10-15:
Using the latest beta version of Domotics V2.3304 which has openzwave Version: 1.3-444-gfd7d67f installed (embedded) I was not able to get the ZME_WALLC-S working using the Aeotec zwave dongle (http://aeotec.com/z-wave-usb-stick Gen5, Model ZW909-C). I was able to include the ZME-WALLC-S controllers into the zwave network but I was not able to trigger any events in Domoticz. 

On the Domoticz forum I found the following discussion: https://www.domoticz.com/forum/viewtopic.php?f=6&t=4454&start=20#p39683 in which a different XML config file for this controller is provided. When I used this file I was able to get events triggered in Domoticz from the controller. 

Maybe somebody with good zwave knowledge can look into the differences of these two files (the original in Domiticz and the one provided in the forum) and tries to explain why the one from the forum is working and the original one not. I want to get into a situation that when upgrading to a newer version of openzwave/Domoticz my setup keeps working using the original files. Maybe the file from the forum can be included in the next release.

I cannot upload the two files I'm referring to thats why I've made the two files available through Dropbox.

Original file: https://dl.dropboxusercontent.com/u/10020964/ZME_WALLC-S.xml.original.zip

Working file from the forum: https://dl.dropboxusercontent.com/u/10020964/ZME_WALLC-S.xml.fromweb.zip

Fishwaldo, 2015-10-19:
There was a duplicate parameter 11 in your file, I assume the 2nd one was meant to be 21.

Fishwaldo, 2015-10-19:
Fixed.

closed 2015-10-19 by Fishwaldo