OpenZWave/open-zwave #673: Error, ERROR: Not enough space in stream buffer

KiwiHC16, 2015-10-16:
Hi All,
After a day or so, I start to have messages in OZ...log file:
Error, ERROR: Not enough space in stream buffer
open-zwave-control-panel run on a raspberry pi platform. I tried with Raspian Wheezy and Jessie with same issue.
I tried with r736, r886, r 929, r1014 and the last git version of open-zwave, same issue.
I tried with few version of openzwave-control-panel, still same issue.
I did not find any good chat around this issue on the net.
What is very surprising is that I never faced this issue in the past. After a crash of the raspberry pi , I did a complete re-installation and since I have this issue.
I think also that after sometime the ozcwp crash. I have activated core dump today. Let see if I get a core tomorrow.
The error message is in the Stream.cpp file from openzwave.
Any idea where to start to progress on the trouble shooting ?
Any workaround ?
Thanks.

Fishwaldo, 2015-10-18:
Is your rpi heavily loaded? That message indicates that we are generating messages faster than we can send to the serial port.

KiwiHC16, 2015-10-19:
Thanks for this indication.
I have this error message again this morning. It’s in the log every second.
The system is totally unresponsive. I tried to change a temperature for a Danfoss Zwave and the radiateur remains at old temperature.
I can see in the log that it receive the request:
———————
2015-10-19 07:48:06.061 Error, ERROR: Not enough space in stream buffer
2015-10-19 07:48:07.064 Error, ERROR: Not enough space in stream buffer
2015-10-19 07:48:08.066 Error, ERROR: Not enough space in stream buffer
2015-10-19 07:48:08.113 Info, Node032, Value::Set - COMMAND_CLASS_THERMOSTAT_SETPOINT - Heating 1 - 1 - 1 - 21.00
2015-10-19 07:48:08.114 Detail,
2015-10-19 07:48:08.117 Detail, Node032, Queuing (WakeUp) ThermostatSetpointCmd_Set (Node=32): 0x01, 0x0d, 0x00, 0x13, 0x20, 0x06, 0x43, 0x01, 0x01, 0x42, 0x08, 0x34, 0x25, 0xda, 0x05
2015-10-19 07:48:08.118 Detail,
2015-10-19 07:48:08.118 Detail, Node032, Queuing (WakeUp) ThermostatSetpointCmd_Get (Node=32): 0x01, 0x0a, 0x00, 0x13, 0x20, 0x03, 0x43, 0x02, 0x01, 0x25, 0xdb, 0x7b
2015-10-19 07:48:09.062 Error, ERROR: Not enough space in stream buffer
2015-10-19 07:48:10.062 Error, ERROR: Not enough space in stream buffer
2015-10-19 07:48:11.061 Error, ERROR: Not enough space in stream buffer
———————————————
We can also see that the issue started at 7:18am (There is no message between 7:11am and 7:18am while normally I have messages every minutes or so.
2015-10-19 07:11:08.753 Detail,
2015-10-19 07:11:08.754 Info, Node034, Received Battery report from node 34: level=71
2015-10-19 07:11:08.754 Detail, Node034, Refreshed Value: old value=71, new value=71, type=byte
2015-10-19 07:11:08.754 Detail, Node034, Changes to this value are not verified
2015-10-19 07:11:08.755 Detail, Node034, Notification: ValueChanged
2015-10-19 07:11:08.755 Info, Notification: Value Changed Home 0x0161e169 Node 34 Genre user Class BATTERY Instance 1 Index 0 Type byte
2015-10-19 07:18:49.373 Error, ERROR: Not enough space in stream buffer
2015-10-19 07:18:50.374 Error, ERROR: Not enough space in stream buffer
2015-10-19 07:18:51.374 Error, ERROR: Not enough space in stream buffer
2015-10-19 07:18:52.374 Error, ERROR: Not enough space in stream buffer
2015-10-19 07:18:53.374 Error, ERROR: Not enough space in stream buffer

From CPU, point of view :
%Cpu(s):  1,0 us,  1,7 sy,  0,0 ni, 96,6 id,  0,3 wa,  0,0 hi,  0,3 si,  0,0 st

Disk space:
pi@zwave ~ $ df -h
Sys. de fichiers Taille Utilisé Dispo Uti% Monté sur
/dev/root          7,2G    4,4G  2,5G  65% /
devtmpfs           214M       0  214M   0% /dev
tmpfs              218M       0  218M   0% /dev/shm
tmpfs              218M     25M  193M  12% /run
tmpfs              5,0M    4,0K  5,0M   1% /run/lock
tmpfs              218M       0  218M   0% /sys/fs/cgroup
/dev/mmcblk0p1      56M     20M   37M  36% /boot
tmpfs               44M       0   44M   0% /run/user/1000

Nothing strange from that side. The CPU can go for one minute to 100% as I run some rrdtool to generate some graph. But none of these at 7:18am this morning when the issue started.

I have 27 node on the Zwave network connected to a Z-Stick S2.

I have one zwave node which is always in a strange state in openzwave-control-panel (not in sleeping mode). Probably a radio coverage issue. Could it create a bad situation on the serial port (node unresponsive for example) ? 

One additional question: Will openzwave be able to configure and discuss with a Node using mesh network if there is no direct link between the dongle and the node at startup ? Will the controller be able to discover the node with the neighbors/topology reports and create a routing to the node ?

Thanks foryour command and help.

closed 2015-10-21 by Fishwaldo

MisterV, 2015-12-02:
I have exactly the same issue, but i do not have any process that have crashed 
How to debug the root cause of this ? 

thanks 
Vincent

openzwave-dev-watcher, 2015-12-02:
To give more information about the contexte 

I am running the version 1.3.482 of the openzwave library
I do not have any issue for months 
My hardware is raspberry

few days ago i had alarm that has been triggered from on device (without 
being able to identify which one) and i have my fibaro switch device that 
were flipping from on to off. But nothing in the OZW log file...
And now i get this error with no space left on stream buffer. 

How do I can debug what is happening ?

openzwave-dev-watcher, 2015-12-03:
I have tested with the last version openzwave and after 24h i have one more 
time the same issue...
Help plz
Vincent

openzwave-dev-watcher, 2015-12-07:
As far as I can tell, this is a OS issue and not anything related to OZW per say. 

The error occurs if the Serial Port Buffers get full. The buffers are controlled by the Operating system and not OZW. It basically means it the OS can’t send messages to the Serial port (fast enough) so a heavily loaded system or other issue could be the cause. 

  
You received this message because you are subscribed to the Google Groups "OpenZWave" group.
To unsubscribe from this group and stop receiving emails from it, send an email to openzwave+unsubscribe@googlegroups.com.
For more options, visit https://groups.google.com/d/optout.

jonferreira, 2018-05-10:
@openzwave-dev-watcher I've been battling with this problem for a couple of weeks now, running 1.4-2888-g54682370-dirty on domoticz.

I just can't see how it's can possibly be a "OS issue" caused by a "heavily loaded system",  given that this a RPI running just Domo and node-red and the problem happens overnight when nothing but power reports happen, etc

RPI CPU averages at 5% and memory at 40%.